margelo/react-native-vision-camera #8: How can i install type definitions? ❓

me1k, 2021-02-28:
## Question

Hi, i saw your post about the Camera library on Reddit and directly wanted to try it out in my Typescript project. I also can see that you use Typescript in the Project. But my project is complaining, that there are no type definitions. How can i install them or use your module now? :)

mrousavy, 2021-02-28:
Hey! Types are available in the latest version, that's `1.0.0-alpha.6`. Which version are you on?

me1k, 2021-02-28:
Thank you for your answer! When i install it with yarn, i installs 1.0.0-alpha.1

mrousavy, 2021-02-28:
Yes, that's because only alphas are available on npm. In the README I have mentioned that you have to install it with the @alpha suffix (Or manually change the version in package.json)

me1k, 2021-03-01:
I changed it manually in the package.json and it works fine now. Thank you very much. I'm looking forward to use your camera component :)

closed 2021-03-01 by mrousavy