margelo/react-native-vision-camera #10: ❓ API docs?

joebernard, 2021-03-01:
I was hoping to explore some API docs for the camera. Do you have those available somewhere or should I just rely on the example app for now?

mrousavy, 2021-03-01:
Hi! I currently don't have docs for web-viewing available, but everything is thoroughly documented in the JSDoc (TypeScript definitions). Ideally I want to automatically build those and host them on GitHub pages, will have to take a look at that when I find some free time. 👍

mrousavy, 2021-03-01:
Let's leave this issue open until docs are available

closed 2021-03-03 by mrousavy

mrousavy, 2021-03-04:
Hey! Forgot to get back to you. Docs are live at [cuvent.github.io/react-native-vision-camera](https://cuvent.github.io/react-native-vision-camera/), and API is automatically generated at [/docs/api](https://cuvent.github.io/react-native-vision-camera/docs/api) - let me know what you think 😄👍

joebernard, 2021-03-04:
Excellent. I'm using `react-native-camera` right now and was always disappointed I could not use the multiple lenses on the later iPhones.  Really excited to experiment with this and having good docs will help.

PR #11 (2021-03-03): Automatically build API documentation
Automatically builds API documentation (A static website in `docs/`, will be hosted using GitHub pages) from the JSDoc comments in the TypeScript source files.

Closes #10 

## Blocker

* [ ] It doesn't pick up `export type ...`, only `export interface ...` works :(
* [ ] The default theme is ugly as hell