margelo/react-native-vision-camera #9: 🐛

bairock, 2021-03-01:
const devices = useCameraDevices('wide-angle-camera')
 const device = devices.back

device id of undefined

crash app

android 10
xiaomi note 7

mrousavy, 2021-03-01:
Please fill out the issue template you have deleted. What version are you on? Share crash logs. Print out the device (`console.log(JSON.stringify(device))`).

bairock, 2021-03-01:
my code 
```
import React from 'react'
import { useState, useEffect, useMemo } from 'react'
import { StyleSheet } from 'react-native'
import { Camera, sortDevices } from 'react-native-vision-camera'

const App = () => {
  const [devices, setDevices] = useState([])
  const device = useMemo(() => devices.find((d) => d.position === 'back'), [devices])

  useEffect(() => {
    const loadDevices = async () => {
      try {
        const availableCameraDevices = await Camera.getAvailableCameraDevices()
        const sortedDevices = availableCameraDevices.sort(sortDevices)
        setDevices(sortedDevices)
      } catch (e) {
        console.error('Failed to get available devices!', e)
      }
    }
    loadDevices()
  }, [])
  if (device && device.id) {
    console.log(JSON.stringify(device))
    return <></>
  }

  return (
    <Camera
      style={StyleSheet.absoluteFill}
      device={device}
      isActive={true}
    />
  )
}
export default App
```
error device id of undefined

```
  static getDerivedStateFromProps(props: CameraProps, state: CameraState): CameraState | null {
    const newCameraId = props.device.id; <===== error
    if (state.cameraId !== newCameraId) return { ...state, cameraId: newCameraId };

    return null;
  }
```

(console.log(JSON.stringify(device)))

```
{
   "devices":[
      "wide-angle-camera"
   ],
   "formats":[
      {
         "autoFocusSystem":"none",
         "colorSpaces":[
            "Array"
         ],
         "fieldOfView":68.25930547065248,
         "frameRateRanges":[
            "Array"
         ],
         "isHighestPhotoQualitySupported":true,
         "maxISO":3200,
         "maxZoom":8,
         "minISO":100,
[... 598 lines cut ...]
         "photoHeight":176,
         "photoWidth":144,
         "supportsPhotoHDR":false,
         "supportsVideoHDR":false,
         "videoHeight":176,
         "videoStabilizationModes":[
            "Array"
         ],
         "videoWidth":144
      }
   ],
   "hasFlash":true,
   "hasTorch":true,
   "id":"0",
   "isMultiCam":false,
   "maxZoom":8,
   "minZoom":1,
   "name":"back (0)",
   "neutralZoom":0,
   "position":"back",
   "supportsDepthCapture":false,
   "supportsLowLightBoost":false,
   "supportsRawCapture":true
}
```

mrousavy, 2021-03-01:
That's because your logic is wrong, you are mounting the `<Camera>` when device is still `undefined`/`null`

bairock, 2021-03-01:
I fixed it but it still crashes

my code
```

import React from 'react'
import { useState, useEffect, useMemo } from 'react'
import { StyleSheet } from 'react-native'
import { Camera, sortDevices } from 'react-native-vision-camera'

const App = () => {
  const [devices, setDevices] = useState([])
  const device = useMemo(() => devices.find((d) => d.position === 'back'), [devices])
  const [permissons, setPermissons] = useState(false)

  const getPermissons = async () => {
    const cameraPermission = await Camera.getCameraPermissionStatus()
    const microphonePermission = await Camera.getMicrophonePermissionStatus()

    if (microphonePermission === 'authorized' && cameraPermission === 'authorized') {
      setPermissons(true)
    }
  }

  useEffect(() => {
    const loadDevices = async () => {
      try {
        const availableCameraDevices = await Camera.getAvailableCameraDevices()
        const sortedDevices = availableCameraDevices.sort(sortDevices)
        setDevices(sortedDevices)
      } catch (e) {
        console.error('Failed to get available devices!', e)
      }
    }
    loadDevices()
    getPermissons()
  }, [])

  if (!device && !permissons) {
    return null
  }

  return (
    <Camera
      style={StyleSheet.absoluteFill}
      device={device}
      isActive={true}
    />
  )

}

export default App

```
logs
```
2021-03-01 20:55:32.271 1808-1808/? E/com.artoolapp: Unknown bits set in runtime_flags: 0x8000
2021-03-01 20:55:32.571 1808-1808/com.artoolapp E/libc: Access denied finding property "ro.vendor.df.effect.conflict"
2021-03-01 20:55:32.586 1808-1808/com.artoolapp E/SoLoader: couldn't find DSO to load: libjscexecutor.so caused by: dlopen failed: library "libjsc.so" not found result: 0
2021-03-01 20:55:32.659 1808-1857/com.artoolapp E/Perf: Fail to get file list com.artoolapp
2021-03-01 20:55:32.659 1808-1857/com.artoolapp E/Perf: getFolderSize() : Exception_1 = java.lang.NullPointerException: Attempt to get length of null array
2021-03-01 20:55:32.660 1808-1857/com.artoolapp E/Perf: Fail to get file list com.artoolapp
2021-03-01 20:55:32.660 1808-1857/com.artoolapp E/Perf: getFolderSize() : Exception_1 = java.lang.NullPointerException: Attempt to get length of null array
2021-03-01 20:55:34.996 1808-1867/com.artoolapp E/ReactNativeJNI: logMarker CREATE_REACT_CONTEXT_END
2021-03-01 20:55:34.996 1808-1866/com.artoolapp E/unknown:ReactNative: ReactInstanceManager.createReactContext: mJSIModulePackage null
2021-03-01 20:55:34.997 1808-1867/com.artoolapp E/ReactNativeJNI: logMarker RUN_JS_BUNDLE_START
2021-03-01 20:55:35.000 1808-1868/com.artoolapp E/unknown:ReactNative: ReactInstanceManager.attachRootViewToInstance()
2021-03-01 20:55:35.110 1808-1868/com.artoolapp E/unknown:ReactRootView: runApplication: call AppRegistry.runApplication
2021-03-01 20:55:35.191 1808-1808/com.artoolapp E/unknown:ReactNative: ReactInstanceManager.attachRootViewToInstance()
2021-03-01 20:55:35.192 1808-1808/com.artoolapp E/unknown:ReactRootView: runApplication: call AppRegistry.runApplication
2021-03-01 20:55:35.732 1808-1867/com.artoolapp E/ReactNativeJNI: logMarker RUN_JS_BUNDLE_END
```

mrousavy, 2021-03-01:
Your `if` is still wrong.

```js
if (!device || !permissions) {
  return null
}
```

The errors logs don't seem to be related to vision camera. See if your apps works when you remove the Camera code.

bairock, 2021-03-01:
sorry for my wrong code, i fixed it but the app still crashes

my error logs
```
2021-03-01 20:55:32.271 1808-1808/? E/com.artoolapp: Unknown bits set in runtime_flags: 0x8000
2021-03-01 20:55:32.571 1808-1808/com.artoolapp E/libc: Access denied finding property "ro.vendor.df.effect.conflict"
2021-03-01 20:55:32.586 1808-1808/com.artoolapp E/SoLoader: couldn't find DSO to load: libjscexecutor.so caused by: dlopen failed: library "libjsc.so" not found result: 0
2021-03-01 20:55:32.659 1808-1857/com.artoolapp E/Perf: Fail to get file list com.artoolapp
2021-03-01 20:55:32.659 1808-1857/com.artoolapp E/Perf: getFolderSize() : Exception_1 = java.lang.NullPointerException: Attempt to get length of null array
2021-03-01 20:55:32.660 1808-1857/com.artoolapp E/Perf: Fail to get file list com.artoolapp
2021-03-01 20:55:32.660 1808-1857/com.artoolapp E/Perf: getFolderSize() : Exception_1 = java.lang.NullPointerException: Attempt to get length of null array
2021-03-01 20:55:34.996 1808-1867/com.artoolapp E/ReactNativeJNI: logMarker CREATE_REACT_CONTEXT_END
2021-03-01 20:55:34.996 1808-1866/com.artoolapp E/unknown:ReactNative: ReactInstanceManager.createReactContext: mJSIModulePackage null
2021-03-01 20:55:34.997 1808-1867/com.artoolapp E/ReactNativeJNI: logMarker RUN_JS_BUNDLE_START
2021-03-01 20:55:35.000 1808-1868/com.artoolapp E/unknown:ReactNative: ReactInstanceManager.attachRootViewToInstance()
2021-03-01 20:55:35.110 1808-1868/com.artoolapp E/unknown:ReactRootView: runApplication: call AppRegistry.runApplication
2021-03-01 20:55:35.191 1808-1808/com.artoolapp E/unknown:ReactNative: ReactInstanceManager.attachRootViewToInstance()
2021-03-01 20:55:35.192 1808-1808/com.artoolapp E/unknown:ReactRootView: runApplication: call AppRegistry.runApplication
2021-03-01 20:55:35.732 1808-1867/com.artoolapp E/ReactNativeJNI: logMarker RUN_JS_BUNDLE_END
```

mrousavy, 2021-03-01:
as I said in https://github.com/cuvent/react-native-vision-camera/issues/9#issuecomment-787913979:
> The errors logs don't seem to be related to vision camera. See if your apps works when you remove the Camera code.

bairock, 2021-03-02:
Crash app

this video shows a crash app

[image]

my code 

```
import React from 'react'
import { useState, useEffect } from 'react'
import { StyleSheet, View, Text, TouchableOpacity } from 'react-native'
import { Camera, useCameraDevices } from 'react-native-vision-camera'

const App = () => {
  const devices = useCameraDevices('wide-angle-camera')
  const device = devices.back
  const [cameraOn, setCameraOn] = useState(false)
  const [permissons, setPermissons] = useState(false)

  const getPermissons = async () => {
    const cameraPermission = await Camera.getCameraPermissionStatus()
    const microphonePermission = await Camera.getMicrophonePermissionStatus()

    if (microphonePermission === 'authorized' && cameraPermission === 'authorized') {
      setPermissons(true)
    }
  }

  useEffect(() => {
    getPermissons()
  }, [])

  if (!permissons) {
    return (
      <View style={{ flex: 1, justifyContent: 'center', alignItems: 'center' }}>
        <Text>No permissons</Text>
      </View>
    )
  }

  if (!device) {
    return (
      <View style={{ flex: 1, justifyContent: 'center', alignItems: 'center' }}>
        <Text>Loading</Text>
      </View>
    )
  }

  if (!cameraOn) {
    return (
      <View style={{ flex: 1, justifyContent: 'center', alignItems: 'center' }}>
        <Text>Turn on the camera</Text>
        <TouchableOpacity onPress={() => setCameraOn(true)} style={{ marginTop: 30, backgroundColor: '#28a745' }}>
          <Text style={{ marginHorizontal: 20, marginVertical: 10, color: 'white' }}>ON</Text>
        </TouchableOpacity>
      </View>
    )
  }

  return (
    <Camera
      style={StyleSheet.absoluteFill}
      device={device}
      isActive={true}
    />
  )
}

export default App

```

mrousavy, 2021-03-02:
Could you try opening your App in Android Studio and debug it? When an error is thrown, you can view which line in the native Kotlin code is causing the crash. Also check Android Studio's Logcat output (<kbd>Cmd</kbd> + <kbd>6</kbd>)

bairock, 2021-03-02:
CameraView.kt 195.
 fun update(changedProps: ArrayList<String>) = GlobalScope.launch(Dispatchers.Main)

[image]

mrousavy, 2021-03-02:
Again, this is not a crash. Please try to learn about Android Studio and debugging. Have you tried to remove the `<Camera>` component in your JS code to find out if it still crashes?

bairock, 2021-03-02:
LOGS

I/CameraManagerGlobal: Connecting to camera service
W/CameraManagerGlobal: [soar.cts] ignore the status update of camera: 2
    [soar.cts] ignore the status update of camera: 3
W/CameraManagerGlobal: ignore the torch status update of camera: 3
D/ExtenderVersion: No versioning extender found. Falling back to default.
I/Process: Sending signal. PID: 23699 SIG: 9

[same problem](https://github.com/flutter/flutter/issues/54619)

zifahm, 2021-03-05:
i can concur with @bairock 
MY app crashed just saying 
```
const Capture = () => {
  const devices = useCameraDevices("wide-angle-camera");
  const device = devices.back;
  const onError = useCallback((error: CameraRuntimeError) => {
    console.error(error);
  }, []);

  if (device == null) {
    return null;
  }

  return (
    <View style={{ flex: 1 }}>
      <Camera
        style={StyleSheet.absoluteFillObject}
        device={device}
        preset="high"
        onError={onError}
      />
    </View>
  );
};
export default Capture;

```

RN:0.64

zifahm, 2021-03-06:
I think my app crashes because its RN 0.64

mrousavy, 2021-03-06:
We're also using RN 0.64 in Cuvent, and the example (on the frame-processor branch) as well.

I'm away for the weekend but will take another look next week! Good luck

mrousavy, 2021-03-06:
Just to confirm, can you guys run the example app?

zifahm, 2021-03-06:
@mrousavy example app works fine on my phone, I even copied all the code from the example app to my app and still it crashes

zifahm, 2021-03-06:
@mrousavy  okay so I found the problem

Seems like the camera only works if you have react native navigation installed 

It doesnt seem to work with a blank react native project. 

I tired a blank react native 0.64.rc-4 with react native vision and the thing failed

Only after installing  react native navigation, Camera seems to work with no problem. 

Any idea how do I fix this?

hannojg, 2021-03-06:
@zifahm Do you have a stack trace for us of the crash?

zifahm, 2021-03-06:
Hmm idk how to do that in android studio, the weird thing is i dont seem to find any crash logs in flipper,

Could you just guide me how would i find the stack trace? @hannojg

hannojg, 2021-03-06:
Start Android studio, open your project (the /android folder) and wait for the gradle sync to complete (important to wait, you can see the in the bottom right corner).Once that’s done open View > Tool Windows > Logcat. Run your application (from Android studio)
In logcat select your device, your App and at the Right side „Show only selected application“.
The output in logcat is the native log from your specific application. 
Now cause the crash with the camera. 
In logcat you should see a red message (and nothing after that as the application has been stopped). This is the stack trace we need.

zifahm, 2021-03-06:
2021-03-06 20:13:18.005 23707-23707/com.rn064 E/SoLoader: Error when loading lib: dlopen failed: library "libjsc.so" not found lib hash: eae0e884bac6a338b8c53618b45e2515 search path is /data/app/com.rn064-oAfb4XOy5uS3nY-08DqAxA==/lib/arm64
2021-03-06 20:13:18.008 23707-23707/com.rn064 E/SoLoader: couldn't find DSO to load: libjscexecutor.so caused by: dlopen failed: library "libjsc.so" not found result: 0

zifahm, 2021-03-06:
@hannojg  okay seems like its  wide enough issue https://github.com/facebook/react-native/issues/25537

mrousavy, 2021-03-06:
Yes that's a pretty common issue and I'm not sure if it is related to VisionCamera. The only thing I thought of is Kotlin, maybe RNN automatically installs Kotlin and VisionCamera doesn't. Could you try with a blank project + install Kotlin (google that) + install VisionCamera? I'll add that to the docs if that's the cause

zifahm, 2021-03-06:
I followed this guide. Still seems to crash @mrousavy 
https://medium.com/mobile-app-development-publication/setup-kotlin-for-android-studio-1bffdf1362e8

zifahm, 2021-03-06:
Last line in versbe dropdown
```
2021-03-06 23:07:14.087 19755-19804/com.rn064 I/cr_LibraryLoader: Loaded native library version number "88.0.4324.181"
2021-03-06 23:07:14.088 19755-19804/com.rn064 I/cr_CachingUmaRecorder: Flushed 4 samples from 4 histograms.
2021-03-06 23:07:14.275 19755-19803/com.rn064 I/ReactNativeJS: Running "RN064" with {"rootTag":1}
2021-03-06 23:07:14.529 19755-19804/com.rn064 I/CameraManagerGlobal: Connecting to camera service
2021-03-06 23:07:14.563 19755-19804/com.rn064 D/ExtenderVersion: No versioning extender found. Falling back to default.
2021-03-06 23:07:14.608 19755-19760/com.rn064 I/zygote64: Compiler allocated 4MB to compile void com.mrousavy.camera.CameraViewModule.getAvailableCameraDevices(com.facebook.react.bridge.Promise)
2021-03-06 23:07:14.665 19755-19755/com.rn064 I/SensorManager: removeAllSensors() [Sensor: LGE Accelerometer] by com.facebook.react.common.ShakeDetector.stop():69

```

mrousavy, 2021-03-07:
Did you try [all the steps in the Troubleshooting guide](https://cuvent.github.io/react-native-vision-camera/docs/troubleshooting)? I updated it earlier today

zifahm, 2021-03-07:
@mrousavy yup. so it works for you in a fresh build without RNN ?

mrousavy, 2021-03-07:
I haven't tried that yet, but I can't think of anything that affects this other than Kotlin being installed. The "libjsc not found" error does not seem to be related, this just often indicates that you have some sort of react native version mismatch. It's just really weird that you're not finding any errors in the logcat output, normally that is spammed with every single detail you can imagine

zifahm, 2021-03-07:
Yeah ikr even flipper does not shill out any logs, damn weird

mrousavy, 2021-03-07:
Flipper shouldn't be your source of debugging for this, Android Logcat picks up more logs than Flipper.

vinkim, 2021-03-09:
Hey! I ran into the same issue when testing out the module in my own application.

Only one of my devices actually printed any meaningful output when crashing. This was logged on a Samsung Galaxy S8+

```
--------- beginning of crash
2021-03-09 12:03:42.381 17026-17026/com.test.app E/AndroidRuntime: FATAL EXCEPTION: main
    Process: com.test.app, PID: 17026
    java.lang.IllegalStateException: Module with the Main dispatcher is missing. Add dependency providing the Main dispatcher, e.g. 'kotlinx-coroutines-android' and ensure it has the same version as 'kotlinx-coroutines-core'
        at kotlinx.coroutines.internal.MainDispatchersKt.throwMissingMainDispatcherException(MainDispatchers.kt:76)
        at kotlinx.coroutines.internal.MissingMainCoroutineDispatcher.missing(MainDispatchers.kt:107)
        at kotlinx.coroutines.internal.MissingMainCoroutineDispatcher.isDispatchNeeded(MainDispatchers.kt:91)
        at kotlinx.coroutines.internal.DispatchedContinuationKt.resumeCancellableWith(DispatchedContinuation.kt:302)
        at kotlinx.coroutines.intrinsics.CancellableKt.startCoroutineCancellable(Cancellable.kt:30)
        at kotlinx.coroutines.intrinsics.CancellableKt.startCoroutineCancellable$default(Cancellable.kt:27)
        at kotlinx.coroutines.CoroutineStart.invoke(CoroutineStart.kt:109)
        at kotlinx.coroutines.AbstractCoroutine.start(AbstractCoroutine.kt:158)
        at kotlinx.coroutines.BuildersKt__Builders_commonKt.launch(Builders.common.kt:56)
        at kotlinx.coroutines.BuildersKt.launch(Unknown Source:1)
        at kotlinx.coroutines.BuildersKt__Builders_commonKt.launch$default(Builders.common.kt:49)
        at kotlinx.coroutines.BuildersKt.launch$default(Unknown Source:1)
        at com.mrousavy.camera.CameraView.update(CameraView.kt:195)
        at com.mrousavy.camera.CameraViewManager.onAfterUpdateTransaction(CameraViewManager.kt:124)
        at com.mrousavy.camera.CameraViewManager.onAfterUpdateTransaction(CameraViewManager.kt:11)
        at com.facebook.react.uimanager.ViewManager.updateProperties(ViewManager.java:51)
        at com.facebook.react.uimanager.NativeViewHierarchyManager.createView(NativeViewHierarchyManager.java:270)
        at com.facebook.react.uimanager.UIViewOperationQueue$CreateViewOperation.execute(UIViewOperationQueue.java:186)
        at com.facebook.react.uimanager.UIViewOperationQueue$DispatchUIFrameCallback.dispatchPendingNonBatchedOperations(UIViewOperationQueue.java:1112)
        at com.facebook.react.uimanager.UIViewOperationQueue$DispatchUIFrameCallback.doFrameGuarded(UIViewOperationQueue.java:1083)
        at com.facebook.react.uimanager.GuardedFrameCallback.doFrame(GuardedFrameCallback.java:29)
        at com.facebook.react.modules.core.ReactChoreographer$ReactChoreographerDispatcher.doFrame(ReactChoreographer.java:175)
        at com.facebook.react.modules.core.ChoreographerCompat$FrameCallback$1.doFrame(ChoreographerCompat.java:85)
        at android.view.Choreographer$CallbackRecord.run(Choreographer.java:947)
        at android.view.Choreographer.doCallbacks(Choreographer.java:761)
        at android.view.Choreographer.doFrame(Choreographer.java:693)
        at android.view.Choreographer$FrameDisplayEventReceiver.run(Choreographer.java:935)
        at android.os.Handler.handleCallback(Handler.java:873)
        at android.os.Handler.dispatchMessage(Handler.java:99)
        at android.os.Looper.loop(Looper.java:214)
        at android.app.ActivityThread.main(ActivityThread.java:7050)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:494)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:965)
```

The only difference I could see between the example app and my app, is that I use **react-navigation** instead of **react-native-navigation**

Looking into the **react-native-navigation** module I see that it has the following dependencies:

```
implementation "org.jetbrains.kotlinx:kotlinx-coroutines-core:$kotlinCoroutinesCore"
implementation "org.jetbrains.kotlinx:kotlinx-coroutines-android:$kotlinCoroutinesCore"

where kotlinCoroutinesCore = 1.3.7

```

Also looking up the error message brought me here: https://stackoverflow.com/questions/52677638/module-with-main-dispatcher-is-missing which has some more information.

My current workaround for now is to add the following to my applications build.gradle

```
// Workaround for react-native-vision-camera
implementation "org.jetbrains.kotlinx:kotlinx-coroutines-android:1.4.2"
```

I Hhaven't dug much further or read too much into this, but maybe replacing
implementation "org.jetbrains.kotlinx:kotlinx-coroutines-android:1.4.2"

mrousavy, 2021-03-09:
@vinkim thank you for the details!! Helps a lot. I've just published 1.0.1 on npm with coroutines-android instead of core, let me know if that works for you.

closed 2021-03-09 by mrousavy

mrousavy, 2021-03-09:
Awesome. I wasn't sure if kotlinx-coroutines-core was included in the kotlinx-coroutines-android package, but if it works for you I assume it is. 🚀 

Thanks again for your awesome research, took me about 20 seconds to see what's going on 👍

guillermoGallo, 2022-04-21:
Hi. There's an issue when trying to load the app using expo start:

[Unhandled promise rejection: TypeError: null is not an object (evaluating 'CameraModule.getCameraPermissionStatus')]

ashrazdev, 2022-11-30:
useCameraDevices is return undefined on release android is there any solution for this?

commit 4639a791 (2021-03-09): Android: Use kotlinx-coroutines-android instead of core (fixes #9)