margelo/react-native-vision-camera #13: 🐛 Android Build Issue

McFly78, 2021-03-03:
Hi,
Thanks very much for this new library

I am experiencing a build issue on Android just after having installed react-native-vision-camera. 
IOS build is working fine.

I am getting the following error message when building with "npx react-native run-android" command.

> Configure project :react-native-vision-camera
Warning: The 'kotlin-android-extensions' Gradle plugin is deprecated. Please use this migration guide (https://goo.gle/kotlin-android-extensions-deprecation) to start working with View Binding (https://developer.android.com/topic/libraries/view-binding) and the 'kotlin-parcelize' plugin.

Deprecated Gradle features were used in this build, making it incompatible with Gradle 7.0.
Use '--warning-mode all' to show the individual deprecation warnings.
See https://docs.gradle.org/6.2/userguide/command_line_interface.html#sec:command_line_warnings

FAILURE: Build completed with 2 failures.

1: Task failed with an exception.
-----------
* Where:
Build file '/Users/nm/Dev/MyApp/node_modules/react-native-vision-camera/android/build.gradle' line: 26

* What went wrong:
A problem occurred evaluating project ':react-native-vision-camera'.
> Failed to apply plugin [id 'org.jlleitschuh.gradle.ktlint']
   > Extension of type 'CommonExtension' does not exist. Currently registered extension types: [ExtraPropertiesExtension, DefaultArtifactPublicationSet, ReportingExtension, SourceSetContainer, JavaPluginExtension, JavaInstallationRegistry, NamedDomainObjectContainer<BaseVariantOutput>, LibraryExtension, KotlinAndroidProjectExtension, KotlinTestsRegistry, AndroidExtensionsExtension, KtlintExtension]

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.
==============================================================================

2: Task failed with an exception.
-----------
* What went wrong:
A problem occurred configuring project ':react-native-vision-camera'.
> compileSdkVersion is not specified.

When opening the project on Android Studio I get the following message:
11:14 AM	Gradle sync failed: Could not run phased build action using Gradle distribution 'https://services.gradle.org/distributions/gradle-6.2-all.zip'.
Consult IDE log for more details (Help | Show Log) (1 s 252 ms)

For the first issue I found the following link if it may help:
https://stackoverflow.com/questions/65179275/the-kotlin-android-extensions-gradle-plugin-is-deprecated

## Environment

React Native: 0.63.4
Android Studio: 4.0
Gradle Plugin: 3.5.3
Gradle: 6.2

* Platform(s):
  * [X] Android

* react-native version: 0.63.4
* react-native-vision-camera version: 1.0.0
* react-native-reanimated version: 1.13.1

Many thanks for your help.

mrousavy, 2021-03-04:
Hm, it looks like ktlint is causing problems for you. Could you try removing it from the `build.gradle` and see if that makes a difference?

McFly78, 2021-03-04:
Thanks very much for your quick answer.
For your information I didn't remove for the moment Ktlint from the react-native-vision-camera build.gradle file .

**What I did**:
I updated **Gradle** version **from 6.2 to 6.5** on my project Gradle-wrapper.properties file:
distributionUrl=https\://services.gradle.org/distributions/gradle-6.5-all.zip

I also **updated Gradle plugin** version from **3.5.3 to 4.1.2** on my project  build.gradle file:
classpath("com.android.tools.build:gradle:4.1.2")

After these modification Ktlint error during building disappeared but I had a new error message telling me to update minSdkVersion from 16 to 21 (minSdkVersion 21 is used by react-native-vision-camera):
minSdkVersion = 21

**After updating minSdkVersion, I have now these warnings and error messages concerning react-native-vision-camera:**
> Configure project :react-native-vision-camera
Warning: The 'kotlin-android-extensions' Gradle plugin is deprecated. Please use this migration guide (https://goo.gle/kotlin-android-extensions-deprecation) to start working with View Binding (https://developer.android.com/topic/libraries/view-binding) and the 'kotlin-parcelize' plugin.

e: /Users/nm/Dev/MyApp/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraViewModule.kt: (152, 74): Unresolved reference: R
e: /Users/nm/Dev/MyApp/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraViewModule.kt: (153, 53): Unresolved reference: CONTROL_ZOOM_RATIO_RANGE
e: /Users/nm/Dev/MyApp/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraViewModule.kt: (188, 46): Unresolved reference: lower
e: /Users/nm/Dev/MyApp/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraViewModule.kt: (189, 46): Unresolved reference: upper
e: /Users/nm/Dev/MyApp/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraViewModule.kt: (243, 51): Unresolved reference: upper
e: /Users/nm/Dev/MyApp/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/utils/CameraCharacteristicsUtils.kt: (73, 92): Unresolved reference: R
e: /Users/nm/Dev/MyApp/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/utils/CameraCharacteristicsUtils.kt: (74, 38): Unresolved reference: CONTROL_ZOOM_RATIO_RANGE
e: /Users/nm/Dev/MyApp/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/utils/CameraCharacteristicsUtils.kt: (77, 26): Unresolved reference: lower
e: /Users/nm/Dev/MyApp/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/utils/CameraCharacteristicsUtils.kt: (77, 46): Unresolved reference: upper
e: /Users/nm/Dev/MyApp/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/utils/CameraCharacteristicsUtils.kt: (77, 64): Unresolved reference: lower

FAILURE: Build failed with an exception.

* What went wrong:
**Execution failed for task ':react-native-vision-camera:compileDebugKotlin'.**

I then also **upgraded Android Studio** **from 4.0 to 4.1.2** but I had the same buildings errors.

Looking into the App example code in example folder I see that you have the following lines in the project build.gradle.
RNNKotlinVersion = "1.3.61"
classpath "org.jetbrains.kotlin:kotlin-gradle-plugin:1.4.30"

I dont't know if I need them but I tried to add them in my build.gradle project file but I am still experiencing the same build errors.

**Here is my updated build.gradle**:
buildscript {
    ext {
        **RNNKotlinVersion = "1.3.61"**
        buildToolsVersion = "29.0.2"
        minSdkVersion = **21**
        compileSdkVersion = 29
        targetSdkVersion = 29
        googlePlayServicesVersion = "17.0.0"
    }
    repositories {
        google()
        jcenter()
    }
    dependencies {
        classpath("com.android.tools.build:gradle:**4.1.2**")
        **classpath "org.jetbrains.kotlin:kotlin-gradle-plugin:1.4.30"**
        // NOTE: Do not place your application dependencies here; they belong
        // in the individual module build.gradle files
    }
}
allprojects {
    repositories {
        mavenLocal()
        maven {
            // All of React Native (JS, Obj-C sources, Android binaries) is installed from npm
            url("$rootDir/../node_modules/react-native/android")
        }
        maven {
            // Android JSC is installed from npm
            url("$rootDir/../node_modules/jsc-android/dist")
        }
        google()
        jcenter()
        maven { url 'https://www.jitpack.io' }
    }
}

**And my updated gradle.wrapper.properties**:
distributionBase=GRADLE_USER_HOME
distributionPath=wrapper/dists
distributionUrl=https\://services.gradle.org/distributions/gradle-**6.5**-all.zip
zipStoreBase=GRADLE_USER_HOME
zipStorePath=wrapper/dists

I don't know what I am missing !

zifahm, 2021-03-05:
Getting the same problem

```
BUILD FAILED in 37s

   39 actionable tasks: 3 executed, 536 up-to-date
e: /home/afi/Documents/Project/Whale/whale/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraViewModule.kt: (152, 74): Unresolved reference: R
e: /home/afi/Documents/Project/Whale/whale/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraViewModule.kt: (153, 53): Unresolved reference: CONTROL_ZOOM_RATIO_RANGE
e: /home/afi/Documents/Project/Whale/whale/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraViewModule.kt: (188, 46): Unresolved reference: lower
e: /home/afi/Documents/Project/Whale/whale/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraViewModule.kt: (189, 46): Unresolved reference: upper
e: /home/afi/Documents/Project/Whale/whale/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraViewModule.kt: (243, 51): Unresolved reference: upper
e: /home/afi/Documents/Project/Whale/whale/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/utils/CameraCharacteristicsUtils.kt: (73, 92): Unresolved reference: R
e: /home/afi/Documents/Project/Whale/whale/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/utils/CameraCharacteristicsUtils.kt: (74, 38): Unresolved reference: CONTROL_ZOOM_RATIO_RANGE
e: /home/afi/Documents/Project/Whale/whale/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/utils/CameraCharacteristicsUtils.kt: (77, 26): Unresolved reference: lower
e: /home/afi/Documents/Project/Whale/whale/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/utils/CameraCharacteristicsUtils.kt: (77, 46): Unresolved reference: upper
e: /home/afi/Documents/Project/Whale/whale/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/utils/CameraCharacteristicsUtils.kt: (77, 64): Unresolved reference: lower
```

McFly78, 2021-03-05:
Finally I found a way to build !!!! :)

I needed to **update SDK version from 29 to 30** in my project build.gradle:

Now my **build.gradle** looks like:
uildscript {
    ext {
        buildToolsVersion = "**30.0.0**"
        minSdkVersion = **21**
        compileSdkVersion = **30**
        targetSdkVersion = **30**
        googlePlayServicesVersion = "17.0.0"
    }
    repositories {
        google()
        jcenter()
    }
    dependencies {
        classpath("com.android.tools.build:gradle:**4.1.2**")
        // NOTE: Do not place your application dependencies here; they belong
        // in the individual module build.gradle files
    }
}
allprojects {
    repositories {
        mavenLocal()
        maven {
            // All of React Native (JS, Obj-C sources, Android binaries) is installed from npm
            url("$rootDir/../node_modules/react-native/android")
        }
        maven {
            // Android JSC is installed from npm
            url("$rootDir/../node_modules/jsc-android/dist")
        }
        google()
        jcenter()
        maven { url 'https://www.jitpack.io' }
    }
}

And my **gradle.wrapper.properties**:
distributionBase=GRADLE_USER_HOME
distributionPath=wrapper/dists
distributionUrl=https://services.gradle.org/distributions/gradle-6.5-all.zip
zipStoreBase=GRADLE_USER_HOME
zipStorePath=wrapper/dists

mrousavy, 2021-03-05:
Oh, that should be a requirement. I've added that to the documentation, thanks for mentioning!

closed 2021-03-05 by mrousavy

McFly78, 2021-03-05:
Thanks !
I think it may be good also to add in the troubleshooting page that it is also needed to update the following Gradle versions to avoid ktlint build issue (same versions as in your app example  :

- update Gradle version to 6.5 on Gradle-wrapper.properties file:
distributionUrl=https://services.gradle.org/distributions/gradle-6.5-all.zip

- update Gradle plugin version to 4.1.2 on build.gradle file:
classpath("com.android.tools.build:gradle:4.1.2")

Also minSdkVersion needs to be 21 "in build.gradle file in order to be able to build (we need at least the same minSdkVersion as react-native-vision-camera module):
minSdkVersion = 21