margelo/react-native-vision-camera #17: Is it possible to use it with react-native-webrtc❓

jsellam, 2021-03-04:
Hi,
Thank you for this library, it's a very good job ! 

I'm working on video chat app with webrtc with react-native-webrtc.
In your documentation you say that we can use Frame Processor to build realtime video chat app. Have you more informations about it ?

react-native-webrc use it own "getUserMedia" implementation. Do you think it's possible to use your library too with webrtc ? and how ?

thx
Jeremie.

mrousavy, 2021-03-04:
Hi!

Yes, it _will_ be possible to use VisionCamera for video chat applications. However I am still working on that part ("Frame Processors", see PR: https://github.com/cuvent/react-native-vision-camera/pull/2), so it is not ready yet.

Once frame processors are ready, they are like a sandbox. You can use plugins for your specific task (send frame over the network). So this means, once frame processors are released you will have to use a WebRTC plugin specifically designed for VisionCamera. Either you write that yourself, or react-native-webrtc does that.

The plugin will have to be written with JSI (C++), but I have a lot of examples on how to do that, and honestly it's not that difficult. I will share more details and examples once Frame Processors are available.

mrousavy, 2021-05-06:
I just released [**2.0.0**](https://github.com/cuvent/react-native-vision-camera/releases/tag/v2.0.0), which finally includes Frame Processors.

This means you can create a WebRTC Frame Processor Plugin and start using WebRTC today. 🎉

> Note: Frame Processors are iOS only for now, I will do Android in a separate PR really soon. 💪

closed 2021-05-06 by mrousavy