margelo/react-native-vision-camera #22: A Dumb Question❓

zifahm, 2021-03-06:
How does the Camera api work in the example app if it does not have "react-native-vision-camera" in its package.json

Just to confirm I even checked inside node_modules and yarn.lock but nope, its not installed in the example app.

Maybe thats why I'm unable to run the Camera api?

zifahm, 2021-03-06:
Okay just saw this in setting.gradel 
```
include ':camera'
project(':camera').projectDir = new File(rootProject.projectDir, '../../android')

```

closed 2021-03-06 by zifahm

mrousavy, 2021-03-06:
Hi, the camera package is not installed because it's the same project. You still need to install it on your end. Can you build it now? Did the settings.gradle line fix it?

zifahm, 2021-03-06:
I tried copying over the android project and doing what the example project did, and still no luck :(