margelo/react-native-vision-camera #76: ✨ Android: Implement custom FPS

mrousavy, 2021-03-17:
Setting custom FPS above 30 is not really supported on Android. I can't seem to get a "format" with FPS higher than 30, even though my system camera can record in 60.

I feel like something in the `getAvailableCameraDevices` code is wrong:

https://github.com/cuvent/react-native-vision-camera/blob/dff8fc71788311c324e6eb80976a6ec50596d3e5/android/src/main/java/com/mrousavy/camera/CameraViewModule.kt#L205-L220

Maybe I shouldn't use `MediaRecorder::class.java` here?:

https://github.com/cuvent/react-native-vision-camera/blob/dff8fc71788311c324e6eb80976a6ec50596d3e5/android/src/main/java/com/mrousavy/camera/CameraViewModule.kt#L201

Here's how I set FPS (that works if I set it to e.g 15)

https://github.com/cuvent/react-native-vision-camera/blob/d85126d883feffcd6189135042420a8d89b153cf/android/src/main/java/com/mrousavy/camera/CameraView.kt#L280-L292

McFly78, 2021-03-23:
Hello,

For your issue "Resolution doesn't seem to correctly apply to Preview"
I confirm you that the Preview Resolution is not good on Android.
The image displayed on the screen is zoomed.

On IOS everything is fine !

Alex1899, 2022-02-24:
having this issue

nica0012, 2022-04-12:
@Alex1899 the zooming is happening for me too on IOS, only for 30FPS (photos and videos). 60FPS the image/video output is exactly as expected (1:1 / 4:3). Been trying to find a solution for 2 days...

mrousavy, 2023-08-18:
Hey! I've rewritten the entire Android codebase of VisionCamera from CameraX to Camera2 in the efforts of [✨ VisionCamera V3](https://github.com/mrousavy/react-native-vision-camera/issues/1376).

I just now completed the Camera2 rewrite and I believe the core structure is running, but there might be some edge cases to iron out.

Multi-cameras should now work! 🎉 

Here's an instruction on how we can test that: https://github.com/mrousavy/react-native-vision-camera/pull/1674#issuecomment-1684104217

mrousavy, 2023-09-01:
some Samsungs say they have 60 FPS but when you try to use 60 FPS it just shows a blackscreen. again, samsung making it really painful.

closed 2023-09-01 by mrousavy