margelo/react-native-vision-camera #77: 🐛 Android: Video doesn't have duration set

mrousavy, 2021-03-17:
On Android the [video duration](https://cuvent.github.io/react-native-vision-camera/docs/api/interfaces/videofile.videofile-1#duration) and [size](https://cuvent.github.io/react-native-vision-camera/docs/api/interfaces/videofile.videofile-1#size) are not set correctly because CameraX doesn't provide that information in the Callback.

https://github.com/cuvent/react-native-vision-camera/blob/dff8fc71788311c324e6eb80976a6ec50596d3e5/android/src/main/java/com/mrousavy/camera/CameraView%2BRecordVideo.kt#L32-L36

Hirbod, 2022-01-13:
@mrousavy would love to have feature parity here. Missing the duration for android.

mrousavy, 2022-01-14:
Did you check if it was there?

Hirbod, 2022-01-14:
Nope, I saw this ticket and I knew it wasn't there couple of weeks ago.

Hirbod, 2022-01-14:
Yes, it is there... `yarn remove react-native-ffmpeg-kit`
:D