margelo/react-native-vision-camera #78: ❓ Possible to not request Microphone Permission to the user in photo mode usage ?

McFly78, 2021-03-17:
Hello,

I would like to know if there is a way that microphone permission is not requested to the user when we just want to take photo and not any movie.

I have tried to use the following camera props but the microphone access is still requesting to the users:
preset= "photo"

Many thanks for your help

mrousavy, 2021-03-17:
Hi!

Sorry, that is currently not supported. The Camera is designed to be used with both Photo and Video capture always being ready.

I will convert this to a feature request and pick this up in the future.

filipef101, 2021-04-12:
It should be possible to capture video without audio permission, so this could be for example a prop like audio={false} like react native camera does

mrousavy, 2021-04-12:
> It should be possible to capture video without audio permission

Interesting to see demand for this.

> a prop like audio={false} like react native camera does

I'm not going to do it this way, since that's going to be a bit ugly when there are multiple props (no video, video without audio, no photo, no frame processors, etc). Instead I'm thinking about creating a config;

```tsx
return <Camera useCases={['photo', 'video-with-audio', 'frame-processor']} />
```

Only thing annoying me is memoization in this case. It's ugly to use `useMemo` here, and it's ugly to prevent those rerenders by unwrapping the array in the Camera component. I'll try to think of something fancy & easy to use.

filipef101, 2021-04-12:
I mean that prop defaulting to true, so what could happen is 

`return <Camera audio={false} photo={false} video={false} frameProcessor={false} />`

mrousavy, 2021-04-12:
> I mean that prop defaulting to true

Exactly, that's something I don't really like. Props that default to true are added confusion, especially in this case since those are booleans and don't have a `enable`, `allow` or `is` prefix.

filipef101, 2021-04-12:
Ah okay, yeah I agree makes more sense to be a prop `useCases`, array or an object, also might be useful to have video without audio?

G0retZ, 2021-06-03:
I'm also interested in this.
Want to build QR code scanner and it should not ask user for microphone permissions.
What should I patch to make this work until the feature will arrive? 😁

mrousavy, 2021-06-03:
With the newest version (2.1.0) you can just remove the `activateAudioSession` and `deactivateAudioSession` calls. I will implement an option to disable audio very soon though (waiting for CameraX to support it)

G0retZ, 2021-06-03:
Just in case if anyone needs this here is a patch to turn off audio related stuff

```
diff --git a/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraView.kt b/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraView.kt
index 4c26b34..bfe99a5 100644
--- a/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraView.kt
+++ b/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraView.kt
@@ -218,9 +218,6 @@ class CameraView(context: Context) : FrameLayout(context), LifecycleOwner {
       val startTime = System.currentTimeMillis()
       Log.i(TAG, "Configuring session...")
       if (ContextCompat.checkSelfPermission(context, Manifest.permission.CAMERA) != PackageManager.PERMISSION_GRANTED) {
-        throw MicrophonePermissionError()
-      }
-      if (ContextCompat.checkSelfPermission(context, Manifest.permission.RECORD_AUDIO) != PackageManager.PERMISSION_GRANTED) {
         throw CameraPermissionError()
       }
       if (cameraId == null) {
diff --git a/node_modules/react-native-vision-camera/ios/CameraView+AVAudioSession.swift b/node_modules/react-native-vision-camera/ios/CameraView+AVAudioSession.swift
index eff11ec..a5f09d4 100644
--- a/node_modules/react-native-vision-camera/ios/CameraView+AVAudioSession.swift
+++ b/node_modules/react-native-vision-camera/ios/CameraView+AVAudioSession.swift
@@ -59,38 +59,6 @@ extension CameraView {
     audioCaptureSession.addOutput(audioOutput!)
   }
 
-  /**
-   Configures the Audio session and activates it. If the session was active it will shortly be deactivated before configuration.
-
-   The Audio Session will be configured to allow background music, haptics (vibrations) and system sound playback while recording.
-   Background audio is allowed to play on speakers or bluetooth speakers.
-   */
-  final func activateAudioSession() {
-    ReactLogger.log(level: .info, message: "Activating Audio Session...")
-
-    do {
-      try AVAudioSession.sharedInstance().updateCategory(AVAudioSession.Category.playAndRecord,
-                                                         options: [.mixWithOthers,
-                                                                   .allowBluetoothA2DP,
-                                                                   .defaultToSpeaker,
-                                                                   .allowAirPlay])
-      audioCaptureSession.startRunning()
-    } catch let error as NSError {
-      switch error.code {
-      case 561_017_449:
-        self.invokeOnError(.session(.audioInUseByOtherApp), cause: error)
-      default:
-        self.invokeOnError(.session(.audioSessionFailedToActivate), cause: error)
-      }
-    }
-  }
-
-  final func deactivateAudioSession() {
-    ReactLogger.log(level: .info, message: "Deactivating Audio Session...")
-
-    audioCaptureSession.stopRunning()
-  }
-
   @objc
   func audioSessionInterrupted(notification: Notification) {
     ReactLogger.log(level: .error, message: "Audio Session Interruption Notification!")
@@ -112,8 +80,6 @@ extension CameraView {
         if isRecording {
           audioQueue.async {
             ReactLogger.log(level: .info, message: "Resuming interrupted Audio Session...", alsoLogToJS: true)
-            // restart audio session because interruption is over
-            self.activateAudioSession()
           }
         }
       } else {
diff --git a/node_modules/react-native-vision-camera/ios/CameraView+RecordVideo.swift b/node_modules/react-native-vision-camera/ios/CameraView+RecordVideo.swift
index 8a958d2..6c43d2a 100644
--- a/node_modules/react-native-vision-camera/ios/CameraView+RecordVideo.swift
+++ b/node_modules/react-native-vision-camera/ios/CameraView+RecordVideo.swift
@@ -45,9 +45,6 @@ extension CameraView: AVCaptureVideoDataOutputSampleBufferDelegate, AVCaptureAud
         let onFinish = { (status: AVAssetWriter.Status, error: Error?) -> Void in
           defer {
             self.recordingSession = nil
-            self.audioQueue.async {
-              self.deactivateAudioSession()
-            }
           }
           ReactLogger.log(level: .info, message: "RecordingSession finished with status \(status.descriptor).")
           if let error = error {
@@ -81,7 +78,6 @@ extension CameraView: AVCaptureVideoDataOutputSampleBufferDelegate, AVCaptureAud
         // Init Audio (optional, async)
         self.audioQueue.async {
           // Activate Audio Session (blocking)
-          self.activateAudioSession()
           guard let recordingSession = self.recordingSession else {
             // recording has already been cancelled
             return

```

mrousavy, 2021-06-03:
doesn't android crash when it doesn't have microphone permission in a video recording?

G0retZ, 2021-06-04:
> doesn't android crash when it doesn't have microphone permission in a video recording?

Sorry, I didn't test that.
It's a temporary workaround for those who don't need audio (and thus video) recording...

dcangulo, 2021-06-04:
Just added difference highlighting. Thanks, @G0retZ for this.
> Just in case if anyone needs this here is a patch to turn off the audio related errors and stuff
> 
> ```diff
> diff --git a/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraView.kt b/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraView.kt
> index 4c26b34..bfe99a5 100644
> --- a/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraView.kt
> +++ b/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraView.kt
> @@ -218,9 +218,6 @@ class CameraView(context: Context) : FrameLayout(context), LifecycleOwner {
>        val startTime = System.currentTimeMillis()
>        Log.i(TAG, "Configuring session...")
>        if (ContextCompat.checkSelfPermission(context, Manifest.permission.CAMERA) != PackageManager.PERMISSION_GRANTED) {
> -        throw MicrophonePermissionError()
> -      }
> -      if (ContextCompat.checkSelfPermission(context, Manifest.permission.RECORD_AUDIO) != PackageManager.PERMISSION_GRANTED) {
>          throw CameraPermissionError()
>        }
>        if (cameraId == null) {
> diff --git a/node_modules/react-native-vision-camera/ios/CameraView+AVAudioSession.swift b/node_modules/react-native-vision-camera/ios/CameraView+AVAudioSession.swift
> index eff11ec..a5f09d4 100644
> --- a/node_modules/react-native-vision-camera/ios/CameraView+AVAudioSession.swift
> +++ b/node_modules/react-native-vision-camera/ios/CameraView+AVAudioSession.swift
> @@ -59,38 +59,6 @@ extension CameraView {
>      audioCaptureSession.addOutput(audioOutput!)
>    }
>  
> -  /**
> -   Configures the Audio session and activates it. If the session was active it will shortly be deactivated before configuration.
> -
> -   The Audio Session will be configured to allow background music, haptics (vibrations) and system sound playback while recording.
> -   Background audio is allowed to play on speakers or bluetooth speakers.
> -   */
> -  final func activateAudioSession() {
> -    ReactLogger.log(level: .info, message: "Activating Audio Session...")
> -
> -    do {
> -      try AVAudioSession.sharedInstance().updateCategory(AVAudioSession.Category.playAndRecord,
> -                                                         options: [.mixWithOthers,
> -                                                                   .allowBluetoothA2DP,
> -                                                                   .defaultToSpeaker,
> -                                                                   .allowAirPlay])
> -      audioCaptureSession.startRunning()
> -    } catch let error as NSError {
> -      switch error.code {
> -      case 561_017_449:
> -        self.invokeOnError(.session(.audioInUseByOtherApp), cause: error)
> -      default:
> -        self.invokeOnError(.session(.audioSessionFailedToActivate), cause: error)
> -      }
> -    }
> -  }
> -
> -  final func deactivateAudioSession() {
> -    ReactLogger.log(level: .info, message: "Deactivating Audio Session...")
> -
> -    audioCaptureSession.stopRunning()
> -  }
> -
>    @objc
>    func audioSessionInterrupted(notification: Notification) {
>      ReactLogger.log(level: .error, message: "Audio Session Interruption Notification!")
> @@ -112,8 +80,6 @@ extension CameraView {
>          if isRecording {
>            audioQueue.async {
>              ReactLogger.log(level: .info, message: "Resuming interrupted Audio Session...", alsoLogToJS: true)
> -            // restart audio session because interruption is over
> -            self.activateAudioSession()
>            }
>          }
>        } else {
> diff --git a/node_modules/react-native-vision-camera/ios/CameraView+RecordVideo.swift b/node_modules/react-native-vision-camera/ios/CameraView+RecordVideo.swift
> index 8a958d2..6c43d2a 100644
> --- a/node_modules/react-native-vision-camera/ios/CameraView+RecordVideo.swift
> +++ b/node_modules/react-native-vision-camera/ios/CameraView+RecordVideo.swift
> @@ -45,9 +45,6 @@ extension CameraView: AVCaptureVideoDataOutputSampleBufferDelegate, AVCaptureAud
>          let onFinish = { (status: AVAssetWriter.Status, error: Error?) -> Void in
>            defer {
>              self.recordingSession = nil
> -            self.audioQueue.async {
> -              self.deactivateAudioSession()
> -            }
>            }
>            ReactLogger.log(level: .info, message: "RecordingSession finished with status \(status.descriptor).")
>            if let error = error {
> @@ -81,7 +78,6 @@ extension CameraView: AVCaptureVideoDataOutputSampleBufferDelegate, AVCaptureAud
>          // Init Audio (optional, async)
>          self.audioQueue.async {
>            // Activate Audio Session (blocking)
> -          self.activateAudioSession()
>            guard let recordingSession = self.recordingSession else {
>              // recording has already been cancelled
>              return
> ```

closed 2021-06-07 by mrousavy

mrousavy, 2021-06-07:
I've implemented this in #168. Try upgrading to [**2.2.0**](https://github.com/cuvent/react-native-vision-camera/releases/tag/v2.2.0) and make sure to check out [the docs for the `supportsPhotoAndVideoCapture` prop](https://cuvent.github.io/react-native-vision-camera/docs/guides/devices#the-supportsphotoandvideocapture-prop) - let me know if it works for you!

G0retZ, 2021-06-07:
Great! Thank you!

PR #168 (2021-06-07): feat: Separate usecases (decouple microphone, video, photo)
Adds

* `photo`
* `video`
* `audio`

props to configure what use-cases to attach. 

* Allows recording videos without audio
* Allows using the camera without microphone permission
* Allows faster camera startup by specifying what use-cases to attach

By default, they're all `false`.

## Side-effects

* (Android) Fixes PreviewView having a way too high resolution
* (Android) Fixes wrong orientation when not using a `format`
* (Android) Make `startRecording` non-blocking
* (Android) Reject with errors instead of crashing the app in `sta [... cut]