margelo/react-native-vision-camera #81: ❓ Is there a way to have information concerning the automatically choosen format ?

McFly78, 2021-03-18:
Hi,
If we don't want to specify a format the Camera automatically chooses the best matching format for the current camera device.

I would like to know if there is a way to know which format has been automatically choosen in order to have all information concerning this format (field of view, ...).

Many thanks !

mrousavy, 2021-03-18:
Hi! There is no way to find out the selected format right now, but I could add a callback/event that gets called once the Format gets chosen. For example:

```tsx
const onFormatChanged = useCallback((format: CameraFormat) => {
  console.log(`Format selected: ${JSON.stringify(format)}`);
}, []);

return <Camera onFormatChanged={onFormatChanged} {...cameraProps} />
```

This might also be a good time to add more events, such as `onWillInitialize`, `onDeviceChanged`, ..

I'll convert this to a feature request and pick this up soon. 👍

mrousavy, 2021-03-29:
A somewhat related issue: https://github.com/cuvent/react-native-vision-camera/issues/114

mrousavy, 2021-03-29:
I started working on this here: https://github.com/cuvent/react-native-vision-camera/pull/84