margelo/react-native-vision-camera #105: 🐛  The 'kotlin-android-extensions' Gradle plugin is deprecated.

grkemtneri, 2021-03-25:
## Type

* [x] I'm experiencing a full app crash
* [ ] A feature is not working correctly

## What
 I get this error when I get build after installing.

```
 ext {
        buildToolsVersion = "30.0.0"
        minSdkVersion = 21
        compileSdkVersion = 30
        targetSdkVersion = 30
 }

 dependencies {
        classpath("com.android.tools.build:gradle:4.1.2")
        classpath "org.jetbrains.kotlin:kotlin-gradle-plugin:1.4.30"
  }

distributionUrl=https\://services.gradle.org/distributions/gradle-6.5-all.zip

```

## Error

`` Configure project :react-native-vision-camera
Warning: The 'kotlin-android-extensions' Gradle plugin is deprecated. Please use this migration guide (https://goo.gle/kotlin-android-extensions-deprecation) to start working with View Binding (https://developer.android.com/topic/libraries/view-binding) and the 'kotlin-parcelize' plugin.`
``

* Platform(s):
  * [ ] iOS
  * [x] Android
* react-native version: 0.64
* react-native-vision-camera version: 1.0.4
* react-native-reanimated version: N/A

mrousavy, 2021-03-25:
That's a warning, not an error. You must be getting another error somewhere, could you share the full logs?

grkemtneri, 2021-03-25:
sorry I missed it. I got a Java heap space error. I solved the problem and have successfully built now. thank you for your quick reply.

closed 2021-03-25 by grkemtneri

Bamorem, 2021-12-29:
Hi, I have the same bug, but on my side it's not just a warning (The Kotlin plugin was loaded in the following projects: ':react-native-pager-view', ':react-native-vision-camera'), see the log:

> Configure project :react-native-vision-camera
Warning: The 'kotlin-android-extensions' Gradle plugin is deprecated. Please use this migration guide (https://goo.gle/kotlin-android-extensions-deprecation) to start working with View Binding (https://developer.android.com/topic/libraries/view-binding) and the 'kotlin-parcelize' plugin.

The Kotlin Gradle plugin was loaded multiple times in different subprojects, which is not supported and may break the build. 
This might happen in subprojects that apply the Kotlin plugins with the Gradle 'plugins { ... }' DSL if they specify explicit versions, even if the versions are equal.
Please add the Kotlin plugin to the common parent project or the root project, then remove the versions in the subprojects.
If the parent project does not need the plugin, add 'apply false' to the plugin line.
See: https://docs.gradle.org/current/userguide/plugins.html#sec:subprojects_plugins_dsl
The Kotlin plugin was loaded in the following projects: ':react-native-pager-view', ':react-native-vision-camera'

> Task :app:checkDebugAarMetadata FAILED

mrousavy, 2021-12-30:
This is still a warning, read the actual error message.

> A failure occurred while executing com.android.build.gradle.internal.tasks.CheckAarMetadataWorkAction
The minCompileSdk (31) specified in a
dependency's AAR metadata (META-INF/com/android/build/gradle/aar-metadata.properties)
is greater than this module's compileSdkVersion (android-30).
Dependency: androidx.camera:camera-view:1.0.0-alpha29.

thisroot, 2022-01-06:
Have same error

lrxiang222, 2022-03-06:
Have same error

lrxiang222, 2022-03-06:
> This is still a warning, read the actual error message.

I was able to get my build working by adding this to my build.gradle file:

classpath 'org.jetbrains.kotlin:kotlin-gradle-plugin:1.3.71'