margelo/react-native-vision-camera #110: 🐛Command Libtool failed with a nonzero exit code

Ppang0405, 2021-03-27:
## Type

* [ X] I'm experiencing a full app crash
* [ ] A feature is not working correctly

## What
Installation fail

### Error

Libtool /Users/truongnguyen/Library/Developer/Xcode/DerivedData/X-dyftosiimfwvyygvyuvlazscpsak/Build/Products/Debug-iphonesimulator/react-native-vision-camera/libreact-native-vision-camera.a normal (in target 'react-native-vision-camera' from project 'Pods')
    cd /Volumes/Untitled/Documents/current_desktop/X/ios/Pods
    export IPHONEOS_DEPLOYMENT_TARGET\=11.0
    /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/libtool -static -arch_only x86_64 -D -syslibroot /Applications/Xcode.app/Contents/Developer/Platforms/iPhoneSimulator.platform/Developer/SDKs/iPhoneSimulator14.4.sdk -L/Users/truongnguyen/Library/Developer/Xcode/DerivedData/X-dyftosiimfwvyygvyuvlazscpsak/Build/Products/Debug-iphonesimulator/react-native-vision-camera -filelist /Users/truongnguyen/Library/Developer/Xcode/DerivedData/X-dyftosiimfwvyygvyuvlazscpsak/Build/Intermediates.noindex/Pods.build/Debug-iphonesimulator/react-native-vision-camera.build/Objects-normal/x86_64/react-native-vision-camera.LinkFileList -dependency_info /Users/truongnguyen/Library/Developer/Xcode/DerivedData/X-dyftosiimfwvyygvyuvlazscpsak/Build/Intermediates.noindex/Pods.build/Debug-iphonesimulator/react-native-vision-camera.build/Objects-normal/x86_64/react-native-vision-camera_libtool_dependency_info.dat -o /Users/truongnguyen/Library/Developer/Xcode/DerivedData/X-dyftosiimfwvyygvyuvlazscpsak/Build/Products/Debug-iphonesimulator/react-native-vision-camera/libreact-native-vision-camera.a

error: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/libtool: file: /Users/truongnguyen/Library/Developer/Xcode/DerivedData/X-dyftosiimfwvyygvyuvlazscpsak/Build/Intermediates.noindex/Pods.build/Debug-iphonesimulator/react-native-vision-camera.build/Objects-normal/x86_64/CameraViewManager-e5b2f19a5a2dc9ad5de82588b81358ee07c213da12b16015a11a8df253e24b68.o is not an object file (not allowed in a library)

## Environment

* Platform(s):
  * [X] iOS
  * [ ] Android
* react-native version: 0.64.0
* react-native-vision-camera version: latest
* react-native-reanimated version: 2.0.1

Ppang0405, 2021-03-27:
More information: My app generates by ignite-cli.

mrousavy, 2021-03-29:
I have never used ignite-cli before, so no idea why that is happening. Try opening an issue at the ignite-cli repo.

Ppang0405, 2021-04-06:
closed. react-native-cli works fine.

closed 2021-04-06 by Ppang0405

AbdullahEslah, 2025-08-11:
I've fixed it after cleaning cache from derived data folder of Xcode