margelo/react-native-vision-camera #121: 🐛 [ANDROID] after the installation Execution failed for task ':app:processDebugResources'.

qesya, 2021-04-21:
Execution failed for task ':app:processDebugResources'. > A failure occurred while executing com.android.build.gradle.internal.tasks.Workers$ActionFacade    > Android resource linking failed

## Type

* [ ] I'm cannot build my project with VisionCamera
* [ ] I'm experiencing a full app crash
* [ ] A feature is not working correctly
* [ ] Other: __________________

## What

## Reproducable sample

```tsx
// code here
```

## Environment

* Platform(s):
  * [ ] iOS
  * [ ] Android
* react-native version: ___
* react-native-vision-camera version: ___
* react-native-reanimated version: N/A

mrousavy, 2021-04-21:
1. Please share your entire logs. The line that says "A failure occured" does not contain any relevant information.
2. Can you run the example app?

mrousavy, 2021-05-06:
closing as it seems stale. let me know if you're still having issues & i'll reopen