margelo/react-native-vision-camera #127: getAvailableCameraDevices() always returns 0 / useCameraDevices() returns undefined

dhirajanand014, 2021-04-25:
Hi Team,

We are using you library to capture images. When using getAvailableCameraDevices() function from Camera, it returns the length 0 or if we use the 'useCameraDevices' hook, the value returned in undefined, Referring the same code as mentioned from the Sample in the API docs. 

Testing this on a Xiomi MI phone. Please look into the issue.

mrousavy, 2021-04-25:
Why did you not fill out the template? Share your versions. Which Xiaomi Mi is it? There are multiple versions afaik.
I believe that device does not return any Camera2 compatible devices, it's legacy

dhirajanand014, 2021-04-25:
Updated the issue with template. Xiomi MI Note 6 Pro is where this issue is occuring and use legacy Camera1 API. What should we do for this?

mrousavy, 2021-04-25:
VisionCamera requires Camera2 compatible devices, because a lot of features we have in this library are not compatible with Camera1.

To also add support for Camera1 devices I would have to implement https://github.com/cuvent/react-native-vision-camera/issues/78 first, since Camera1 devices will not support Preview + Photo Capture + Video Capture + Frame Processing. I understand that there are still many phones out there using Camera1, but after all it is a _deprecated_ API and I have a few features that have higher priority than supporting legacy devices (such as the Frame Processors PR).

On second thought, the Xiaomi MI Note 6 Pro was released 2018, that should be using Camera2 by now, no?

Try opening your project in Android Studio, then hit Command + Shift + O (two times) to open the `CameraViewModule.kt` file. Put a breakpoint in that line: 

https://github.com/cuvent/react-native-vision-camera/blob/5e2e9b6912e820708f3888c32d6375e3dcc91273/android/src/main/java/com/mrousavy/camera/CameraViewModule.kt#L127

and try stepping through the code to find out what devices are actually available, and why they don't get returned in that function.

dhirajanand014, 2021-04-25:
Cannot find the file CameraViewModule.kt

mrousavy, 2021-04-25:
> The cameraId returned in 0

is there only a single camera ID? what if you press "continue", does it not hit the breakpoint again with the next camera?

> which is legacy

how did you determine that?

mrousavy, 2021-04-25:
Does this get printed in your Logcat console (in Android Studio, hit command + 6):

https://github.com/cuvent/react-native-vision-camera/blob/5e2e9b6912e820708f3888c32d6375e3dcc91273/android/src/main/java/com/mrousavy/camera/CameraViewModule.kt#L136-L140

dhirajanand014, 2021-04-26:
Yes this gets printed and the debugger gets a second hit with value 1. However, the above statements is executed and printed.

mrousavy, 2021-04-26:
So for both camera devices (0 and 1) the above line (`"Skipping Camera #(id) because ...."`) will be printed? Could you put a breakpoint here:
https://github.com/cuvent/react-native-vision-camera/blob/5e2e9b6912e820708f3888c32d6375e3dcc91273/android/src/main/java/com/mrousavy/camera/CameraViewModule.kt#L135

and hover over the `hardwareLevel` variable - tell me what value it has (do this for both cameras).

Then put a breakpoint here: https://github.com/cuvent/react-native-vision-camera/blob/5e2e9b6912e820708f3888c32d6375e3dcc91273/android/src/main/java/com/mrousavy/camera/CameraViewModule.kt#L144

and tell me if that breakpoint is getting hit at all.

dhirajanand014, 2021-05-09:
Hi,

Sorry for the delay in reply. I checked with what you mentioned to check. 

Yes, for both the cameras for my phone - the hardware level value is got as 2. - 

First snapshot for first breakpoint - 

[image]

Second snapshot for second breakpoint  - 

[image]
- 

The line of log gets executed and printed. Not getting any clue how to resolve this.

kabus202, 2021-05-09:
@dhirajanand014  can you try this on device
go to setting >> additional setting >> developer options >> disable "Turn on MIUI optimization"

mrousavy, 2021-05-10:
I see, thanks for the detailled infos.

The problem here is that LEGACY devices do not support as many use-cases as normal Camera devices (and iPhones), which is the standard I was going for with VisionCamera.

So for a LEGACY device, you cannot add the "video-capture", "photo-capture", and "preview" use-case at the same time, meaning you could only add "photo-capture" and "preview", or "video-capture" and "preview".

See this table in the link I'm logging in your console:

[image]

With LIMITED, this already looks a bit better:

[image]

but those devices still aren't as good as iPhones, since iPhones can attach "photo-capture" + "video-capture" + "preview" without problems, all in MAXIMUM resolution (except preview obviously). Either way, since LIMITED devices support all three types of capturing at the same time, this is enough for VisionCamera.

And lastly, FULL devices:

[image]

Since VisionCamera allows you to record videos and capture photos in the same camera session without any quirks, I can't think of a nice way to add support for LEGACY devices. What happens if the user tries to use a LEGACY device and tries to start recording? What happens if he wants to record, but takes a picture instead? How will he be able to control whether he wants to add the "video-capture" or the "photo-capture" use-case - remember that he cannot add both?

These are all things that will result in a very ugly API, especially if this is only a legacy-support feature and will not be needed at all anymore in the near future (LEGACY devices are not being produced anymore I think?).

Does anyone know how many Android devices are still on LEGACY level?

dhirajanand014, 2021-05-12:
The issue to get the camera working itselfs stops us at the code point where we call the getAvailableCameraDevices method. Here both the cameras are returning value 0 right? How does the camera view work even without them?

mrousavy, 2021-05-12:
Sorry I don't understand the question, does it still return 0 devices even with my patch/PR?

dhirajanand014, 2021-05-13:
Tried with the steps mentioned and it dont work. Tried building the app and now shows only white screen. Nothing happens.

mrousavy, 2021-05-13:
A white screen? Are there any errors shown? Are there any logs in the metro bundler? Are there any logs in Android Studio Logcat? Did the breakpoint get hit?

dhirajanand014, 2021-05-15:
I thing you can merge.

mrousavy, 2021-05-17:
What does that mean? Did you test it or not?

yfunk, 2021-05-18:
> Does anyone know how many Android devices are still on LEGACY level?

I was also interested in this, so I did a bit of research. According to data from http://camera2probe.com ([raw](https://camera2probe.com/device/)) for 3630 unique devices starting at API 23 the split between supported hardware levels (for select minimum API versions) looks like this:

### Devices API 23 or newer 
All versions with reliable data. _84.90% market share according to Android Studio._
| Support   |        |
|-----------|--------|
| `LEVEL_3` | 22.59% |
| `FULL`    | 19.15% |
| `LIMITED` | 17.36% |
| `LEGACY`  | 40.91% |

### Devices API 25 or newer
_66.20% market share._
| Support   |        |
|-----------|--------|
| `LEVEL_3` | 29.57% |
| `FULL`    | 22.44% |
| `LIMITED` | 19.66% |
| `LEGACY`  | 28.33% |

### Devices API 27 or newer
Officially supported Android versions according to [Wikipedia](https://en.wikipedia.org/wiki/Android_version_history#Overview). _53.5% market share._
| Support   |        |
|-----------|--------|
| `LEVEL_3` | 35.05% |
| `FULL`    | 23.96% |
| `LIMITED` | 20.68% |
| `LEGACY`  | 20.31% |

### Devices API 29 or newer
"New" Android versions. _8.20% market share._
| Support   |        |
|-----------|--------|
| `LEVEL_3` | 47.20% |
| `FULL`    | 26.24% |
| `LIMITED` | 20.75% |
| `LEGACY`  | 5.81% |

Since React Native and this library currently support Android back to API 21, where almost half the devices have hardware level `LEGACY`, a feature like described in #78 would definitely be nice to have in the future.

mrousavy, 2021-05-19:
Those numbers are way higher than I expected. Thanks for the insight @yfunk.

It definitely makes sense to support LEGACY devices, whichever solution it is going to be. I want to try to not change a thing for the library/API to support LEGACY devices, since LEGACY devices do support 3 use-cases, even if video recording will be in PREVIEW sized frames:

[image]

> (at least that's how I interpreted [this table](https://developer.android.com/reference/android/hardware/camera2/CameraDevice#regular-capture))

Also, Snapchat works on LEGACY devices right? Snapchat also has three use-cases; PREVIEW, photo-capture and video-capture. They even have a fourth use-case, in-app-processing for those dog-mask filters, but I think that can be chained to the PREVIEW use-case.

mrousavy, 2021-05-19:
Also, are those market shares realistic values? How many of those phones actually do have e.g. snapchat or other apps installed? What if they're all just drug dealer or old grandma phones? @yfunk

mrousavy, 2021-05-19:
I investigated a bit, taking a look as to how it would be possible with Camera2. We have two approaches:

### Prioritize photo-capture

* Use normal PREVIEW buffer for the preview use-case
* Use PREVIEW sized buffer/frames for video-capture, so video recordings are in poor quality (PREVIEW size is very small on those LEGACY phones)
* Use normal MAXIMUM buffers for photo-capture

### Prioritize video-capture

* Use normal PREVIEW buffer for the preview use-case
* Use MAXIMUM or RECORD buffer/frames for the video-capture to achieve good quality video recordings
* Use PREVIEW buffers for photo-capture by simply screenshotting the PreviewView (same as `takeSnapshot` currently) - this does not support flash and results in poor quality still-images.

I have created a ticket at the google issue tracker asking for advice on this, but I think this is going to be a bad situation either way. Again, LEGACY devices aren't going to be around for long, new devices are being built and the percentages go down. Also, not sure how many people with LEGACY devices download modern apps (since VisionCamera is built for the Cuvent mobile app).

#78 would probably be the best solution to this, I'll try to find a clean API for that.

yfunk, 2021-05-19:
> Also, are those market shares realistic values? How many of those phones actually do have e.g. snapchat or other apps installed? What if they're all just drug dealer phones, airport tablets, old grandma phones, etc?

The data I used for this is from the Android Developer [Distribution dashboard](https://developer.android.com/about/dashboards), which represents all active Android devices during a 7-day period ending on January 8, 2021. This is what Google provides as a guideline for choosing the `minSdkVersion`, but I agree that this doesn't necessarily reflect actual users who would download modern apps.

I found [data from AppBrain SDK](https://www.appbrain.com/stats/top-android-sdk-versions) (app monetization), which may be better suited for reflecting potential app users and suggests a significantly higher market share of newer devices. I added it in the comment above.

closed 2021-06-07 by mrousavy

mrousavy, 2021-06-07:
I've implemented this in #168. Try upgrading to [**2.2.0**](https://github.com/cuvent/react-native-vision-camera/releases/tag/v2.2.0) and make sure to check out [the docs for the `supportsPhotoAndVideoCapture` prop](https://cuvent.github.io/react-native-vision-camera/docs/guides/devices#the-supportsphotoandvideocapture-prop) - let me know if it works for you!

K-Leon, 2021-07-20:
I just tested it - it looks like this issue isn't solved completely. 

react-native-camera for example falls back to Camera1 - which works - Camera2 ist just producing a black image. I even noticed the usage of these "crazy" hardware in newer devices. 

Is there any easy method to solve this? (Maybe read the images from Camera1 if necessary)

PR #168 (2021-06-07): feat: Separate usecases (decouple microphone, video, photo)
Adds

* `photo`
* `video`
* `audio`

props to configure what use-cases to attach. 

* Allows recording videos without audio
* Allows using the camera without microphone permission
* Allows faster camera startup by specifying what use-cases to attach

By default, they're all `false`.

## Side-effects

* (Android) Fixes PreviewView having a way too high resolution
* (Android) Fixes wrong orientation when not using a `format`
* (Android) Make `startRecording` non-blocking
* (Android) Reject with errors instead of crashing the app in `sta [... cut]