margelo/react-native-vision-camera #144: 🐛 The Swift pod `VisionCamera` depends upon `React-Core`, which does not define modules.

dan10gc, 2021-05-18:
## What

After installing the library on a react native project `0.63.2` I get the error below in `Logs` when running `npx pod-install` .

If I add  `use_modular_headers` I get this issue at build time for `glog` pod. -> ` Import of module 'glog.glog.log_severity' appears within namespace 'google'`

Any guidance on how to get past this issue is must appreciated. 

### Logs

```
The Swift pod `VisionCamera` depends upon `React-Core`, which does not define modules. To opt into those targets generating module maps (which is necessary to import them from Swift when building as static libraries), you may set `use_modular_headers!` globally in your Podfile, or specify `:modular_headers => true` for particular dependencies.
```

## Reproducable sample

```tsx
// code here
```

## Environment

* Platform(s):
  * [ ] iOS
  * [ ] Android
* react-native version: ___
* react-native-vision-camera version: ___
* react-native-reanimated version: N/A

mrousavy, 2021-05-18:
Hm, I believe [the `React-Core` dependency](https://github.com/cuvent/react-native-vision-camera/blob/310ad5fc4c6c9d7b5aa71ff6458e60bdf4fac6e3/VisionCamera.podspec#L43-L45) shouldn't be used since it is an umbrella pod.
I wonder how you could install Reanimated, since they [also specify `React-Core` in dependencies](https://github.com/software-mansion/react-native-reanimated/blob/c3cf8b7875405fdaa07753928cd7725ded62f20f/RNReanimated.podspec#L67) 🤔 

Could you try removing the `React-Core` dependency from `VisionCamera.podspec` and see if it builds? There might be some things missing that you have to add via another dependency..

mrousavy, 2021-05-24:
@dan10gc 🏓

dan10gc, 2021-05-24:
@mrousavy I apologize for not getting back to you earlier. So I was actually able to successfully install the pod files for vision-camera after I updated my project to  react-native `0.64.0`. I think one of the changes was they updated the `RCT-Folly` pod they were using and that got rid of the error. I then followed all the troubleshooting steps in the documentation and I was able to use the `<Camera/>` component. :)

closed 2021-05-24 by mrousavy

kaiyes, 2022-02-23:
Can confirm this error goes away once you update to `0.64.1` or later