margelo/react-native-vision-camera #146: 🐛 Selfie on Android Pixel 2 XL (OS version 11) is rotated

jphilipstevens, 2021-05-21:
## What
Platform: Android
Tested OS version: Android 11
Devices tested: Pixel 2 XL and Pixel 3a
library version: 1.0.11

I notice using both my app and your example is that Selfie images are rotated 90 degrees counterclockwise. I checked the metadata on my app and I can see Orientation is 0 (I believe this is 0 degrees?).

at the moment I cannot move up to 2.0 as reanimaed 2,0 has debugging issues due to TurboModules.

## Reproducable sample

1. Open the example app
2. switch to selfie camera
3. capture an image

The preview of the captured image appears rotated 90 degrees counter clockwise

## Environment

* Platform(s):
  * [ ] iOS
  * [ X] Android
* react-native version: __0.64_
* react-native-vision-camera version: _1.0.11__
* react-native-reanimated version: N/A

mrousavy, 2021-05-21:
> 90 degrees counterclockwise
so it's horizontal? does it look stretched or anything?

* You said "The preview of the captured image", does this mean the actual image is not rotated, only the small thumbnail?
* Is it only the captured image itself? 
* Does the live camera Preview in the app look correct?
* Do video recordings have the correct orientation?

jphilipstevens, 2021-05-21:
Thanks for replying. Here is an example: https://imgur.com/eZZwzHd 

I used the example app to take the selfie. in my case the phone was in portrait and the couch, lamp in the house should be verticle.

- You said "The preview of the captured image", does this mean the actual image is not rotated, only the small thumbnail?
  - its the actual image
- Is it only the captured image itself?
  - I believe the image is captured in the wrong orientation
- Does the live camera Preview in the app look correct?
  -  yup, the live capture looks fine
- Do video recordings have the correct orientation?
  - I just tried the example app in this repo and the picture is flipped horizontally.

mrousavy, 2021-05-21:
> the couch, lamp in the house should be vertical.
haha, I hope so!

> and the picture is flipped horizontally
yeah that's a limitation from CameraX.

Okay so that's definitely a bug, I'll try to find out where it comes from. I wonder what rotation value the `photo` object has in this line: https://github.com/cuvent/react-native-vision-camera/blob/160ec779404be276f4953d3d760bc21d0dddb1db/android/src/main/java/com/mrousavy/camera/CameraView%2BTakePhoto.kt#L74

jphilipstevens, 2021-05-21:
I checked the metadata and what I noticed is that ORientation is 0 on the JS side.

looking at CameraX there was some mention of a potential issue here:
https://stackoverflow.com/questions/63921260/android-camerax-image-rotated

Also, the video is flipped on the front camera is another issue. I can log that separately.

I am starting work soon so if I can I can look at some of this. We are using this lib in a future production release of a social network.

mrousavy, 2021-05-21:
Currently I don't have phone rotation implemented, and I think that by implementing phone rotation (letting CameraX know how the phone is currently rotated), this issue is going to get resolved.

I'll try to take a look at that soon 👍 

Interested about your social media app and how VisionCamera will play a role in that ;)

zifahm, 2021-05-21:
same here, ran the example app and selfie is rotated and also the video for the selfie camera is flipped as well

jphilipstevens, 2021-05-28:
@mrousavy let me know if you need testing. I am closing off some work. Note we are going to release with the vision camera 1.x because of issues with reanimated 2. I plan to review it after our initial release.

mrousavy, 2021-06-07:
@jphilipstevens can you test if the problem still occurs in [**2.2.0**](https://github.com/cuvent/react-native-vision-camera/releases/tag/v2.2.0)?

jphilipstevens, 2021-06-12:
@mrousavy Sorry for missing this. I've been swamped with work and few personal matters. I can test on 2.2.0 but I do have an issue with reanimated 2.0. The team is using 1.x until Turbo modules have debugging support.

I will give this version a spin though. I have a branch around (well like 10 branches my own experiments haha) with reanimated 2.x

mrousavy, 2021-06-12:
@jphilipstevens VisionCamera 2.x has nothing to do with Reanimated 2.x, you can even use VisionCamera 2.x without Reanimated installed at all.

jphilipstevens, 2021-06-13:
🤔 Oh, I will check my experiments from the past. I'll get the app on 2.x starting today and test. I am a bit swamped jumping between server-side and some leadership work but I want this to work as I have been impressed with the lib. 👍

zifahm, 2021-06-18:
This is still rotated

jphilipstevens, 2021-06-22:
Im going to test out 2.x today with the fixes. I'll report back on my findings

jphilipstevens, 2021-06-22:
I just tested and the image needs to be rotated 90 degrees clockwise.

Here is the resulting selfie image: https://ibb.co/BKngx6G

Here is what the camera preview shows: https://ibb.co/qC4Bwkg

jphilipstevens, 2021-06-25:
Ok i made a pr for this: https://github.com/mrousavy/react-native-vision-camera/pull/228

i am not sure the reason but I linked a SO where the matrix of data needs to be rotated. I could not validate on a breadth of devices, or why the flip leaves it rotated 90 degrees?

jphilipstevens, 2021-06-26:
fyi, i used patch-project on vision camera 2.4.1 in our project, and it's working. The team is doing a general QA on our app with a few android devices over the next week.

fedeerbes, 2021-09-28:
Hi @jphilipstevens, can you share your patch file? How was the result of your QA team?
We're planning to move from react-native-camera cause some issues with low-mid end android devices and this library looks amazing @mrousavy. =)

jphilipstevens, 2021-10-01:
@fedeerbes I had a PR here: https://github.com/mrousavy/react-native-vision-camera/pull/228

But it may not be an thorough fix. More of a workaround that is working right now. YMMV

viljark, 2021-11-26:
I'm getting rotated selfies with all of my Android phones (LG V30 - Android 9, Samsung Galaxy A12 - Android 11, Samsung Galaxy 7), but not with my Android tablet (which defaults to the landscape mode when rotation is locked, maybe sensor placement difference?).
iOS is ok.

jzxchiang1, 2021-12-01:
Yeah, I'm seeing this with my Pixel 3a emulator as well. The captured photo is also incorrectly rotated 90 degrees counterclockwise.

rodgomescrn, 2021-12-08:
It's too bad to launch an app in production and randomly deal with this issue

iam doing something like that

if (devices.includes(device)){
   matrix.postRotate(90f);
}

wilsonw926, 2021-12-11:
+1 for samsung galaxy s10e.

didithilmy, 2022-01-07:
I think I know where the issue is.

In the ImageProxy.save method, if `flipHorizontally` is set to true, it invokes the `flipImage` method [1], which flips the image without taking the `Orientation` metadata into consideration. When the image is saved back to the tmp path, the EXIF metadata is stripped, including `Orientation`.

I think this is why the issue only exists on front-facing camera.

[1] https://github.com/mrousavy/react-native-vision-camera/blob/main/android/src/main/java/com/mrousavy/camera/utils/ImageProxy.save.kt#L43

closed 2022-01-11 by mrousavy

SayeedAfridi, 2023-11-09:
@mrousavy I am using version `3.6.4`. This issue is still there. Selfie is rotated 90 degree after taking photo.

mrousavy, 2023-11-10:
Yea again - as mentioned in all those other issues, Orientation isn't explicitly implemented in VisionCamera V3 - so there are edge cases where it's not correct.

I'm raising money here and once the pool is full I'll implement orientation once and for all: https://github.com/mrousavy/react-native-vision-camera/issues/1891

FazilMuhammed, 2023-11-16:
@mrousavy please help.Selfie is rotated 90 degree after taking photo. any solution. couple months I am stuck with this issue

mrousavy, 2023-11-17:
@FazilMuhammed 

[image]

SayeedAfridi, 2023-11-21:
@FazilMuhammed apply this patch for now...
```
diff --git a/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraView+TakePhoto.kt b/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraView+TakePhoto.kt
index 88c085f..9f22575 100644
--- a/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraView+TakePhoto.kt
+++ b/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraView+TakePhoto.kt
@@ -7,6 +7,7 @@ import android.graphics.BitmapFactory
 import android.graphics.ImageFormat
 import android.graphics.Matrix
 import android.hardware.camera2.*
+import androidx.exifinterface.media.ExifInterface
 import android.util.Log
 import com.facebook.react.bridge.Arguments
 import com.facebook.react.bridge.ReadableMap
@@ -71,7 +72,35 @@ private fun writePhotoToFile(photo: CameraSession.CapturedPhoto, file: File) {
     val imageBytes = ByteArray(byteBuffer.remaining()).apply { byteBuffer.get(this) }
     val bitmap = BitmapFactory.decodeByteArray(imageBytes, 0, imageBytes.size)
     val matrix = Matrix()
-    matrix.preScale(-1f, 1f)
+
+    // Fix rotation of the mirrored image
+    val exif = ExifInterface(imageBytes.inputStream())
+    val orientation = exif.getAttributeInt(ExifInterface.TAG_ORIENTATION, ExifInterface.ORIENTATION_UNDEFINED)
+
+    when (orientation) {
+      ExifInterface.ORIENTATION_ROTATE_180 -> {
+        matrix.setRotate(180f)
+        matrix.postScale(-1f, 1f)
+      }
+      ExifInterface.ORIENTATION_FLIP_VERTICAL -> {
+        matrix.setRotate(180f)
+      }
+      ExifInterface.ORIENTATION_TRANSPOSE -> {
+        matrix.setRotate(90f)
+      }
+      ExifInterface.ORIENTATION_ROTATE_90 -> {
+        matrix.setRotate(90f)
+        matrix.postScale(-1f, 1f)
+      }
+      ExifInterface.ORIENTATION_TRANSVERSE -> {
+        matrix.setRotate(-90f)
+      }
+      ExifInterface.ORIENTATION_ROTATE_270 -> {
+        matrix.setRotate(-90f)
+        matrix.postScale(-1f, 1f)
+      }
+    }
+
     val processedBitmap =
       Bitmap.createBitmap(bitmap, 0, 0, bitmap.width, bitmap.height, matrix, false)
     FileOutputStream(file).use { stream ->
```

BarbieriMatheus, 2024-01-31:
I had this problem too.

```
Tested on: Oneplus A5010 (Android 10);
{ "react": "18.2.0", "react-native": "0.72.9", "react-native-vision-camera": "^3.8.2" }
```

PR #719 (2022-01-11): fix: Fix image rotated when flipped in Android

## What

This PR fixes a bug that causes image to rotate when using a front camera in Android.

## Changes
This PR changes the `flipImage` method in ImageProxy.save.kt to fix the image rotation based on Exif `Orientation` value.

## Tested on

Pixel 5, Android 11

## Related issues
Fixes #146