margelo/react-native-vision-camera #147: 🐛  [Android] Unmounting CameraView throw react-native-reanimated issue

elgorditosalsero, 2021-05-25:
## What

When unmounting the Camera component (wrapped in another one), throws an `Error: Frame Processors are not enabled. Make sure you install react-native-reanimated 2.1.0 or above!`.

[image]

On iOS everything works fine...

Btw, `react-native-reanimated` is installed because others libs depends on it and they works fine.

### Logs

```
2021-05-25 16:01:12.685 21551-21551/com.smarteven I/com.smarteven: Rejecting re-init on previously-failed class java.lang.Class<androidx.camera.extensions.ExtensionsManager$2>: java.lang.NoClassDefFoundError: Failed resolution of: Landroidx/camera/extensions/impl/InitializerImpl$OnExtensionsDeinitializedCallback;
2021-05-25 16:01:12.685 21551-21551/com.smarteven I/com.smarteven:     at com.google.common.util.concurrent.ListenableFuture androidx.camera.extensions.ExtensionsManager.init(android.content.Context) (ExtensionsManager.java:120)
2021-05-25 16:01:12.685 21551-21551/com.smarteven I/com.smarteven:     at java.lang.Object com.mrousavy.camera.CameraViewModule$getAvailableCameraDevices$1.invokeSuspend(java.lang.Object) (CameraViewModule.kt:102)
2021-05-25 16:01:12.685 21551-21551/com.smarteven I/com.smarteven:     at void kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(java.lang.Object) (ContinuationImpl.kt:33)
2021-05-25 16:01:12.685 21551-21551/com.smarteven I/com.smarteven:     at void kotlinx.coroutines.DispatchedTask.run() (DispatchedTask.kt:106)
2021-05-25 16:01:12.685 21551-21551/com.smarteven I/com.smarteven:     at void android.os.Handler.handleCallback(android.os.Message) (Handler.java:873)
2021-05-25 16:01:12.685 21551-21551/com.smarteven I/com.smarteven:     at void android.os.Handler.dispatchMessage(android.os.Message) (Handler.java:99)
2021-05-25 16:01:12.685 21551-21551/com.smarteven I/com.smarteven:     at void android.os.Looper.loop() (Looper.java:205)
2021-05-25 16:01:12.685 21551-21551/com.smarteven I/com.smarteven:     at void android.app.ActivityThread.main(java.lang.String[]) (ActivityThread.java:6991)
2021-05-25 16:01:12.685 21551-21551/com.smarteven I/com.smarteven:     at java.lang.Object java.lang.reflect.Method.invoke(java.lang.Object, java.lang.Object[]) (Method.java:-2)
2021-05-25 16:01:12.685 21551-21551/com.smarteven I/com.smarteven:     at void com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run() (RuntimeInit.java:493)
2021-05-25 16:01:12.685 21551-21551/com.smarteven I/com.smarteven:     at void com.android.internal.os.ZygoteInit.main(java.lang.String[]) (ZygoteInit.java:884)
2021-05-25 16:01:12.685 21551-21551/com.smarteven I/com.smarteven: Caused by: java.lang.ClassNotFoundException: Didn't find class "androidx.camera.extensions.impl.InitializerImpl$OnExtensionsDeinitializedCallback" on path: DexPathList[[zip file "/data/app/com.smarteven-LiELhOZIgGJtrNIBbCJSYw==/base.apk"],nativeLibraryDirectories=[/data/app/com.smarteven-LiELhOZIgGJtrNIBbCJSYw==/lib/arm64, /data/app/com.smarteven-LiELhOZIgGJtrNIBbCJSYw==/base.apk!/lib/arm64-v8a, /system/lib64, /system/product/lib64, /system/vendor/lib64]]
2021-05-25 16:01:12.685 21551-21551/com.smarteven I/com.smarteven:     at java.lang.Class dalvik.system.BaseDexClassLoader.findClass(java.lang.String) (BaseDexClassLoader.java:134)
2021-05-25 16:01:12.685 21551-21551/com.smarteven I/com.smarteven:     at java.lang.Class java.lang.ClassLoader.loadClass(java.lang.String, boolean) (ClassLoader.java:379)
2021-05-25 16:01:12.685 21551-21551/com.smarteven I/com.smarteven:     at java.lang.Class java.lang.ClassLoader.loadClass(java.lang.String) (ClassLoader.java:312)
2021-05-25 16:01:12.685 21551-21551/com.smarteven I/com.smarteven:     at com.google.common.util.concurrent.ListenableFuture androidx.camera.extensions.ExtensionsManager.init(android.content.Context) (ExtensionsManager.java:120)
2021-05-25 16:01:12.685 21551-21551/com.smarteven I/com.smarteven:     at java.lang.Object com.mrousavy.camera.CameraViewModule$getAvailableCameraDevices$1.invokeSuspend(java.lang.Object) (CameraViewModule.kt:102)
2021-05-25 16:01:12.685 21551-21551/com.smarteven I/com.smarteven:     at void kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(java.lang.Object) (ContinuationImpl.kt:33)
2021-05-25 16:01:12.685 21551-21551/com.smarteven I/com.smarteven:     at void kotlinx.coroutines.DispatchedTask.run() (DispatchedTask.kt:106)
2021-05-25 16:01:12.685 21551-21551/com.smarteven I/com.smarteven:     at void android.os.Handler.handleCallback(android.os.Message) (Handler.java:873)
2021-05-25 16:01:12.685 21551-21551/com.smarteven I/com.smarteven:     at void android.os.Handler.dispatchMessage(android.os.Message) (Handler.java:99)
2021-05-25 16:01:12.685 21551-21551/com.smarteven I/com.smarteven:     at void android.os.Looper.loop() (Looper.java:205)
2021-05-25 16:01:12.685 21551-21551/com.smarteven I/com.smarteven:     at void android.app.ActivityThread.main(java.lang.String[]) (ActivityThread.java:6991)
2021-05-25 16:01:12.685 21551-21551/com.smarteven I/com.smarteven:     at java.lang.Object java.lang.reflect.Method.invoke(java.lang.Object, java.lang.Object[]) (Method.java:-2)
2021-05-25 16:01:12.685 21551-21551/com.smarteven I/com.smarteven:     at void com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run() (RuntimeInit.java:493)
2021-05-25 16:01:12.685 21551-21551/com.smarteven I/com.smarteven:     at void com.android.internal.os.ZygoteInit.main(java.lang.String[]) (ZygoteInit.java:884)
2021-05-25 16:01:12.686 21551-21551/com.smarteven I/com.smarteven: Rejecting re-init on previously-failed class java.lang.Class<androidx.camera.extensions.ExtensionsManager$2>: java.lang.NoClassDefFoundError: Failed resolution of: Landroidx/camera/extensions/impl/InitializerImpl$OnExtensionsDeinitializedCallback;
2021-05-25 16:01:12.686 21551-21551/com.smarteven I/com.smarteven:     at com.google.common.util.concurrent.ListenableFuture androidx.camera.extensions.ExtensionsManager.init(android.content.Context) (ExtensionsManager.java:120)
2021-05-25 16:01:12.686 21551-21551/com.smarteven I/com.smarteven:     at java.lang.Object com.mrousavy.camera.CameraViewModule$getAvailableCameraDevices$1.invokeSuspend(java.lang.Object) (CameraViewModule.kt:102)
2021-05-25 16:01:12.686 21551-21551/com.smarteven I/com.smarteven:     at void kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(java.lang.Object) (ContinuationImpl.kt:33)
2021-05-25 16:01:12.686 21551-21551/com.smarteven I/com.smarteven:     at void kotlinx.coroutines.DispatchedTask.run() (DispatchedTask.kt:106)
2021-05-25 16:01:12.686 21551-21551/com.smarteven I/com.smarteven:     at void android.os.Handler.handleCallback(android.os.Message) (Handler.java:873)
2021-05-25 16:01:12.686 21551-21551/com.smarteven I/com.smarteven:     at void android.os.Handler.dispatchMessage(android.os.Message) (Handler.java:99)
2021-05-25 16:01:12.686 21551-21551/com.smarteven I/com.smarteven:     at void android.os.Looper.loop() (Looper.java:205)
2021-05-25 16:01:12.686 21551-21551/com.smarteven I/com.smarteven:     at void android.app.ActivityThread.main(java.lang.String[]) (ActivityThread.java:6991)
2021-05-25 16:01:12.686 21551-21551/com.smarteven I/com.smarteven:     at java.lang.Object java.lang.reflect.Method.invoke(java.lang.Object, java.lang.Object[]) (Method.java:-2)
2021-05-25 16:01:12.686 21551-21551/com.smarteven I/com.smarteven:     at void com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run() (RuntimeInit.java:493)
2021-05-25 16:01:12.686 21551-21551/com.smarteven I/com.smarteven:     at void com.android.internal.os.ZygoteInit.main(java.lang.String[]) (ZygoteInit.java:884)
2021-05-25 16:01:12.686 21551-21551/com.smarteven I/com.smarteven: Caused by: java.lang.ClassNotFoundException: Didn't find class "androidx.camera.extensions.impl.InitializerImpl$OnExtensionsDeinitializedCallback" on path: DexPathList[[zip file "/data/app/com.smarteven-LiELhOZIgGJtrNIBbCJSYw==/base.apk"],nativeLibraryDirectories=[/data/app/com.smarteven-LiELhOZIgGJtrNIBbCJSYw==/lib/arm64, /data/app/com.smarteven-LiELhOZIgGJtrNIBbCJSYw==/base.apk!/lib/arm64-v8a, /system/lib64, /system/product/lib64, /system/vendor/lib64]]
2021-05-25 16:01:12.686 21551-21551/com.smarteven I/com.smarteven:     at java.lang.Class dalvik.system.BaseDexClassLoader.findClass(java.lang.String) (BaseDexClassLoader.java:134)
2021-05-25 16:01:12.686 21551-21551/com.smarteven I/com.smarteven:     at java.lang.Class java.lang.ClassLoader.loadClass(java.lang.String, boolean) (ClassLoader.java:379)
[... cut ...]
        in Camera (at Photo.tsx:19)
        in RCTView (at View.js:34)
        in View (at Photo.tsx:15)
        in SEPhotoBS (at Add.tsx:132)
        in RCTView (at View.js:34)
        in View (at Add.tsx:64)
        in SEAddActivityBS (at Activities.tsx:156)
```

## Reproducable sample

1. Wrap `<Camera />` inside a component
2. Mount it
3. Unmount it

## Environment

* Platform(s):
  * [ ] iOS
  * [x ] Android
* react-native version: 0.64.1
* react-native-vision-camera version: 2.0.0
* react-native-reanimated version: 2.1.0

mrousavy, 2021-05-25:
My bad! I'll fix this asap

mrousavy, 2021-05-25:
Check out [v2.0.1](https://github.com/cuvent/react-native-vision-camera/releases/tag/v2.0.1), let me know if this fixes it for you!

closed 2021-05-25 by elgorditosalsero