margelo/react-native-vision-camera #153: ✨ Return mime (content type) from video recordings

jphilipstevens, 2021-05-28:
## What

I noticed that the Video File object does not include the mime type. Including it in the VideFile would benefit those using the video to send over networks.

## Alternatives/Workarounds

I am going to use https://www.npmjs.com/package/react-native-mime-types in order to get the mime from the file

```javascript
import * as mime from "react-native-mime-types";

camera.current.startRecording({
    flash: flash,
    onRecordingError: (error) => {
        onStoppedRecording();
    },
    onRecordingFinished: (video) => {
        const videoSource = {
            path: video.path,
            mimeType: mime.lookup(video.path),
        };
        onMediaCaptured(video, "video");
        onStoppedRecording();
    },
});

```

jphilipstevens, 2021-05-28:
threw this one in quickly. Let me know if you need more details

mrousavy, 2021-05-28:
I believe videos are _always_ `mov` on iOS and `mp4` on Android, but that's a good feature request - I'll take a look when I can 👍

jphilipstevens, 2021-06-03:
@mrousavy point me to the native code where I set this and I can code it locally. We use patch-project so I think I could make some tweaks on 1.x and experiment.
After a few weeks of making sleep optional, I may finally have some time to experiment hahahaha

erennyuksell, 2023-08-28:
@mrousavy can we add this feature?

mrousavy, 2023-08-28:
@erennyuksell you can add this feature if you want yes

mrousavy, 2024-01-30:
Hey @jphilipstevens - what approach did you go for? I'm not sure how I would get the mime-type, but effectively it just always is `video/${codec}`, where `codec` is either `mp4` or `mov`, right?

maxbbb, 2024-03-22:
Came across this looking into something unrelated, but wanted to note the mime-type for `mov` is `video/quicktime`, not `video/mov`. So for iOS it would be `video/quicktime` and Android `video/mp4`

Code-Victor, 2024-11-01:
Please, what is the mime type for pictures on Android  and iOS