margelo/react-native-vision-camera #162: 🐛 Camera and Microphone permission errors are confused on Android

G0retZ, 2021-06-03:
## What

The library reports wrong permissions error.
I've found the place that needs a change so I guess it's a simple typo
Just swap these 2 lines

https://github.com/cuvent/react-native-vision-camera/blob/b6a67d5cedb762f69a29c02b6f4184892f4d606f/android/src/main/java/com/mrousavy/camera/CameraView.kt#L221
https://github.com/cuvent/react-native-vision-camera/blob/b6a67d5cedb762f69a29c02b6f4184892f4d606f/android/src/main/java/com/mrousavy/camera/CameraView.kt#L224

Should I create a PR or you will do this by yourself?

### Logs

Not needed

## Reproducable sample

Just give only camera or only mic permission and see the error result.

## Environment

any

G0retZ, 2021-06-03:
Here is the patch if someone needs it:

`diff --git a/node_modules/react-native-vision-camera/android/build/.transforms/511557eeebd9ea819a6492c074d783a3.bin b/node_modules/react-native-vision-camera/android/build/.transforms/511557eeebd9ea819a6492c074d783a3.bin
index 4c26b34..ea56f24 100644
--- a/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraView.kt
+++ b/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraView.kt
@@ -218,10 +218,10 @@ class CameraView(context: Context) : FrameLayout(context), LifecycleOwner {
       val startTime = System.currentTimeMillis()
       Log.i(TAG, "Configuring session...")
       if (ContextCompat.checkSelfPermission(context, Manifest.permission.CAMERA) != PackageManager.PERMISSION_GRANTED) {
-        throw MicrophonePermissionError()
+        throw CameraPermissionError()
       }
       if (ContextCompat.checkSelfPermission(context, Manifest.permission.RECORD_AUDIO) != PackageManager.PERMISSION_GRANTED) {
-        throw CameraPermissionError()
+        throw MicrophonePermissionError()
       }
       if (cameraId == null) {
         throw NoCameraDeviceError()
`

mrousavy, 2021-06-03:
Oh, nice catch! Would you mind creating a PR? I'm on mobile

PR #163 (2021-06-03): Resolve missing permission errors confusion
## What

This PR fixes a typo in a CameraView.kt

## Changes

* Swapped permission errors to throw.

## Tested on

* iPhone 11 , iOS 14.3
* Nokia 5.3, Android 10

## Related issues

 * Resolves #162
closed 2021-06-03 by mrousavy