margelo/react-native-vision-camera #164: ❓

G0retZ, 2021-06-04:
## Question

Do you have any estimation on Frame processor for Android?
Or mb you need some help with it?
Desperately need that :-)

mrousavy, 2021-06-04:
I don't have an estimation, as it's really hard to plan this. After all, it's new "technology" that no one has used like this before.

I will start with the development next week, and I think most of the work is just getting the correct CMake setup. If you have experience with CMake, C++ or JNI, let me know.

G0retZ, 2021-06-04:
Unfortunately I'm not that experienced in CMake, C++ or JNI
But once upon a time I was building and integrating pjsip (C++ and swig) with OPUS codec(before official support) and with the Android project.
From what I remember, the JNI Android API is much more reduced in comparison to the Java/Kotlin API.
And the key thing about image processing was that frames were gathered from Java code (Android API) and then passed to the C++ code (pjsip) that reused it's common code base for encoding and SIP sessions...
And also JNI works in the same Java threads (can create and use them).
So from that time I learned that JNI might be good for heavy calculations but not for heavy platform API usage.

Hope this info helps somehow.

PR #196 (2021-06-27): feat: Frame Processors for Android
implements a super amazing feature for android 

resolves #164 
resolves #212 

## Blockers

Frame Processors work on Android - and they're incredibly fast! 🤩 But before merging & releasing this, I have two blockers which only affect the CI/GitHub actions but I'd still like to have them resolved before this gets merged:

* This issue: https://github.com/software-mansion/react-native-reanimated/issues/2129, as the CI fails and can't build. I have used `patch-package` to hotfix it, but I'd rather wait for a new Reanimated release. (2.3.0)
* The example app can' [... cut]
closed 2021-06-27 by mrousavy