margelo/react-native-vision-camera #167: 🐛 iOS release build error

dcangulo, 2021-06-04:
## What
Doesn't build on release mode but builds on debug mode.

### Logs
```
<unknown>:0: error: module map file ‘/Users/davidangulo/Library/Developer/Xcode/DerivedData/project_mobile-erivebwhxgzkcgcenpcxgkdgfiqi/Build/Intermediates.noindex/ArchiveIntermediates/project_mobile/BuildProductsPath/Release-iphoneos/VisionCamera/VisionCamera.modulemap’ not found
1 error generated.
<unknown>:0: error: failed to emit precompiled header ‘/Users/davidangulo/Library/Developer/Xcode/DerivedData/project_mobile-erivebwhxgzkcgcenpcxgkdgfiqi/Build/Intermediates.noindex/ArchiveIntermediates/project_mobile/PrecompiledHeaders/project_mobile-Bridging-Header-swift_2G0UQ3CJSS46M-clang_2YKEVN7TTCF9E.pch’ for bridging header ‘/Users/davidangulo/Documents/projects/project_mobile/ios/project_mobile-Bridging-Header.h’
3 errors generated.
```

## Reproducable sample

```tsx
// code here
```

## Environment

* Platform(s):
  * [X] iOS
  * [ ] Android
* react-native version: 0.64.1
* react-native-vision-camera version: 2.1.0
* react-native-reanimated version: 1.13.2

## Solution
What worked for me is this answer: https://stackoverflow.com/a/46523347/9375533

Aside from setting `platform :ios` to `11.0`, I also need to update `iOS Deployment Target` to `11.0`

[image]

Maybe we should add it to the docs? Specifically https://github.com/cuvent/react-native-vision-camera/blob/main/docs/docs/guides/TROUBLESHOOTING.mdx

mrousavy, 2021-06-07:
Just to confirm - setting the iOS Deployment Target to 11.0 fixes those build issues for you? If you set it back to 10.0, does it fail again?

dcangulo, 2021-06-07:
@mrousavy Yes

mrousavy, 2021-06-07:
Alright, thank you for the detailed report! Added info to the TROUBLESHOOTING doc in https://github.com/cuvent/react-native-vision-camera/commit/27bd8eae231148fe949c84bc98f0a951e1764d07.

closed 2021-06-07 by mrousavy

OmerMessing, 2022-07-04:
@mrousavy I have this issue and I have both 'platform :ios' set to 11.0 and iOS Deployment Target is 11.0. I still get an error.

fatal error: module map file '/Users/**********/Library/Developer/Xcode/DerivedData/CookingApp-brkrxhnvzpztyzeyxljqkgaaddav/Build/Products/Debug-iphoneos/VisionCamera/VisionCamera.modulemap' not found