margelo/react-native-vision-camera #184: 🐛 frameProcessorFps not respected?

G0retZ, 2021-06-08:
## What

I've  created a frame processor using ImageLabeler as template.
It completes in 200ms per frame
So I tried `frameProcessorFps={2}` and the default one, but anyway Frames got dropped.
Should it throttle frames before FrameProcessor gets in play or am I missing something?

### Logs

```
2021-06-08 12:38:17.886942+0300  Warning: Frame Processor function took 192 ms to execute. This blocks the video queue from recording, optimize your frame processor!
2021-06-08 12:38:17.887228+0300  [native] VisionCamera.captureOutput(_:didDrop:from:): Dropped a Frame. This might indicate that your Frame Processor is doing too much work. Either throttle the frame processor's frame rate, or optimize your frame processor's execution speed. Frame drop reason: FrameWasLate
2021-06-08 12:38:17.888389+0300  [javascript] [native] VisionCamera.captureOutput(_:didDrop:from:): Dropped a Frame. This might indicate that your Frame Processor is doing too much work. Either throttle the frame processor's frame rate, or optimize your frame processor's execution speed. Frame drop reason: FrameWasLate
```

## Reproducable sample

The frame processor itself

```objectiveC
//
//  VisionCameraQRScanner.m
//
//  Created by Sergei Mitrofanov on 07.06.21.
//

#import <VisionCamera/FrameProcessorPlugin.h>
#import <UIKit/UIKit.h>
#import <React/RCTLog.h>
#import <React/RCTUtils.h>

@interface QRCodeFrameProcessorPlugin : NSObject

+ (CIDetector*) detector;

@end

@implementation QRCodeFrameProcessorPlugin

+ (CIDetector*) detector {
  static CIDetector* detector = nil;
  if (detector == nil) {
    NSDictionary *detectorOptions = @{@"CIDetectorAccuracy": @"CIDetectorAccuracyHigh"};
    detector = [CIDetector detectorOfType:CIDetectorTypeQRCode context:nil options:detectorOptions];
  }
  return detector;
}

static inline id scanQRCodes(CMSampleBufferRef buffer, NSArray* arguments) {
  CVImageBufferRef cvImage = CMSampleBufferGetImageBuffer(buffer);
  CIImage *ciImage = [[CIImage alloc] initWithCVPixelBuffer:cvImage];

  NSArray* features = [[QRCodeFrameProcessorPlugin detector] featuresInImage:ciImage];
  NSMutableArray* results = [NSMutableArray arrayWithCapacity:features.count];
  for (CIQRCodeFeature* feature in features) {
    [results addObject:feature.messageString];
  }

  return results;
}

VISION_EXPORT_FRAME_PROCESSOR(scanQRCodes)

@end
```

## Environment

* Platform(s):
  * [X] iOS
  * [ ] Android
* react-native version: 0.64.1
* react-native-vision-camera version: 2.2.0
* react-native-reanimated version: 2.2.0

G0retZ, 2021-06-08:
I tried to log the workout itself like this but for unknown reason I got no logs at all

```tsx
  const frameProcessor = useFrameProcessor((frame) => {
    'worklet';

    console.log('Boom frame');

    const codes = scanQRCodes(frame);
    console.log('Boom', codes);
    if (codes && codes.length > 0) {
      setScannedData(codes[0]);
    }
  }, []);
```

mrousavy, 2021-06-08:
Hi!

The frame processor itself looks good to me, at what resolution are you running this on? If you didn't choose a format, chances are that you are receiving 4k frames. Try to get the `frame.width` property to find this out. Ideally you want to resize (downscale) the frame before running AI on it.

As for the logging, I just remembered that I didn't bind `console.log` yet. You can log using `_log`, which prints to the native Xcode console - I'll try to find a workaround to get the logs into your metro console as well!

G0retZ, 2021-06-08:
Thank you for fast answer!
Gotcha, will try to set a format now.

mrousavy, 2021-06-08:
Right, try choosing a format with a `videoWidth` of something low like 720 or even lower. I'm also working on a plugin to freely resize buffers right now: https://github.com/mrousavy/vision-camera-resize-plugin

mrousavy, 2021-06-08:
Also, `frameProcessorFps` is respected as far as I can see. The frame processor still gets executed twice a second (2 FPS), but the Video Queue itself runs at a higher FPS (for video recordings). 

If you don't use video recordings at all, you can set `fps` to something acceptable for the user (e.g. `30` or `60`), run your frame processor at 2 FPS and optimize the execution speed by downsizing the frames (or using a smaller format in the first place). I can also add a functionality to disable the warning that gets printed.

G0retZ, 2021-06-08:
Oh, I got it.
So it throttles the frames but this doesn't mean frame processor is allowed to execute any longer than 33ms.

I guess this comment made me think that I have all the time between throttled frames to execute.
```tsx
...
     * If you're using higher values, always check your Xcode/Android Studio Logs to make sure your frame processors are executing fast enough
     * without blocking the video recording queue.
...
```

So should I read it as I will get 33ms for the Frame processor to complete a task or less than that if `frameProcessorFps` set lower?

G0retZ, 2021-06-08:
Unfortunately even 192*192 took 170 ms to scan the QR Code...

Is it possible to pass a callback or a promise handle into frame processor arguments just like to a react native module?

mrousavy, 2021-06-08:
It is allowed to execute longer, that's perfectly fine. 

If your camera is set to 30 FPS, then that means each frame has 33ms to be processed. Remember that the video recorder is also a sort of frame processing.
If you take longer than 33ms, the next frame will be dropped, but that's only a problem if you're recording videos since it looks like there's a small stutter.

If you don't use video recordings, 200ms is a reasonable time for the frame processor to execute - you can run it 5 times a second (5 FPS), which might be enough for your use-case.

mrousavy, 2021-06-08:
what does that frame processor do? I'm pretty sure there are QR code scanners that are faster than 200ms

mrousavy, 2021-06-08:
In other words: Frame Processors have exactly `1 / FPS`ms to execute, regardless of what `frameProcessorFps` is set to. `fps` determines how fast they have to execute. `frameProcessorFps` determines how often they get executed if you don't want to run them on every frame.

"have to" execute is a bold statement, since there are use cases where you would be perfectly fine with dropped frames. For example, if you don't use video recordings, frames can be dropped - that's perfectly fine. If you do use video recordings and like 1 out of 20 frames gets dropped, that might be fine for you too, the resulting video might look a bit stuttery, but I haven't experimented with that yet.

> Is it possible to pass a callback or a promise handle into the frame processor arguments just like to a react native module?

I have implemented that functionality, but I didn't test it yet. It should work perfectly fine, just remember that the callbacks have to be worklets:

```js
  const frameProcessor = useFrameProcessor((frame) => {
    'worklet';

    scanQRCodes(frame, (codes) => {
        _log(`QR code plugin completed! Codes: ${codes}`)
    });
  }, []);
```

or dispatch to JS:

```js
  const onQrCodeRead = useCallback((codes) => {
    console.log(`Got QR code: ${codes}`)
    navigation.push("ProductPage", { articleId: codes[0] })
  }, []);
  const frameProcessor = useFrameProcessor((frame) => {
    'worklet';

    scanQRCodes(frame, (codes) => {
        _log(`QR code plugin completed! Codes: ${codes}`)
       runOnJS(onQrCodeRead)(codes)
    });
  }, []);
```

and if you take a look at the docs, the type for this would be a `RCTResponseSenderBlock`:
[image]

(intended usage: first argument is result, second argument is error)

G0retZ, 2021-06-08:
Thank you a lot for your help!
Now this all is much clearer for me.
Will report once will got the satisfying result for my case :-)

G0retZ, 2021-06-08:
Ok, looks like it doesn't work after all =(
I feel there is something wrong happening.

Tried to use `runOnJS` and every time I get "Don't try to clone an object backed by a different Runtime" and app hangs immediately
Even if I just call a log function like this
```tsx
const logCodes = (codes) => {
 console.log('--- Codes = ', codes);
}

const frameProcessor = useFrameProcessor((frame) => {
    'worklet';

    const scannedCodes = scanQRCodes(frame);
    runOnJS(logCodes)(scannedCodes)
  }, []);
```

Tried to use SharedValue  like this
```tsx
  const randomWidth = useSharedValue<string | undefined>(undefined);
  const frameProcessor = useFrameProcessor((frame) => {
    'worklet';

    // eslint-disable-next-line no-undef
    _log('Boom frame');

    const codes = scanQRCodes(frame);
    // eslint-disable-next-line no-undef
    if (codes && codes.length > 0) {
      _log(`Boooom ${codes}`);
      // runOnJS(onQRCodeDetected)(codes[0]);
      randomWidth.value = codes[0];
    }
  }, []);
  console.log('--------- Boom', randomWidth);
```

And still no luck - The value never changes despite of there are actual native logs about data processing results.

Struggling for hours with this ='(
What am I missing with this worklet?

G0retZ, 2021-06-08:
Happens in `getValue` here https://github.com/cuvent/react-native-vision-camera/blob/be30d55df5ac6b588108add4023668120f36706c/ios/Frame%20Processor/FrameProcessorRuntimeManager.mm#L151

G0retZ, 2021-06-09:
For now implemented with a crutch by dispatching scan results via react native emitter module...
Thanks anyway.

mrousavy, 2021-06-09:
Sorry, was gone yesterday. That's weird, did you manage to get it working with a SharedValue? Because that definitely works. I will also try to investigate the `runOnJS` issue now.

Also, I wonder if this works with Hermes 🤔

G0retZ, 2021-06-09:
@mrousavy yes, I tried with SharedValue also.
But for some reason despite of being set in the worklet it was not updated in the UI component and wasn't trigger...

So I just gave up on both of these =/

mrousavy, 2021-06-09:
I'll create an example app demonstrating SharedValues soon, but just to confirm - your issue is resolved for now and your frame processor works?

mrousavy, 2021-06-09:
Btw, I have just released [v2.3.0](https://github.com/cuvent/react-native-vision-camera/releases/tag/v2.3.0) which replaces the `CMSampleBufferRef` with a custom type (`Frame*`) to also provide information about orientation. This means, you need to change the parameter:

```diff
- static inline id somePlugin(CMSampleBufferRef buffer, NSArray* arguments) {
+ static inline id somePlugin(Frame* frame, NSArray* arguments) {
```

and to get the actual `CMSampleBufferRef` you can simply use `frame.buffer`

G0retZ, 2021-06-09:
For now I'm using a bit ugly workaround:
I created a native iOS module that  dispatches an event with payload to React Native side via `[self sendEventWithName:@"onQRCodesScanned" body:@{@"codes": codes}];`.
From the Frame processor after I got result I dispatch it using the module and listen for it on RN side.

It doesn't look like an elegant and easy to use solution but at least it works.
So far I don't think this solution is acceptable for a public `react-native-vision-camera` plugin.

mrousavy, 2021-06-09:
Alright, I'm moving this to a discussion since the QR code plugin you're building isn't finished yet - let's keep the conversation going on there.

closed 2021-06-09 by mrousavy