margelo/react-native-vision-camera #190: ❓ How to stop camera usage

G0retZ, 2021-06-09:
## Question

Currently I have and app utilizing react navigating.
When I start the app all is fine.
When I navigate to the screen with the `Camera` view, all is also fine.
And there is a green dot  above network state on iPhone, signaling that camera is in use.

But when I navigate back to the main screen (`CameraView` is gone) the green dot signal stays.
Which means the camera is still in use.

When I leave the app the indicator dot disappears.

But when I return to the app, the dot appears again. Despite there is no more `CameraView` on the screen.

So how to stop camera usage when `CameraView` is not on the screen?

My current usage code:

```tsx
  const isForeground = visible && appState === 'active';
...
      {device ? (
        <Camera
          ref={camera}
          style={StyleSheet.absoluteFill}
          device={device}
          torch={visible && flash ? 'on' : 'off'}
          isActive={isForeground}
          frameProcessor={Platform.select({
            ios: frameProcessor,
            default: undefined,
          })}
          frameProcessorFps={2}
          photo={isForeground}       // <--- added flag here to check if this hash effect - no effect
          video={isForeground}       // <--- added flag here to check if this hash effect - no effect
        />
      ) : null}
...
```

mrousavy, 2021-06-09:
1. You should only set `isActive` to `false`, `photo` and `video` should always stay true
2. Can you log `isForeground`? I just tested it using react-native-navigation and it works perfectly fine if I use a `isForeground` variable, did you read the ["Lifecycle" documentation](https://cuvent.github.io/react-native-vision-camera/docs/guides/lifecycle)?

mrousavy, 2021-06-09:
Seems weird to me that the green dot stays when you navigate back to the main screen, as that means that the Camera screen does not get destroyed correctly which seems to be wrong 🤔

G0retZ, 2021-06-09:
Thank for fast feedback!
So.
1. Ok
2. Yes, I debugged with `console.log` and the `isForgeround` value changes to `false` once the screen with camera goes off the screen. It is currently placed in a `PagerView` along with other screens...
 And yes, I read the Lifecycle docs about "camera pause" and I guess it is not the same as "camera stop".
 3. In my case `visible` is true if current `ViewPager` page index is `1` (the page with the camera view.
 4. I'm on 2.2.0 currently

G0retZ, 2021-06-09:
> Seems weird to me that the green dot stays when you navigate back to the main screen, as that means that the Camera screen does not get destroyed correctly which seems to be wrong 🤔

Hum 🤔
Is there an easy way to proof or disproof this? :sweat_smile:

mrousavy, 2021-06-09:
If I pass `isActive={false}` to my Camera, the Camera pauses/freezes and the green dot at the top disappears. I'm currently on react-native-vision-camera 2.1.0.

G0retZ, 2021-06-09:
Just checked again by force quit the app and relaunch.
No issue.
Looks like it was a chaotic glitch 😅

mrousavy, 2021-06-09:
Perfect! 👍  

Maybe it is a hot-reload/fast-refresh issue, if you can reproduce it again please create a bug report 🙏

closed 2021-06-09 by mrousavy

icodeface, 2023-05-09:
Got same issue on a normal View, any updates ?

amjadbouhouch, 2023-05-26:
@icodeface did you resolve the issue ? I'm facing this issue right now

icodeface, 2023-05-26:
> @icodeface did you resolve the issue ? I'm facing this issue right now

We found the issue was from our side, the Camara was not released correctly.

kavuri, 2023-10-01:
> > @icodeface did you resolve the issue ? I'm facing this issue right now
> 
> We found the issue was from our side, the Camara was not released correctly.

How to release the camera? In my case, I am navigating away to a different tab

mrousavy, 2023-10-03:
Either pass `false` to `isActive` or just unmount the Camera component.