margelo/react-native-vision-camera #195: 🐛 'folly/Optional.h' file not found on Xcode version 12.5

iwantooxxoox, 2021-06-10:
## What

Can't run on Xcode v12.5, it shows 'folly/Optional.h' file not found.
Not sure if it's relative to https://github.com/facebook/flipper/issues/2215
### Logs

[image]

## Reproducable sample

1. Create a new react native project
2. Install react-native-vision-camera 
3. Build and run on Xcode v12.5

## Environment

* Platform(s):
  * [V] iOS
  * [ ] Android
* react-native version: 0.63.4
* react-native-vision-camera version: 2.3.0
* react-native-reanimated version: N/A

mrousavy, 2021-06-10:
This might be related to the following line: https://github.com/cuvent/react-native-vision-camera/blob/26cf21ff5f79d98c37b80e4e488ae01d02248448/VisionCamera.podspec#L46

could you try removing it, running pod install, and then seeing if that fixes it for you? There might be some additional imports missing which you have to the Podfile, for example React-callinvoker.

LupulescuAlexandru, 2021-06-12:
I have the same problem as OP, and after removing that line I got this error:
[image]
I also followed the troubleshooting steps in the documentation.

mrousavy, 2021-07-09:
You could try editing `node_modules/react-native-vision-camera/VisionCamera.podspec`, and make the following change:
```diff
-  s.dependency "React"
+  s.dependency 'ReactCommon/turbomodule/core'
+  s.dependency 'Folly'
```

then in your own `Podfile` add 
```
use_modular_headers!
```
to the top. [Here](https://github.com/expo/expo/blob/42da468f1e1ad3e3f20bbde5f2b8fc4f794edb55/packages/expo-modules-core/ios/ExpoModulesCore.podspec)'s an example podspec that should work. Let me know if that fixes the issue for you because I cannot reproduce it.

axeldelafosse, 2021-07-13:
I was able to make it work but I had to set `"DEFINES_MODULE" => "YES"` in `ReactCommon.podspec` and `Folly.podspec` 

`react-native-vision-camera+2.4.2-beta.9.patch`

```
diff --git a/node_modules/react-native-vision-camera/VisionCamera.podspec b/node_modules/react-native-vision-camera/VisionCamera.podspec
index 8cfe1c6..6cddd88 100644
--- a/node_modules/react-native-vision-camera/VisionCamera.podspec
+++ b/node_modules/react-native-vision-camera/VisionCamera.podspec
@@ -2,6 +2,11 @@ require "json"
 
 package = JSON.parse(File.read(File.join(__dir__, "package.json")))
 
+folly_flags = '-DFOLLY_NO_CONFIG -DFOLLY_MOBILE=1 -DFOLLY_USE_LIBCPP=1'
+folly_compiler_flags = folly_flags + ' ' + '-Wno-comma -Wno-shorten-64-to-32'
+folly_version = '2020.01.13.00'
+boost_compiler_flags = '-Wno-documentation'
+
 Pod::Spec.new do |s|
   s.name         = "VisionCamera"
   s.version      = package["version"]
@@ -19,6 +24,12 @@ Pod::Spec.new do |s|
     "HEADER_SEARCH_PATHS" => "\"$(PODS_TARGET_SRCROOT)/ReactCommon\" \"$(PODS_TARGET_SRCROOT)\" \"$(PODS_ROOT)/Headers/Private/React-Core\" \"$(PODS_ROOT)/Headers/Public/React-hermes\" \"$(PODS_ROOT)/Headers/Public/hermes-engine\""
   }
   s.requires_arc = true
+  s.compiler_flags = folly_compiler_flags + ' ' + boost_compiler_flags
+  s.xcconfig = {
+    "CLANG_CXX_LANGUAGE_STANDARD" => "c++14",
+    "HEADER_SEARCH_PATHS" => "$(inherited) \"$(PODS_TARGET_SRCROOT)/ReactCommon\" \"$(PODS_ROOT)/boost-for-react-native\" \"$(PODS_ROOT)/glog\" \"$(PODS_ROOT)/Folly\" \"${PODS_ROOT}/Headers/Public/React-hermes\" \"${PODS_ROOT}/Headers/Public/hermes-engine\"",
+    "OTHER_CFLAGS" => "$(inherited)" + " " + folly_flags
+  }
 
   # All source files that should be publicly visible
   # Note how this does not include headers, since those can nameclash.
@@ -43,6 +54,7 @@ Pod::Spec.new do |s|
   ]
 
   s.dependency "React-callinvoker"
-  s.dependency "React"
   s.dependency "React-Core"
+  s.dependency "ReactCommon/turbomodule/core"
+  s.dependency "Folly", folly_version
 end
diff --git a/node_modules/react-native-vision-camera/ios/React Utils/JSIUtils.mm b/node_modules/react-native-vision-camera/ios/React Utils/JSIUtils.mm
index ccd0476..2c64c69 100644
--- a/node_modules/react-native-vision-camera/ios/React Utils/JSIUtils.mm	
+++ b/node_modules/react-native-vision-camera/ios/React Utils/JSIUtils.mm	
@@ -191,7 +191,7 @@ RCTResponseSenderBlock convertJSIFunctionToCallback(jsi::Runtime &runtime, const
     wrapperWasCalled = YES;
   };
 
-  if (RCTTurboModuleBlockCopyEnabled()) {
+  if (RCTTurboModuleEnabled()) {
     return [callback copy];
   }
 
```

`react-native+0.63.4.patch`

```
diff --git a/node_modules/react-native/ReactCommon/ReactCommon.podspec b/node_modules/react-native/ReactCommon/ReactCommon.podspec
index 765ec10..9499ba0 100644
--- a/node_modules/react-native/ReactCommon/ReactCommon.podspec
+++ b/node_modules/react-native/ReactCommon/ReactCommon.podspec
@@ -32,7 +32,8 @@ Pod::Spec.new do |s|
   s.source                 = source
   s.header_dir             = "ReactCommon" # Use global header_dir for all subspecs for use_frameworks! compatibility
   s.compiler_flags         = folly_compiler_flags + ' ' + boost_compiler_flags
-  s.pod_target_xcconfig    = { "HEADER_SEARCH_PATHS" => "\"$(PODS_ROOT)/boost-for-react-native\" \"$(PODS_ROOT)/Folly\" \"$(PODS_ROOT)/DoubleConversion\" \"$(PODS_ROOT)/Headers/Private/React-Core\"",
+  s.pod_target_xcconfig    = { "DEFINES_MODULE" => "YES",
+                               "HEADER_SEARCH_PATHS" => "\"$(PODS_ROOT)/boost-for-react-native\" \"$(PODS_ROOT)/Folly\" \"$(PODS_ROOT)/DoubleConversion\" \"$(PODS_ROOT)/Headers/Private/React-Core\"",
                                "USE_HEADERMAP" => "YES",
                                "CLANG_CXX_LANGUAGE_STANDARD" => "c++14" }
 
diff --git a/node_modules/react-native/scripts/.packager.env b/node_modules/react-native/scripts/.packager.env
new file mode 100644
index 0000000..361f5fb
--- /dev/null
+++ b/node_modules/react-native/scripts/.packager.env
@@ -0,0 +1 @@
+export RCT_METRO_PORT=8081
diff --git a/node_modules/react-native/third-party-podspecs/Folly.podspec b/node_modules/react-native/third-party-podspecs/Folly.podspec
index 35a0522..8d2706c 100644
--- a/node_modules/react-native/third-party-podspecs/Folly.podspec
+++ b/node_modules/react-native/third-party-podspecs/Folly.podspec
@@ -52,7 +52,8 @@ Pod::Spec.new do |spec|
                         'folly/net/detail/*.h',
                         'folly/portability/*.h'
   spec.libraries           = "stdc++"
-  spec.pod_target_xcconfig = { "USE_HEADERMAP" => "NO",
+  spec.pod_target_xcconfig = { "DEFINES_MODULE" => "YES",
+                               "USE_HEADERMAP" => "NO",
                                "CLANG_CXX_LANGUAGE_STANDARD" => "c++14",
                                "HEADER_SEARCH_PATHS" => "\"$(PODS_TARGET_SRCROOT)\" \"$(PODS_ROOT)/boost-for-react-native\" \"$(PODS_ROOT)/DoubleConversion\"" }
 
```

matheusmatos, 2021-07-13:
Playing around solution suggested by @axeldelafosse I was able to build it without touching RN codebase.

## To make work with RN 0.63

We need to do two changes in the module. First one in `VisionCamera.podspec`:

```diff
# filename: VisionCamera.podspec

+ folly_flags = '-DFOLLY_NO_CONFIG -DFOLLY_MOBILE=1 -DFOLLY_USE_LIBCPP=1'
+ folly_compiler_flags = folly_flags + ' ' + '-Wno-comma -Wno-shorten-64-to-32'
+ folly_version = '2020.01.13.00'
+ boost_compiler_flags = '-Wno-documentation'

...

s.requires_arc = true
+ s.compiler_flags = folly_compiler_flags + ' ' + boost_compiler_flags
+ s.xcconfig = {
+   "CLANG_CXX_LANGUAGE_STANDARD" => "c++14",
+   "HEADER_SEARCH_PATHS" => "$(inherited) \"$(PODS_TARGET_SRCROOT)/ReactCommon\" \"$(PODS_ROOT)/boost-for-react-native\" \"$(PODS_ROOT)/glog\" \"$(PODS_ROOT)/Folly\" \"${PODS_ROOT}/Headers/Public/React-hermes\" \"${PODS_ROOT}/Headers/Public/hermes-engine\"",
+   "OTHER_CFLAGS" => "$(inherited)" + " " + folly_flags
+ }

...

- s.dependency "React-callinvoker"
- s.dependency "React"
- s.dependency "React-Core"
+ s.dependency "React-callinvoker"
+ s.dependency "React-Core"
+ s.dependency "ReactCommon/turbomodule/core"
+ s.dependency "Folly", folly_version
```

Also this simple change in `ios/React Utils/JSIUtils.mm`:

```diff
- if (RCTTurboModuleBlockCopyEnabled()) {
+ if (RCTTurboModuleEnabled()) {
```

Then in your main Podfile:

```diff
- config = use_native_modules!
- use_react_native!(:path => config["reactNativePath"])
+ config = use_native_modules!
+ reactNativePath = config["reactNativePath"] ||= "../node_modules/react-native"
+ use_react_native!(:path => reactNativePath)

...

+ pod 'ReactCommon/turbomodule/core', :path => "#{reactNativePath}/ReactCommon", :modular_headers => true
+ pod 'Folly', :podspec => "#{reactNativePath}/third-party-podspecs/Folly.podspec", :modular_headers => true
```

P.S: The fixed Folly version is extracted from [RN0.63 release notes](https://github.com/react-native-community/releases/blob/master/CHANGELOG.md#ios-specific-12). I just kept it based on @axeldelafosse suggestion and [RNReanimated podscec](https://github.com/software-mansion/react-native-reanimated/blob/master/RNReanimated.podspec).

## To make work with RN 0.64 using changes above 👆🏻

Basically we just need to change `Folly` to `RCT-Folly`:

```diff
# filename: VisionCamera.podspec

- "HEADER_SEARCH_PATHS" => "$(inherited) \"$(PODS_TARGET_SRCROOT)/ReactCommon\" \"$(PODS_ROOT)/boost-for-react-native\" \"$(PODS_ROOT)/glog\" \"$(PODS_ROOT)/Folly\" \"${PODS_ROOT}/Headers/Public/React-hermes\" \"${PODS_ROOT}/Headers/Public/hermes-engine\"",
+ "HEADER_SEARCH_PATHS" => "$(inherited) \"$(PODS_TARGET_SRCROOT)/ReactCommon\" \"$(PODS_ROOT)/boost-for-react-native\" \"$(PODS_ROOT)/glog\" \"$(PODS_ROOT)/RCT-Folly\" \"${PODS_ROOT}/Headers/Public/React-hermes\" \"${PODS_ROOT}/Headers/Public/hermes-engine\"",

...

- s.dependency "Folly", folly_version
+ s.dependency "RCT-Folly", folly_version
```

Also in your main Podfile:
```diff
- pod 'Folly', :podspec => "#{reactNativePath}/third-party-podspecs/Folly.podspec", :modular_headers => true
+ pod 'RCT-Folly', :podspec => "#{reactNativePath}/third-party-podspecs/RCT-Folly.podspec", :modular_headers => true
```

👉🏻  ...**However** I believe that we don't have problem now with 0.64 version. I"ll update the PR #273 to make these changes only in 0.63 version. What you guys think?

PS: I don't understand why we need to switch from Folly to RTC-Folly as I didn't find anything in the release notes. But it works.

## To make work with Expo 42 as a Config Plugin in Managed Workflow

Expo 42 uses RN 0.63. Using `expo prebuild` I tested the solution above and it works. Now we just need to add a [mod](https://docs.expo.io/guides/config-plugins/#what-are-mods) to the `expo-plugin/withVisionCamera` to insert these two lines in main Podfile (after updating the RNVisionCamera module) - I"ll open a PR for that.

That's it! If anyone still has problems just share it!

✌️

mrousavy, 2021-07-14:
Wow, amazing research @matheusmatos and @axeldelafosse! 😲🤩

I'll research a bit more on what those Modular Headers are exactly and if there's any workaround because I can't sleep good at night if we patch the user's Podfile like that (adjusting RN pods), but if not we'll use those changes!

mrousavy, 2021-07-14:
Maybe this fixes things for the future: https://github.com/facebook/react-native/pull/31858

mrousavy, 2021-07-26:
So as far as I know, Expo 43 will come with React Native 0.64, which fixes all those build issues.
I think I will not merge your change @matheusmatos, so that VisionCamera only works on Expo 43 and above, just for the sake of simplicity (the `VisionCamera.podspec` would be very messy and users are required to edit their `Podfile` which is bad DX).
As for non-expo users, most of them are already on RN 0.64 anyways.
Thank you for providing a workaround people can use until then, as well as for your detailed research. :v:

mrousavy, 2021-07-27:
Just found a commit that fixes this issue (I believe) and it made it to RN 0.65! https://github.com/facebook/react-native/commit/8aea93022bad4ed57ff6b738528a43f26fc335ac

pke, 2021-11-10:
Gosh this is all so fragile in the build process :(

kranthi1216, 2024-02-21:
After updating React Native to 0.73.4, I am seeing the following: 'Pods/Headers/Public/RCT-Folly/folly/Optional'.h:474:36 Binary expression ('const bool' and 'const std::nullopt_t') has invalid operands.can help in resolving.