margelo/react-native-vision-camera #198: Not supported - use activeFormat.supportedColorSpaces'

maguirekrist, 2021-06-12:
## What
React Native Vision Camera crashes on IOS when running through physical device. The crash appears to manifest itself through the xCode debugger and a runtime error. The issue originated when I placed an Animated View outside of the Camera's container (which is a View). It sometimes works, but then randomly stops, within a few minutes of a new build. 

### Logs

```
2021-06-11 22:25:26.747003-0400 ReactNativeEhugs[1772:592497] *** Terminating app due to uncaught exception 'NSInvalidArgumentException', reason: '*** -[AVCaptureDevice setActiveColorSpace:] Not supported - use activeFormat.supportedColorSpaces'
*** First throw call stack:
(0x18def186c 0x1a2f0ac50 0x1a82661dc 0x12266d6c0 0x12267df14 0x1a8266050 0x18f21b890 0x18f1543a0 0x18f216830 0x1a8289090 0x1a8287444 0x1a8280ec0 0x1a8282b20 0x102bf8b34 0x102c10f18 0x102c12160 0x12266bbcc 0x12266d6c0 0x122675354 0x1226760c0 0x122682644 0x1d999d814 0x1d99a476c)
libc++abi.dylib: terminating with uncaught exception of type NSException
*** Terminating app due to uncaught exception 'NSInvalidArgumentException', reason: '*** -[AVCaptureDevice setActiveColorSpace:] Not supported - use activeFormat.supportedColorSpaces'
terminating with uncaught exception of type NSException
```

## Reproducable sample
This is pointless, but I'll link repo tomorrow. 

```tsx
// code here
```

## Environment

* Platform(s):
  * [ ] iOS
  * [ ] Android
* react-native version: latest
* react-native-vision-camera version: latest
* react-native-reanimated version: latest

mrousavy, 2021-06-12:
> This is pointless
It isn't, without a reproduction of the issue I cannot debug this.

Are you using the `colorSpace` prop?

mrousavy, 2021-06-12:
I have a blind guess - please try again with [2.4.1](https://github.com/cuvent/react-native-vision-camera/releases/tag/v2.4.1).

closed 2021-06-24 by mrousavy