margelo/react-native-vision-camera #210: 🐛 Preview stretched out when rotating to landscape (Android)

LupulescuAlexandru, 2021-06-16:
## What

The preview seems to get stretched out if rotated to landscape and auto-rotate is turned on. The problem does not appear if the device is locked on portrait, however the resulting image will be rotated.
Demo here: https://www.youtube.com/watch?v=0Dqsre4a5T4

## Reproducable sample

1. Git clone && run the [example](https://github.com/cuvent/react-native-vision-camera/tree/main/example) on an Android phone 
2. Make sure that auto-rotate is turned on
3. Rotate the phone to landscape

## Environment

* Device: Samsung Galaxy Note 8
* Platform(s):
  * [ ] iOS
  * [X] Android 9
* react-native version: 0.64.2
* react-native-vision-camera version: 2.4.1
* react-native-reanimated version: N/A

mrousavy, 2021-06-16:
I see, are you using a custom `format={...}`?

LupulescuAlexandru, 2021-06-16:
No, I didn't modify anything inside the example app.

mrousavy, 2021-06-16:
Example app uses a custom `format={...}`. I'll try to find out what's causing this, but I didn't notice it because our app doesn't support rotation.

LupulescuAlexandru, 2021-06-16:
Oh, you're right, sorry for not noticing. I tested it right now and the behavior is the same even without the format prop.

Kypsis, 2021-07-06:
Any news or progress on the rotation issue?

mrousavy, 2021-07-06:
I didn't continue investigating/debugging this because I was busy with working on Frame Processors for Android. I'll try to get to this soon, but if anyone else wants to contribute feel free to do so 🤗

Kypsis, 2021-07-06:
At the moment we are using the orientation change tied to setState + useEffect combo to conditionally render the camera view (basically quick mount/unmount) to reset the preview. This is definitely not ideal. Since our use case involves recording in landscape the portrait to landscape transition is quite common as people tend to press record in portrait before setting the phone in landscape and into a phone holder for instance. As for personally contributing I have the same issue as this gentleman (not familiar with coroutines): [https://github.com/mrousavy/react-native-vision-camera/issues/213#issuecomment-864224835](https://github.com/mrousavy/react-native-vision-camera/issues/213#issuecomment-864224835)

Also very excited for the frame processors on android as well. Very nice seeing someone pushing the RN community to new heights like this 😄 👍

closed 2021-07-26 by mrousavy

mrousavy, 2022-01-04:
Maybe check also if #715 changes anything for you guys?

PR #301 (2021-07-26): feat: Support rotation

## What

* Supports phone rotation for the Preview View
* Supports phone rotation for taking pictures
* Supports phone rotation for recording views
* Supports phone rotation for the Frame Processor

## Changes

## Tested on

## Related issues

* Fixes #287 
* Fixes #213 
* Fixes #210