margelo/react-native-vision-camera #211: ✨ Wide Angle Camera

farhanmeo, 2021-06-17:
## What

Currently the library works with IOS wide angle camera but unable to get the available cameras in Android So that would be great since all latest mobiles have built in multiple cameras 

## Alternatives/Workarounds

Currently there is no alternative for Android to use wide angle camera

mrousavy, 2021-06-17:
Do you mean the _ultra_ wide angle camera? You should be able to get that, but _multi-cameras_ (so smooth switch-over while zooming) is not supported on Android at the moment (limitation by the native android camera library "CameraX")
I have reported this an they're working on it, but I don't have an ETA.

mrousavy, 2023-08-18:
Hey! I've rewritten the entire Android codebase of VisionCamera from CameraX to Camera2 in the efforts of [✨ VisionCamera V3](https://github.com/mrousavy/react-native-vision-camera/issues/1376).

I just now completed the Camera2 rewrite and I believe the core structure is running, but there might be some edge cases to iron out. Can you try and test the PR #1674 for me to see if you can still reproduce this issue here?

Here's an instruction on how you can test that: https://github.com/mrousavy/react-native-vision-camera/pull/1674#issuecomment-1684104217

If the issue cannot be reproduced with that version/PR anymore, then hoorayy, I fixed it! 🎉 
Otherwise please let me know and I'll keep this issue open to keep track of it.

Thank you!

closed 2023-09-01 by mrousavy