margelo/react-native-vision-camera #214: Hi , Where is this vision-camera-qrcode-scanner plugin ❓

ballbrk, 2021-06-18:
## Question

In the doc, you've a plugin called vision-camera-qrcode-scanner but  when click on link go 404

mrousavy, 2021-06-18:
Oh right, sorry, I lied. That plugin doesn't exist yet. I wanted to create it when Frame Processors are available for Android, which is what I'm working on now. You might be interested in this discussion: https://github.com/cuvent/react-native-vision-camera/discussions/188

closed 2021-06-24 by mrousavy

rodgomesc, 2021-08-16:
the link is showing 404 for me too. I'm starting a longtime relationship with this lib, if this plugin doesn't exist I'll start working on it today

mrousavy, 2021-08-16:
Link above is dead because VisionCamera moved from cuvent/ to mrousavy/. New link to the discussion is: https://github.com/mrousavy/react-native-vision-camera/discussions/188.

The plugin does not yet exist, excited to help you get started creating it :)

rodgomesc, 2021-08-17:
I finished the qrcode scanner on android and it's extremely easy to implement, unfortunately the company I work for didn't allow it to be open source, as it will be part of an authentication module, but if anyone wants to start implementing it and have any difficulties i will be happy to help

I'm going to play with the face detector now

mrousavy, 2021-08-17:
> it's extremely easy to implement

Happy to hear my API is easy to use 😄

rodgomesc, 2021-08-17:
I had never done anything in java and even so it was very easy, this lib will be the future of camera api for react native, is there any chat that you are available, to talk without spamming the issues?

mrousavy, 2021-08-17:
@rodgomesc I could create a discord for VisionCamera (+ my other OSS libraries) 🤔

mrousavy, 2021-08-17:
otherwise there are github discussions in this repo

rodgomesc, 2021-08-17:
> I could create a discord for VisionCamera (+ my other OSS libraries)

it would be perfect