margelo/react-native-vision-camera #227: ❓ Disable default "wide-angle-camera" as a default

dkunin9, 2021-06-24:
First of all, thank you for such brilliant library 👍 🥇 

Just wanted to know how to set the default zoom value of camera to normal for both platforms? 🙂But still have an ability to zoom out if ` ["wide-angle-camera"]` exists on the device 

P.s. I read your [documentation page](https://mrousavy.github.io/react-native-vision-camera/docs/guides/devices) and looked to the example project but still can't figure out how to disable ` ["wide-angle-camera"]` as a default zoom once the camera is open.

Thank you in advance!

mrousavy, 2021-06-29:
Sorry what do you mean with "normal"? Maybe this discussion answers your question: https://github.com/mrousavy/react-native-vision-camera/discussions/226

dkunin9, 2021-06-29:
> Sorry what do you mean with "normal"? Maybe this discussion answers your question: #226

yes, sure!
thank you ✌️

closed 2021-06-29 by dkunin9