margelo/react-native-vision-camera #232: 🐛 Camera start and stop issues

G0retZ, 2021-06-28:
Looks like camera device is struggled to start and stop immediately

## What

We are building an app with React Navigation.
So each screen is actually a view, that is manipulated by the library on navigation.

I have found that if I "open a screen" with the CameraView being active by default, there are problems:
- the frame processor doesn't get called on iOS
- on some Android devices turning on flashlight leads to crash.

If I add a delay before making CameraView active like so:

```
  useEffect(() => {
    if (!visible) {
      setTrueVisible(false);
      return () => {};
    }
    const timeout = setTimeout(() => setTrueVisible(true), 20);
    return () => clearTimeout(timeout);
  }, [visible, setTrueVisible]);

```

And the same applies to the CameraView destroying.
When I navigate back, the camera stays active (can be seen by green dot indicator in iPhone).

But if I first make it inactive and then navigate back after a small delay - all is fine.

Tried to update the library to the lates version (2.4.1) - same result.

Is this delay mandatory or smith else is the source of the issue?

### Logs

The problem was found by lack of logs.
Actually there are no errors in the logs. Just actual misbehavior...
¯\_(ツ)_/¯

## Reproducable sample

Try to add this view and alter `visible` prop. Mb by button.

```tsx
export default ({ visible, flash, onScannedCodes }: Props) => {
  const camera = useRef<Camera>(null);
  const { status } = useCameraPermissions();
  const devices = useCameraDevices();
  const appState = useAppState();
  const device = useMemo(
    () => (status === PermissionStatus.allowed ? devices.back : undefined),
    [devices, status]
  );
  const isForeground = useMemo(() => visible && appState === 'active', [
    visible,
    appState,
  ]);

  useEffect(() => {
    const subscription = eventEmitter?.addListener(
      'onQRCodesScanned',
      (result) => onScannedCodes?.(result.codes)
    );
    return () => subscription?.remove();
  }, []);

  const frameProcessor = useFrameProcessor(
    (frame) => {
      'worklet';

      scanQRCodes(frame);
    },
    [scanQRCodes]
  );

  return device ? (
    <Camera
      ref={camera}
      style={StyleSheet.absoluteFill}
      device={device}
      torch={visible && flash ? 'on' : 'off'}
      isActive={isForeground}
      frameProcessor={frameProcessor}
      frameProcessorFps={2}
      photo
      video
    />
  ) : null;
};
```

## Environment

* Platform(s):
  * [V] iOS
  * [V] Android
* react-native version: 0.64.1
* react-native-vision-camera version: 2.2.0
* react-native-reanimated version: 2.2.0

G0retZ, 2021-06-28:
A released question I asked earlier !190

mrousavy, 2021-06-28:
1. This happens on both iOS and Android?
2. Can you log `isForeground`? Does it actually have `true` when you expect the camera to open?
3. Can you run the project in Xcode/Android Studio and share the native logs with me? Maybe there's something being logged there...
4. Does it work if you use the following code:
```tsx
  return device ? (
    <Camera
      ref={camera}
      style={StyleSheet.absoluteFill}
      device={device}
      isActive={true}
      frameProcessor={frameProcessor}
      frameProcessorFps={2}
      video={true}
    />
  ) : null;
```
?

G0retZ, 2021-06-28:
1. On Android looks like it works (no frame processor, just `takeSnapshot`s
2. Yes, I logged, it is true.
3. Logs are coming (the case without delay)
4. Same result unfortunately
5. Happens always despite of hot reload or cold start

G0retZ, 2021-06-28:
The iOS log right after navigating to the screen with the camera:

```log
2021-06-28 19:06:29.182929+0300 emma dev[2144:890642] [javascript] 'currentlyFocusedField is deprecated and will be removed in a future release. Use currentlyFocusedInput', '\nTransitioner@http://192.168.0.11:8081/index.bundle?platform=ios&dev=true&minify=false&modulesOnly=false&runModule=true&app=com.emma-app.dev:136882:36\nStackView@http://192.168.0.11:8081/index.bundle?platform=ios&dev=true&minify=false&modulesOnly=false&runModule=true&app=com.emma-app.dev:133740:36\nNavigator@http://192.168.0.11:8081/index.bundle?platform=ios&dev=true&minify=false&modulesOnly=false&runModule=true&app=com.emma-app.dev:133039:38\nKeyboardAwareNavigator@http://192.168.0.11:8081/index.bundle?platform=ios&dev=true&minify=false&modulesOnly=false&runModule=true&app=com.emma-app.dev:133153:38\nNavigationContainer@http://192.168.0.11:8081/index.bundle?platform=ios&dev=true&minify=false&modulesOnly=false&runModule=true&app=com.emma-app.dev:130856:38\nRCTView\nView\nErrorBoundaries@http://192.168.0.11:8081/index.bundle?platform=ios&dev=true&minify=false&modulesOnly=false&runModule=true&app=com.emma-app.dev:463503:36\nNavigator@http://192.168.0.11:8081/index.bundle?platform=ios&dev=true&minify=false&modulesOnly=false&runModule=true&app=com.emma-app.dev:461476:36\nConnect(Navigator)@http://192.168.0.11:8081/index.bundle?platform=ios&dev=true&minify=false&modulesOnly=false&runModule=true&app=com.emma-app.dev:120098:43\nRCTView\nView\nApp@http://192.168.0.11:8081/index.bundle?platform=ios&dev=true&minify=false&modulesOnly=false&runModule=true&app=com.emma-app.dev:124366:36\nConnect(App)@http://192.168.0.11:8081/index.bundle?platform=ios&dev=true&minify=false&modulesOnly=false&runModule=true&app=com.emma-app.dev:120098:43\nRNCSafeAreaProvider\nSafeAreaProvider@http://192.168.0.11:8081/index.bundle?platform=ios&dev=true&minify=false&modulesOnly=false&runModule=true&app=com.emma-app.dev:123005:24\nProvider@http://192.168.0.11:8081/index.bundle?platform=ios&dev=true&minify=false&modulesOnly=false&runModule=true&app=com.emma-app.dev:119675:21\nRoot@http://192.168.0.11:8081/index.bundle?platform=ios&dev=true&minify=false&modulesOnly=false&runModule=true&app=com.emma-app.dev:119477:36\nRCTView\nView\nRCTView\nView\nAppContainer@http://192.168.0.11:8081/index.bundle?platform=ios&dev=true&minify=false&modulesOnly=false&runModule=true&app=com.emma-app.dev:69137:36'
2021-06-28 19:06:29.344607+0300 emma dev[2144:890471] [native] VisionCamera.didSetProps(_:): Updating 13 prop(s)...
2021-06-28 19:06:29.344777+0300 emma dev[2144:891916] [native] VisionCamera.configureCaptureSession(): Configuring Session...
2021-06-28 19:06:29.344805+0300 emma dev[2144:891916] [native] VisionCamera.configureCaptureSession(): Initializing Camera with device com.apple.avfoundation.avcapturedevice.built-in_video:6...
2021-06-28 19:06:29.344843+0300 emma dev[2144:891916] [native] VisionCamera.configureCaptureSession(): Adding Video input...
2021-06-28 19:06:29.349154+0300 emma dev[2144:891916] [native] VisionCamera.configureCaptureSession(): Adding Photo output...
2021-06-28 19:06:29.351966+0300 emma dev[2144:891916] [native] VisionCamera.configureCaptureSession(): Adding Video Data output...
2021-06-28 19:06:29.352410+0300 emma dev[2144:891916] [native] VisionCamera.invokeOnInitialized(): Camera initialized!
2021-06-28 19:06:29.352472+0300 emma dev[2144:891916] [native] VisionCamera.configureCaptureSession(): Session successfully configured!
2021-06-28 19:06:29.352950+0300 emma dev[2144:891916] [native] VisionCamera.configureFormat(): Configuring Format...
2021-06-28 19:06:29.352956+0300 emma dev[2144:891916] [native] VisionCamera.configureDevice(): Configuring Device...
2021-06-28 19:06:29.352986+0300 emma dev[2144:891916] [native] VisionCamera.configureDevice(): Device successfully configured!
2021-06-28 19:06:29.353137+0300 emma dev[2144:891916] [native] VisionCamera.didSetProps(_:): Starting Session...
2021-06-28 19:06:29.975627+0300 emma dev[2144:891916] [native] VisionCamera.didSetProps(_:): Started Session!
```

G0retZ, 2021-06-28:
Looks like it just skipped the frame processor adding?

G0retZ, 2021-06-28:
This is Android logs for the same case:

```log
2021-06-28 19:09:09.627 929-929/? W/adbd: timeout expired while flushing socket, closing
2021-06-28 19:09:10.069 17592-20176/com.emmaprod E/ReactNativeJS: 'currentlyFocusedField is deprecated and will be removed in a future release. Use currentlyFocusedInput', '\nTransitioner@http://localhost:8081/index.android.bundle?platform=android&dev=true&minify=false&app=com.emmaprod&modulesOnly=false&runModule=true:135346:36\nStackView@http://localhost:8081/index.android.bundle?platform=android&dev=true&minify=false&app=com.emmaprod&modulesOnly=false&runModule=true:132204:36\nNavigator@http://localhost:8081/index.android.bundle?platform=android&dev=true&minify=false&app=com.emmaprod&modulesOnly=false&runModule=true:131503:38\nKeyboardAwareNavigator@http://localhost:8081/index.android.bundle?platform=android&dev=true&minify=false&app=com.emmaprod&modulesOnly=false&runModule=true:131617:38\nNavigationContainer@http://localhost:8081/index.android.bundle?platform=android&dev=true&minify=false&app=com.emmaprod&modulesOnly=false&runModule=true:129320:38\nRCTView\nView\nErrorBoundaries@http://localhost:8081/index.android.bundle?platform=android&dev=true&minify=false&app=com.emmaprod&modulesOnly=false&runModule=true:460506:36\nNavigator@http://localhost:8081/index.android.bundle?platform=android&dev=true&minify=false&app=com.emmaprod&modulesOnly=false&runModule=true:458504:36\nConnect(Navigator)@http://localhost:8081/index.android.bundle?platform=android&dev=true&minify=false&app=com.emmaprod&modulesOnly=false&runModule=true:118562:43\nRCTView\nView\nApp@http://localhost:8081/index.android.bundle?platform=android&dev=true&minify=false&app=com.emmaprod&modulesOnly=false&runModule=true:122830:36\nConnect(App)@http://localhost:8081/index.android.bundle?platform=android&dev=true&minify=false&app=com.emmaprod&modulesOnly=false&runModule=true:118562:43\nRNCSafeAreaProvider\nSafeAreaProvider@http://localhost:8081/index.android.bundle?platform=android&dev=true&minify=false&app=com.emmaprod&modulesOnly=false&runModule=true:121469:24\nProvider@http://localhost:8081/index.android.bundle?platform=android&dev=true&minify=false&app=com.emmaprod&modulesOnly=false&runModule=true:118139:21\nRoot@http://localhost:8081/index.android.bundle?platform=android&dev=true&minify=false&app=com.emmaprod&modulesOnly=false&runModule=true:117941:36\nRCTView\nView\nRCTView\nView\nAppContainer@http://localhost:8081/index.android.bundle?platform=android&dev=true&minify=false&app=com.emmaprod&modulesOnly=false&runModule=true:68170:36'
2021-06-28 19:09:10.111 17592-17592/com.emmaprod E/libc: Access denied finding property "vendor.camera.aux.packagelist"
2021-06-28 19:09:10.104 17592-17592/com.emmaprod W/com.emmaprod: type=1400 audit(0.0:21979): avc: denied { read } for name="u:object_r:persist_camera_prop:s0" dev="tmpfs" ino=2223 scontext=u:r:untrusted_app:s0:c232,c256,c512,c768 tcontext=u:object_r:persist_camera_prop:s0 tclass=file permissive=0
2021-06-28 19:09:10.344 17592-17592/com.emmaprod W/CameraView: CameraViewModule::getAvailableCameraDevices took: 240 ms
2021-06-28 19:09:11.054 17592-17592/com.emmaprod W/CameraX-core_ca: type=1400 audit(0.0:21980): avc: denied { read } for name="u:object_r:persist_camera_prop:s0" dev="tmpfs" ino=2223 scontext=u:r:untrusted_app:s0:c232,c256,c512,c768 tcontext=u:object_r:persist_camera_prop:s0 tclass=file permissive=0
2021-06-28 19:09:11.068 17592-20629/com.emmaprod E/libc: Access denied finding property "persist.vendor.camera.privapp.list"
2021-06-28 19:09:11.070 953-24003/? E/CameraService: tom han set sys.camera 000 ....
2021-06-28 19:09:11.114 17592-17592/com.emmaprod W/Binder:17592_7: type=1400 audit(0.0:21981): avc: denied { read } for name="u:object_r:persist_camera_prop:s0" dev="tmpfs" ino=2223 scontext=u:r:untrusted_app:s0:c232,c256,c512,c768 tcontext=u:object_r:persist_camera_prop:s0 tclass=file permissive=0
2021-06-28 19:09:11.120 17592-24079/com.emmaprod E/libc: Access denied finding property "vendor.camera.aux.packagelist"
2021-06-28 19:09:11.121 17592-24079/com.emmaprod E/libc: Access denied finding property "vendor.camera.aux.packagelist"
2021-06-28 19:09:11.114 17592-17592/com.emmaprod W/Binder:17592_7: type=1400 audit(0.0:21982): avc: denied { read } for name="u:object_r:persist_camera_prop:s0" dev="tmpfs" ino=2223 scontext=u:r:untrusted_app:s0:c232,c256,c512,c768 tcontext=u:object_r:persist_camera_prop:s0 tclass=file permissive=0
2021-06-28 19:09:11.511 17592-20075/com.emmaprod W/OpenGLRenderer: Surface doesn't have any previously queued frames, nothing to readback from
2021-06-28 19:09:11.527 17592-17592/com.emmaprod E/SurfaceViewImpl: PreviewView.SurfaceViewImplementation.getBitmap() failed with error 3
2021-06-28 19:09:11.884 753-753/? W/provider@2.4-se: type=1400 audit(0.0:21983): avc: denied { read } for name="u:object_r:default_prop:s0" dev="tmpfs" ino=2152 scontext=u:r:hal_camera_default:s0 tcontext=u:object_r:default_prop:s0 tclass=file permissive=0
2021-06-28 19:09:11.890 753-753/? E/libc: Access denied finding property "persist.wa.log.level"
2021-06-28 19:09:11.894 753-753/? E/walog/SCamera: module name: defualt -> HMD_M582
2021-06-28 19:09:11.895 753-753/? E/libc: Access denied finding property "persist.wa.log.level"
2021-06-28 19:09:11.895 753-753/? E/libc: Access denied finding property "persist.wa.log.level"
[... cut ...] o ML Kit.
2021-06-28 19:09:23.875 3623-3623/? E/bal.app.midtes: Not starting debugger since process cannot load the jdwp agent.
2021-06-28 19:09:24.265 3623-3623/? W/ContextImpl: Calling a method in the system process without a qualified user: android.app.ContextImpl.startService:1570 android.content.ContextWrapper.startService:669 com.longcheertel.fmradio.FMAdapterApp.onCreate:81 android.app.Instrumentation.callApplicationOnCreate:1190 android.app.ActivityThread.handleBindApplication:6479 

```

G0retZ, 2021-06-28:
And Android camera component looks like this:

```tsx
import React, { useMemo, useRef } from 'react';
import { StyleSheet, NativeModules } from 'react-native';
import { Camera, useCameraDevices } from 'react-native-vision-camera';

import useAppState from '../../../hooks/useAppState';
import { useCameraPermissions } from '../hooks';
import { PermissionStatus } from '../types';
import useInterval from '../../../hooks/useInterval';

const qrcodeLocalImage = NativeModules.QRCodeLocalImage;

type Props = {
  visible: boolean;
  flash: boolean;
  onScannedCodes?: (codes: string[]) => void;
};

export default ({ visible, flash, onScannedCodes }: Props) => {
  const camera = useRef<Camera>(null);
  const { status } = useCameraPermissions();
  const devices = useCameraDevices();
  const appState = useAppState();
  const device = useMemo(
    () => (status === PermissionStatus.allowed ? devices.back : undefined),
    [devices, status]
  );
  const isForeground = useMemo(() => visible && appState === 'active', [
    visible,
    appState,
  ]);

  useInterval(
    async () => {
      if (camera.current && visible) {
        try {
          const photo = await camera.current?.takeSnapshot({
            quality: 80,
            skipMetadata: true,
          });

          if (photo && qrcodeLocalImage) {
            onScannedCodes?.(await qrcodeLocalImage.decode(photo.path));
          }
        } catch (e) {
          // NoOp
        }
      }
    },
    isForeground && device ? 500 : undefined
  );

  return device ? (
    <Camera
      ref={camera}
      style={StyleSheet.absoluteFill}
      device={device}
      torch={isForeground && flash ? 'on' : 'off'}
      isActive={isForeground}
      photo
    />
  ) : null;
};
```

mrousavy, 2021-06-28:
Oh okay, so the issue is that the frame processor is not set? 

The title of the issue is a bit misleading. "Camera start and stop issues" sounds like the camera blackscreens.

Anyways, I think I already found a fix. I haven't tested the change, but here's the diff: https://github.com/mrousavy/react-native-vision-camera/commit/b10b2c10fc6084f13a3a667b970910c0d7541b34

I have released this (and Frame Processors for Android! 🎉) under the `beta` tag on npm (see [Release 2.4.2-beta.3](https://github.com/mrousavy/react-native-vision-camera/releases/tag/v2.4.2-beta.2)).

```
$ npm i react-native-vision-camera@beta
```

G0retZ, 2021-06-28:
But what about camera not stopping on view unmount?
I can record a video of this...

G0retZ, 2021-06-28:
Will be happy to get Android Frame Processor also :-)🎉

mrousavy, 2021-06-28:
> I can record a video of this...
Please do, because I cannot reproduce it in the example app.

Add the following code to your page:

```tsx
useEffect(() => {
  console.log("mounted")
  return () => {
    console.log("unmounted!")
  }
}, [])
```

, then start the app clean (no fast-refresh) and showcase the issue

G0retZ, 2021-06-28:
@mrousavy so here it the record
https://youtube.com/shorts/Yzv0afk7aJ4?feature=share

The green indicator dot on the top right of the screen stays after going back. And never goes off.
If I send app to background it disappears.
But appears back once app gets resumed.

Here are the logs:

```
 LOG  mounted
 WARN  [native] VisionCamera.captureOutput(_:didDrop:from:): Dropped a Frame. This might indicate that your Frame Processor is doing too much work. Either throttle the frame processor's frame rate, or optimize your frame processor's execution speed. Frame drop reason: FrameWasLate
 LOG  unmounted!
```

Visually unmounted log happened after the view was animated to go offscreen.

G0retZ, 2021-06-28:
Also will try to update the lib, then will report on it's behavior

mrousavy, 2021-06-28:
Hm, that's really weird. Maybe there's a memory leak? Are you using react-navigation or react-native-navigation?

G0retZ, 2021-06-28:
We are using react-navigation

G0retZ, 2021-06-29:
Tried with the latest `2.4.2-beta.7`
The issue is still here.
@mrousavy  You mentioned something about the leak.
How there can be a leak if a component is unmounted? 🤔

G0retZ, 2021-06-29:
Also, looks like this fix actually doesn't help #b10b2c1
I need a delay anyway to let frame processor work.
Actually it looks like frame processor gets stuck after 1 run 🤔
And start to work normally only after disabling CameraView and enabling again.

G0retZ, 2021-06-29:
View with CameraView mounted and before disabled (Frame processor called only once):

```log
2021-06-29 15:27:02.643754+0300 emma dev[2737:1147307] FrameProcessorBindings: Setting new frame processor...
2021-06-29 15:27:02.643803+0300 emma dev[2737:1147307] FrameProcessorBindings: Adapting Shareable value from function (conversion to worklet)...
2021-06-29 15:27:02.643914+0300 emma dev[2737:1147307] FrameProcessorBindings: Successfully created worklet!
2021-06-29 15:27:02.654980+0300 emma dev[2737:1147299] FrameProcessorBindings: Converting worklet to Objective-C callback...
2021-06-29 15:27:02.655318+0300 emma dev[2737:1147299] FrameProcessorBindings: Frame processor set!
2021-06-29 15:27:02.657990+0300 emma dev[2737:1147134] [native] VisionCamera.didSetProps(_:): Updating 13 prop(s)...
2021-06-29 15:27:02.658195+0300 emma dev[2737:1147551] [native] VisionCamera.configureCaptureSession(): Configuring Session...
2021-06-29 15:27:02.658223+0300 emma dev[2737:1147551] [native] VisionCamera.configureCaptureSession(): Initializing Camera with device com.apple.avfoundation.avcapturedevice.built-in_video:6...
2021-06-29 15:27:02.658256+0300 emma dev[2737:1147551] [native] VisionCamera.configureCaptureSession(): Adding Video input...
2021-06-29 15:27:02.662385+0300 emma dev[2737:1147551] [native] VisionCamera.configureCaptureSession(): Adding Photo output...
2021-06-29 15:27:02.663920+0300 emma dev[2737:1147551] [native] VisionCamera.configureCaptureSession(): Adding Video Data output...
2021-06-29 15:27:02.664479+0300 emma dev[2737:1147551] [native] VisionCamera.invokeOnInitialized(): Camera initialized!
2021-06-29 15:27:02.664535+0300 emma dev[2737:1147551] [native] VisionCamera.configureCaptureSession(): Session successfully configured!
2021-06-29 15:27:02.665129+0300 emma dev[2737:1147551] [native] VisionCamera.configureFormat(): Configuring Format...
2021-06-29 15:27:02.665137+0300 emma dev[2737:1147551] [native] VisionCamera.configureDevice(): Configuring Device...
2021-06-29 15:27:02.665144+0300 emma dev[2737:1147551] [native] VisionCamera.configureDevice(): Device successfully configured!
2021-06-29 15:27:02.665277+0300 emma dev[2737:1147551] [native] VisionCamera.didSetProps(_:): Starting Session...
2021-06-29 15:27:03.259933+0300 emma dev[2737:1147551] [native] VisionCamera.didSetProps(_:): Started Session!
2021-06-29 15:27:10.769634+0300 emma dev[2737:1147307] FrameProcessorBindings: Setting new frame processor...
2021-06-29 15:27:10.769747+0300 emma dev[2737:1147307] FrameProcessorBindings: Adapting Shareable value from function (conversion to worklet)...
2021-06-29 15:27:10.769829+0300 emma dev[2737:1147307] FrameProcessorBindings: Successfully created worklet!
2021-06-29 15:27:10.769945+0300 emma dev[2737:1147296] FrameProcessorBindings: Converting worklet to Objective-C callback...
2021-06-29 15:27:10.770171+0300 emma dev[2737:1147296] FrameProcessorBindings: Frame processor set!
2021-06-29 15:27:10.794642+0300 emma dev[2737:1147296] Frame processor called!
2021-06-29 15:27:10.795332+0300 emma dev[2737:1147296] Metal API Validation Enabled
2021-06-29 15:27:10.856776+0300 emma dev[2737:1147134] [native] VisionCamera.didSetProps(_:): Updating 1 prop(s)...
2021-06-29 15:27:10.857034+0300 emma dev[2737:1147554] [native] VisionCamera.didSetProps(_:): Stopping Session...
2021-06-29 15:27:10.985358+0300 emma dev[2737:1147554] [native] VisionCamera.didSetProps(_:): Stopped Session!
```

G0retZ, 2021-06-29:
Logs after CameraView reenabled. Frame processor gets called constantly:
```
2021-06-29 15:27:15.703789+0300 emma dev[2737:1147134] [native] VisionCamera.didSetProps(_:): Updating 1 prop(s)...
2021-06-29 15:27:15.704219+0300 emma dev[2737:1147296] [native] VisionCamera.didSetProps(_:): Starting Session...
2021-06-29 15:27:16.259640+0300 emma dev[2737:1147296] [native] VisionCamera.didSetProps(_:): Started Session!
2021-06-29 15:27:16.261332+0300 emma dev[2737:1147296] Frame processor called!
2021-06-29 15:27:16.782095+0300 emma dev[2737:1147556] Frame processor called!
2021-06-29 15:27:17.294113+0300 emma dev[2737:1147556] Frame processor called!
2021-06-29 15:27:17.825266+0300 emma dev[2737:1147296] Frame processor called!
2021-06-29 15:27:18.330277+0300 emma dev[2737:1147296] Frame processor called!
2021-06-29 15:27:18.857436+0300 emma dev[2737:1147296] Frame processor called!
2021-06-29 15:27:19.361439+0300 emma dev[2737:1147296] Frame processor called!
2021-06-29 15:27:19.893778+0300 emma dev[2737:1147624] Frame processor called!
2021-06-29 15:27:20.426882+0300 emma dev[2737:1147300] Frame processor called!
2021-06-29 15:27:20.529367+0300 emma dev[2737:1147134] [native] VisionCamera.didSetProps(_:): Updating 1 prop(s)...
2021-06-29 15:27:20.529628+0300 emma dev[2737:1147300] [native] VisionCamera.didSetProps(_:): Stopping Session...
2021-06-29 15:27:20.695193+0300 emma dev[2737:1147300] [native] VisionCamera.didSetProps(_:): Stopped Session!
2021-06-29 15:27:22.628997+0300 emma dev[2737:1147307] FrameProcessorBindings: Removing frame processor...
2021-06-29 15:27:22.629324+0300 emma dev[2737:1147134] FrameProcessorBindings: Frame processor removed!
```

mrousavy, 2021-06-29:
Can you reproduce this in the example app?

G0retZ, 2021-06-29:
Trying to build this one https://github.com/mrousavy/vision-camera-image-labeler/tree/main/example
It fails...
But what I see so far, there is no case where CameraView gets unmounted while app is on the screen.
Once build, will add this case and check if it reproduces...

G0retZ, 2021-06-29:
Analyzing the difference in logs between first deactivate than unmount and unmount while was active.
The case with the issue lacks these logs:
```log
[2861:1181594] [native] VisionCamera.didSetProps(_:): Stopping Session...
[2861:1181594] [native] VisionCamera.didSetProps(_:): Stopped Session!
```

G0retZ, 2021-06-30:
@mrousavy thank you a lot with your help!
I finally managed to make example app build and run (it's a bit outdated and not fully set), replicated the case with mount/unmount and saw no issue with camera hang I reported. Looks like we have a leak somewhere...
I altered the example code a bit to test:
```tsx
import * as React from 'react';
import { TouchableOpacity, Text } from 'react-native';

import { StyleSheet, View, ActivityIndicator } from 'react-native';
import {
  useCameraDevices,
  useFrameProcessor,
} from 'react-native-vision-camera';
import { Camera } from 'react-native-vision-camera';
import { labelImage } from 'vision-camera-image-labeler';

export default function App() {
  const [hasPermission, setHasPermission] = React.useState(false);
  const [active, setActive] = React.useState(true);
  const [mount, setMount] = React.useState(true);
  const devices = useCameraDevices('wide-angle-camera');
  const device = devices.back;

  React.useEffect(() => {
    (async () => {
      const status = await Camera.requestCameraPermission();
      setHasPermission(status === 'authorized');
    })();
  }, []);

  const frameProcessor = useFrameProcessor((frame) => {
    const labels = labelImage(frame);
    _log(`Labels: ${labels.join(', ')}`);
  }, []);

  return (
    <View style={styles.container}>
      {device != null && mount && hasPermission ? (
        <Camera
          style={StyleSheet.absoluteFill}
          device={device}
          isActive={active}
          frameProcessor={frameProcessor}
          frameProcessorFps={5}
        />
      ) : (
        <ActivityIndicator size="large" color="white" />
      )}
      <TouchableOpacity
        style={styles.button}
        onPress={() => {
          setActive(!active);
        }}
      >
        <Text style={{ fontSize: 24 }}>
          {active ? 'Deactivate' : 'Activate'}
        </Text>
      </TouchableOpacity>
      <TouchableOpacity
        style={styles.button}
        onPress={() => {
          setMount(!mount);
        }}
      >
        <Text style={{ fontSize: 24 }}>{mount ? 'Unmount' : 'Mount'}</Text>
      </TouchableOpacity>
    </View>
  );
}

const styles = StyleSheet.create({
  container: {
    flex: 1,
    alignItems: 'center',
    justifyContent: 'center',
  },
  button: {
    padding: 24,
    margin: 24,
    backgroundColor: 'rgba(255,255,255,0.7)',
  },
  box: {
    width: 60,
    height: 60,
    marginVertical: 20,
  },
});
```

mrousavy, 2021-07-08:
Hey @G0retZ! Can you check if https://github.com/mrousavy/react-native-vision-camera/pull/265 fixes the issue for you?

closed 2021-07-12 by mrousavy

vzhovnitsky, 2022-01-20:
Having same issue with "react-native-vision-camera": "^2.12.0",
Using it with react-navigation. Green dot stays even after I close the screen with Camera

```javascript
const isFocused = useIsFocused();
const devices = useCameraDevices();
const device = devices.back;
const [frameProcessor, barcodes] = useScanBarcodes([BarcodeFormat.QR_CODE]);

{device && (
                <Camera
                    style={StyleSheet.absoluteFill}
                    device={device}
                    isActive={isFocused}
                    frameProcessor={frameProcessor}
                    frameProcessorFps={5}
                    torch={flashOn ? 'on' : 'off'}
                />
)}
```

arthwood, 2022-04-01:
I have the same issue but Android only. The camera stays active as soon as I visit this screen and persists through whole app session until it's moved to background:

```typescript
import { isDefined, Nullable } from '@core'
import { useIsFocused } from '@react-navigation/native'
import React, { FC, useEffect, useMemo, useRef, useState } from 'react'
import { Text, View } from 'react-native'
import { Camera, CameraDevice, CameraPermissionStatus, useCameraDevices } from 'react-native-vision-camera'

interface CameraState {
  cameraDevice?: CameraDevice
  cameraPermissionStatus?: CameraPermissionStatus
}

interface CameraReadyState extends CameraState {
  cameraDevice: CameraDevice
  cameraPermissionStatus: 'authorized'
}

const isCameraStateReady = (cameraState: CameraState): cameraState is CameraReadyState =>
  isDefined(cameraState.cameraDevice) && cameraState.cameraPermissionStatus === 'authorized'

const PictureScreen: FC = () => {
  const isFocused = useIsFocused()
  const cameraRef = useRef<Nullable<Camera>>(null)
  const { back: cameraDevice } = useCameraDevices()
  const [cameraPermissionStatus, setCameraPermissionStatus] = useState<CameraPermissionStatus>()

  const cameraState = useMemo<CameraState>(() => ({ cameraDevice, cameraPermissionStatus }), [
    cameraDevice,
    cameraPermissionStatus,
  ])
  const isCameraAuthorized = cameraPermissionStatus === 'authorized'

  useEffect(() => {
    Camera.getCameraPermissionStatus().then(setCameraPermissionStatus)
  }, [setCameraPermissionStatus])

  useEffect(() => {
    if (!isCameraAuthorized) Camera.requestCameraPermission().then(setCameraPermissionStatus)
  }, [isCameraAuthorized, setCameraPermissionStatus])

  return (
    <View>
      {isCameraStateReady(cameraState) ? (
        <Camera audio={false} device={cameraState.cameraDevice} isActive={isFocused} photo ref={cameraRef} video />
      ) : (
        <View>
          <Text>Loading camera...</Text>
        </View>
      )}
    </View>
  )
}
```

`isFocused` is propertly changing from `true` to `false` when navigating from this screen and `isCameraStateReady(cameraState)` is still `true`.

mrousavy, 2023-11-18:
Hey! I just found out that I really forgot to close and dispose the locked Camera resources, so I just fixed that in this PR: https://github.com/mrousavy/react-native-vision-camera/pull/2174 
Now the Camera fully & synchronously closes all resources (CameraSession, CameraDevice, OpenGL context, Video & Photo outputs, Photo Synchronizer) once the view gets removed from the React view hierarchy ("unmounted"), and things like Flash, Torch, NFC, and other Camera components should be working again.

There is still a small issue that causes once Camera component to turn into a blackscreen when navigating back and forth between two Camera components, that's a pretty rare edge case but I will still try to fix that soon when I have some free time.

If you appreciate my time, expertise and dedication to this project, pleas [💖 consider sponsoring me on GitHub 💖](https://github.com/sponsors/mrousavy) to support the development of this project.