margelo/react-native-vision-camera #237: 🐛  audio flag is ignored when recording video with Android device

Kypsis, 2021-07-06:
## What

The audio flag seems to have no effect. It will record audio regardless if it's true or false on Pixel 4a.

## Reproducable sample

react-native-vision-camera example code.

mrousavy, 2021-07-06:
Right, this is a limitation of CameraX. I have created an issue in the CameraX issue tracker and the Google team is working on this. Currently, audio will simply not be recorded when either the permission has not been granted, or the audio session failed to set up (e.g. while on a call or any other error)

Kypsis, 2021-07-06:
I did try playing with microphone permissions as workaround but it actually red errored when I revoked microphone permission and crashed the camera.

mrousavy, 2021-07-07:
can you share the _native_ logs (Android Studio/Logcat) for when it crashed?

Kypsis, 2021-07-07:
```
2021-07-07 12:34:09.478 32687-32728/com.mrousavy.camera.example D/ReactNativeJS: state: BEGAN
2021-07-07 12:34:09.690 32687-32728/com.mrousavy.camera.example I/ReactNativeJS: calling startRecording()...
2021-07-07 12:34:09.692 32687-32728/com.mrousavy.camera.example I/ReactNativeJS: called startRecording()!
2021-07-07 12:34:09.697 32687-32753/com.mrousavy.camera.example D/Camera2CameraImpl: {Camera@3fb9680[id=0]} Issue capture request
2021-07-07 12:34:09.697 32687-32753/com.mrousavy.camera.example D/CaptureSession: Issuing capture request.
2021-07-07 12:34:09.701 32687-32753/com.mrousavy.camera.example D/UseCaseAttachState: Active and attached use case: [androidx.camera.core.Preview-0b672842-e184-45c2-b716-7fc9f7a777aa267873987, androidx.camera.core.VideoCapture-faf31d9d-9edb-4548-b576-5faf5ba21b04233554752, androidx.camera.core.ImageCapture-d8b3846a-7efe-44f0-87c1-4f67d2ecdb1b106943097] for camera: 0
2021-07-07 12:34:09.704 32687-32753/com.mrousavy.camera.example D/CaptureSession: Attempting to submit CaptureRequest after setting
2021-07-07 12:34:09.705 32687-32753/com.mrousavy.camera.example D/CaptureSession: Issuing request for session.
2021-07-07 12:34:09.712 32687-32687/com.mrousavy.camera.example I/VideoCapture: startRecording
2021-07-07 12:34:09.749 32687-32728/com.mrousavy.camera.example E/ReactNativeJS: 'Recording failed!', { userInfo: null,
      message: 'An unknown error occurred while trying to start a video recording!',
      cause: 
       { stacktrace: 'java.lang.NullPointerException: Attempt to invoke virtual method \'void android.media.AudioRecord.startRecording()\' on a null object reference\n\tat androidx.camera.core.VideoCapture.startRecording(VideoCapture.java:389)\n\tat com.mrousavy.camera.CameraView_RecordVideoKt.startRecording(CameraView+RecordVideo.kt:44)\n\tat com.mrousavy.camera.CameraView_RecordVideoKt$startRecording$1.invokeSuspend(Unknown Source:13)\n\tat kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:33)\n\tat kotlinx.coroutines.DispatchedTask.run(DispatchedTask.kt:106)\n\tat android.os.Handler.handleCallback(Handler.java:938)\n\tat android.os.Handler.dispatchMessage(Handler.java:99)\n\tat android.os.Looper.loop(Looper.java:223)\n\tat android.app.ActivityThread.main(ActivityThread.java:7664)\n\tat java.lang.reflect.Method.invoke(Native Method)\n\tat com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:592)\n\tat com.android.internal.os.ZygoteInit.main(ZygoteInit.java:947)\n',
         message: 'Attempt to invoke virtual method \'void android.media.AudioRecord.startRecording()\' on a null object reference' },
      code: 'capture/unknown' }
2021-07-07 12:34:09.788 32687-32728/com.mrousavy.camera.example I/ReactNativeJS: stopped recording video!
2021-07-07 12:34:11.433 32687-32728/com.mrousavy.camera.example D/ReactNativeJS: state: ACTIVE
2021-07-07 12:34:11.434 32687-32728/com.mrousavy.camera.example D/ReactNativeJS: state: END
2021-07-07 12:34:11.435 32687-32728/com.mrousavy.camera.example I/ReactNativeJS: calling stopRecording()...
2021-07-07 12:34:11.439 32687-32687/com.mrousavy.camera.example I/VideoCapture: stopRecording
2021-07-07 12:34:11.440 32687-32754/com.mrousavy.camera.example D/Camera2CameraImpl: {Camera@3fb9680[id=0]} Issue capture request
2021-07-07 12:34:11.440 32687-32754/com.mrousavy.camera.example D/CaptureSession: Issuing capture request.
2021-07-07 12:34:11.448 32687-32728/com.mrousavy.camera.example I/ReactNativeJS: called stopRecording()!
2021-07-07 12:34:11.460 32687-32754/com.mrousavy.camera.example D/UseCaseAttachState: Active and attached use case: [androidx.camera.core.Preview-0b672842-e184-45c2-b716-7fc9f7a777aa267873987, androidx.camera.core.VideoCapture-faf31d9d-9edb-4548-b576-5faf5ba21b04233554752, androidx.camera.core.ImageCapture-d8b3846a-7efe-44f0-87c1-4f67d2ecdb1b106943097] for camera: 0
2021-07-07 12:34:11.461 32687-32754/com.mrousavy.camera.example D/CaptureSession: Attempting to submit CaptureRequest after setting
2021-07-07 12:34:11.461 32687-32754/com.mrousavy.camera.example D/CaptureSession: Issuing request for session.
2021-07-07 12:34:11.464 32687-32754/com.mrousavy.camera.example D/Camera2CameraImpl: {Camera@3fb9680[id=0]} Use case androidx.camera.core.VideoCapture@debc340 UPDATED
2021-07-07 12:34:11.464 32687-32754/com.mrousavy.camera.example D/UseCaseAttachState: Active and attached use case: [androidx.camera.core.Preview-0b672842-e184-45c2-b716-7fc9f7a777aa267873987, androidx.camera.core.VideoCapture-faf31d9d-9edb-4548-b576-5faf5ba21b04233554752, androidx.camera.core.ImageCapture-d8b3846a-7efe-44f0-87c1-4f67d2ecdb1b106943097] for camera: 0
2021-07-07 12:34:11.465 32687-32754/com.mrousavy.camera.example D/CaptureSession: Attempting to submit CaptureRequest after setting
2021-07-07 12:34:11.465 32687-32754/com.mrousavy.camera.example D/CaptureSession: Issuing request for session.
```

mrousavy, 2021-07-07:
_probably_ fixed with https://github.com/mrousavy/react-native-vision-camera/commit/1e64215d375ff2b3de6104cca479c120ff982932 (#257)

closed 2021-07-07 by mrousavy

Kypsis, 2021-07-07:
Can confirm, that revoking microphone permission does not crash anymore. However hot reload breaks the example app now with:
```
2021-07-07 14:34:19.460 10995-10995/com.mrousavy.camera.example D/SurfaceViewImpl: Surface destroyed.
2021-07-07 14:34:19.460 10995-10995/com.mrousavy.camera.example D/SurfaceViewImpl: Surface invalidated androidx.camera.core.SurfaceRequest@17eea9d
2021-07-07 14:34:19.460 10995-10995/com.mrousavy.camera.example D/DeferrableSurface: surface closed,  useCount=1 closed=true androidx.camera.core.SurfaceRequest$2@f3eb12
2021-07-07 14:34:19.461 10995-10995/com.mrousavy.camera.example D/CameraView: Lifecycle went from RESUMED -> RESUMED (isActive: true | isAttachedToWindow: true)
2021-07-07 14:34:19.472 10995-11059/com.mrousavy.camera.example I/ReactNativeJS: Re-rendering camera page with inactive camera. Device: "back (0)" (4000x2000 @ 60fps)
2021-07-07 14:34:19.477 10995-10995/com.mrousavy.camera.example I/VideoCapture: stopRecording
2021-07-07 14:34:19.477 10995-10995/com.mrousavy.camera.example D/CameraView: Lifecycle went from RESUMED -> CREATED (isActive: true | isAttachedToWindow: false)
2021-07-07 14:34:19.478 10995-10995/com.mrousavy.camera.example W/unknown:ReactNative: Packager connection already open, nooping.
2021-07-07 14:34:19.478 10995-11060/com.mrousavy.camera.example I/ReactNative: [GESTURE HANDLER] Tearing down gesture handler registered for root view com.reactnativenavigation.react.ReactView{b8ac545 V.E...... .......D 0,0-1080,2296 #1}
2021-07-07 14:34:19.478 10995-11084/com.mrousavy.camera.example D/Camera2CameraImpl: {Camera@91b4f16[id=0]} Use case androidx.camera.core.VideoCapture@abdb4c8 UPDATED
2021-07-07 14:34:19.483 10995-11084/com.mrousavy.camera.example D/UseCaseAttachState: Active and attached use case: [androidx.camera.core.Preview-30caeae1-e972-4f11-8320-622c0cd8361646895467, androidx.camera.core.VideoCapture-c1d5f56d-fca2-4cad-b382-e7d70d49dfce180204744, androidx.camera.core.ImageCapture-d9b8ebf5-34d6-440e-81d4-76022971e63f176888161] for camera: 0
2021-07-07 14:34:19.487 10995-11084/com.mrousavy.camera.example D/CaptureSession: Attempting to submit CaptureRequest after setting
2021-07-07 14:34:19.488 10995-11084/com.mrousavy.camera.example D/CaptureSession: Issuing request for session.
2021-07-07 14:34:19.493 10995-10995/com.mrousavy.camera.example E/unknown:ReactInstanceManager: destroyRootView called
2021-07-07 14:34:19.493 10995-10995/com.mrousavy.camera.example E/unknown:ReactInstanceManager: destroyRootView called, unmountReactApplication
2021-07-07 14:34:19.497 10995-11084/com.mrousavy.camera.example D/Camera2CameraImpl: {Camera@91b4f16[id=0]} Use cases [Preview:androidx.camera.core.Preview-30caeae1-e972-4f11-8320-622c0cd83616, androidx.camera.core.VideoCapture@abdb4c8, ImageCapture:androidx.camera.core.ImageCapture-d9b8ebf5-34d6-440e-81d4-76022971e63f] now DETACHED for camera
2021-07-07 14:34:19.497 10995-11084/com.mrousavy.camera.example D/UseCaseAttachState: All use case: [] for camera: 0
2021-07-07 14:34:19.497 10995-11084/com.mrousavy.camera.example D/Camera2CameraImpl: {Camera@91b4f16[id=0]} Resetting Capture Session
2021-07-07 14:34:19.500 10995-11084/com.mrousavy.camera.example D/Camera2CameraImpl: {Camera@91b4f16[id=0]} Releasing session in state OPENED
2021-07-07 14:34:19.501 10995-11084/com.mrousavy.camera.example D/UseCaseAttachState: Active and attached use case: [] for camera: 0
2021-07-07 14:34:19.502 10995-11084/com.mrousavy.camera.example D/UseCaseAttachState: Active and attached use case: [] for camera: 0
2021-07-07 14:34:19.502 10995-11084/com.mrousavy.camera.example D/Camera2CameraImpl: {Camera@91b4f16[id=0]} Closing camera.
2021-07-07 14:34:19.502 10995-11084/com.mrousavy.camera.example D/Camera2CameraImpl: {Camera@91b4f16[id=0]} Transitioning camera internal state: OPENED --> CLOSING
2021-07-07 14:34:19.503 10995-11084/com.mrousavy.camera.example D/CameraStateRegistry: Recalculating open cameras:
    Camera                                       State                 
    -------------------------------------------------------------------
    Camera@8681069[id=1]                         UNKNOWN               
    Camera@91b4f16[id=0]                         CLOSING               
    -------------------------------------------------------------------
    Open count: 1 (Max allowed: 1)
2021-07-07 14:34:19.503 10995-11084/com.mrousavy.camera.example D/CameraStateMachine: New public camera state CameraState{type=CLOSING, error=null} from CLOSING and null
2021-07-07 14:34:19.503 10995-11084/com.mrousavy.camera.example D/CameraStateMachine: Publishing new public camera state CameraState{type=CLOSING, error=null}
2021-07-07 14:34:19.503 10995-10995/com.mrousavy.camera.example D/StreamStateObserver: Update Preview stream state to IDLE
2021-07-07 14:34:19.503 10995-11084/com.mrousavy.camera.example D/Camera2CameraImpl: {Camera@91b4f16[id=0]} Resetting Capture Session
2021-07-07 14:34:19.503 10995-11084/com.mrousavy.camera.example D/Camera2CameraImpl: {Camera@91b4f16[id=0]} Releasing session in state CLOSING
2021-07-07 14:34:19.503 10995-11084/com.mrousavy.camera.example D/CaptureSession: onSessionFinished()
2021-07-07 14:34:19.511 10995-11060/com.mrousavy.camera.example W/unknown:ViewManagerPropertyUpdater: Could not find generated setter for class com.facebook.react.uimanager.LayoutShadowNode
[... cut ...]
        at com.mrousavy.camera.CameraViewModule.lambda$dYYdgYuKJvU8OejKuKT5KLSehwU(Unknown Source:0)
        at com.mrousavy.camera.-$$Lambda$CameraViewModule$dYYdgYuKJvU8OejKuKT5KLSehwU.run(Unknown Source:2)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167)
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641)
        at java.lang.Thread.run(Thread.java:923)
2021-07-07 14:34:21.586 10995-11063/com.mrousavy.camera.example I/Process: Sending signal. PID: 10995 SIG: 9
```

mrousavy, 2021-07-07:
Can you try with https://github.com/mrousavy/react-native-vision-camera/pull/258 to see if that fixes the hot-reload bug for you?

Kypsis, 2021-07-07:
Fails to compile unless 
```kotlin
HybridData?
```
is changed to 
 ```kotlin
HybridData
```
in **src/main/java/com/mrousavy/camera/frameprocessor/FrameProcessorRuntimeManager.kt** at line 64

But once it compiles it does not crash anymore 👍

mrousavy, 2021-07-07:
thanks, merged.