margelo/react-native-vision-camera #238: 🐛  An unknown camera error occured on Samsung Galaxy Note10+

Kypsis, 2021-07-06:
## What

The camera preview will not initialise on Samsung Galaxy Note10+. `const devices = useCameraDevices(); ` remains undefined.

### Logs

`LOG  Running "CameraPage" with {"initialProps":{"componentId":"Component1"},"rootTag":1}
 LOG  re-rendering camera page without active camera
 WARN  Possible Unhandled Promise Rejection (id: 0):
unknown/unknown: [unknown/unknown] An unknown camera error occured.
[unknown/unknown]: [unknown/unknown] An unknown camera error occured.
    at construct (native)
    at apply (native)
    at _construct (http://localhost:8081/index.bundle?platform=android&dev=true&minify=false&app=com.mrousavy.camera.example&modulesOnly=false&runModule=true:118386:28)
    at Wrapper (http://localhost:8081/index.bundle?platform=android&dev=true&minify=false&app=com.mrousavy.camera.example&modulesOnly=false&runModule=true:118339:64)
    at construct (native)
    at _createSuperInternal (http://localhost:8081/index.bundle?platform=android&dev=true&minify=false&app=com.mrousavy.camera.example&modulesOnly=false&runModule=true:118135:322)
    at call (native)
    at CameraError (http://localhost:8081/index.bundle?platform=android&dev=true&minify=false&app=com.mrousavy.camera.example&modulesOnly=false&runModule=true:118148:26)
    at construct (native)
    at _createSuperInternal (http://localhost:8081/index.bundle?platform=android&dev=true&minify=false&app=com.mrousavy.camera.example&modulesOnly=false&runModule=true:118135:322)
    at apply (native)
    at CameraRuntimeError (http://localhost:8081/index.bundle?platform=android&dev=true&minify=false&app=com.mrousavy.camera.example&modulesOnly=false&runModule=true:118203:27)
    at tryParseNativeCameraError (http://localhost:8081/index.bundle?platform=android&dev=true&minify=false&app=com.mrousavy.camera.example&modulesOnly=false&runModule=true:118226:38)
    at getAvailableCameraDevices$ (http://localhost:8081/index.bundle?platform=android&dev=true&minify=false&app=com.mrousavy.camera.example&modulesOnly=false&runModule=true:117201:102)
    at call (native)
    at tryCatch (http://localhost:8081/index.bundle?platform=android&dev=true&minify=false&app=com.mrousavy.camera.example&modulesOnly=false&runModule=true:117446:23)
    at invoke (http://localhost:8081/index.bundle?platform=android&dev=true&minify=false&app=com.mrousavy.camera.example&modulesOnly=false&runModule=true:117619:32)
    at anonymous (http://localhost:8081/index.bundle?platform=android&dev=true&minify=false&app=com.mrousavy.camera.example&modulesOnly=false&runModule=true:117489:30)
    at call (native)
    at tryCatch (http://localhost:8081/index.bundle?platform=android&dev=true&minify=false&app=com.mrousavy.camera.example&modulesOnly=false&runModule=true:117446:23)
    at invoke (http://localhost:8081/index.bundle?platform=android&dev=true&minify=false&app=com.mrousavy.camera.example&modulesOnly=false&runModule=true:117519:30)
    at anonymous (http://localhost:8081/index.bundle?platform=android&dev=true&minify=false&app=com.mrousavy.camera.example&modulesOnly=false&runModule=true:117531:21)
    at tryCallOne (http://localhost:8081/index.bundle?platform=android&dev=true&minify=false&app=com.mrousavy.camera.example&modulesOnly=false&runModule=true:30482:16)
    at anonymous (http://localhost:8081/index.bundle?platform=android&dev=true&minify=false&app=com.mrousavy.camera.example&modulesOnly=false&runModule=true:30583:27)
    at apply (native)
    at anonymous (http://localhost:8081/index.bundle?platform=android&dev=true&minify=false&app=com.mrousavy.camera.example&modulesOnly=false&runModule=true:31111:26)
    at _callTimer (http://localhost:8081/index.bundle?platform=android&dev=true&minify=false&app=com.mrousavy.camera.example&modulesOnly=false&runModule=true:30999:17)
    at _callImmediatesPass (http://localhost:8081/index.bundle?platform=android&dev=true&minify=false&app=com.mrousavy.camera.example&modulesOnly=false&runModule=true:31038:17)
    at callImmediates (http://localhost:8081/index.bundle?platform=android&dev=true&minify=false&app=com.mrousavy.camera.example&modulesOnly=false&runModule=true:31256:33)
    at __callImmediates (http://localhost:8081/index.bundle?platform=android&dev=true&minify=false&app=com.mrousavy.camera.example&modulesOnly=false&runModule=true:8072:35)
    at anonymous (http://localhost:8081/index.bundle?platform=android&dev=true&minify=false&app=com.mrousavy.camera.example&modulesOnly=false&runModule=true:7850:34)
    at __guard (http://localhost:8081/index.bundle?platform=android&dev=true&minify=false&app=com.mrousavy.camera.example&modulesOnly=false&runModule=true:8055:15)
    at flushedQueue (http://localhost:8081/index.bundle?platform=android&dev=true&minify=false&app=com.mrousavy.camera.example&modulesOnly=false&runModule=true:7849:21)
    at invokeCallbackAndReturnFlushedQueue (http://localhost:8081/index.bundle?platform=android&dev=true&minify=false&app=com.mrousavy.camera.example&modulesOnly=false&runModule=true:7842:33)
 LOG  re-rendering camera page without active camera`

## Reproducable sample

react-native-vision-camera example code.

mrousavy, 2021-07-06:
Could you please correctly format the error logs? use three backticks (```) for code blocks

Kypsis, 2021-07-06:
I'll add a remark here just incase it might help. I have a side project in Flutter and there the camera module on Samsung Galaxy Note10+ for whatever reason is noticeably slower to init than Pixel 4a for example.

mrousavy, 2021-07-07:
Could you please share the native logs in Android Studio/Logcat?

Kypsis, 2021-07-07:
```
2021-07-07 12:41:23.437 6996-6996/? I/.camera.exampl: Late-enabling -Xcheck:jni
2021-07-07 12:41:23.468 6996-6996/? I/.camera.exampl: Unquickening 22 vdex files!
2021-07-07 12:41:23.503 6996-6996/? D/ActivityThread: setConscryptValidator
2021-07-07 12:41:23.504 6996-6996/? D/ActivityThread: setConscryptValidator - put
2021-07-07 12:41:23.568 6996-6996/com.mrousavy.camera.example D/ActivityThread: handleBindApplication()++ app=com.mrousavy.camera.example
2021-07-07 12:41:23.577 6996-6996/com.mrousavy.camera.example I/.camera.exampl: The ClassLoaderContext is a special shared library.
2021-07-07 12:41:23.732 6996-6996/com.mrousavy.camera.example D/ActivityThread: handleBindApplication() -- skipGraphicsSupport=false
2021-07-07 12:41:23.733 6996-6996/com.mrousavy.camera.example D/ActivityThread: ActivityThread::handleMakeApplication() data=AppBindData{appInfo=ApplicationInfo{e24dd84 com.mrousavy.camera.example}} appContext=android.app.ContextImpl@948b96d appContext.mOpPackageName=com.mrousavy.camera.example appContext.mBasePackageName=com.mrousavy.camera.example appContext.mPackageInfo=android.app.LoadedApk@3c459a2 data.restrictedBackupMode= false data.providers= [] Caller=android.app.ActivityThread.handleBindApplication:7446 android.app.ActivityThread.access$1500:301 android.app.ActivityThread$H.handleMessage:2148 android.os.Handler.dispatchMessage:106 android.os.Looper.loop:246 
2021-07-07 12:41:23.735 6996-6996/com.mrousavy.camera.example D/LoadedApk: LoadedApk::makeApplication() appContext=android.app.ContextImpl@335b633 appContext.mOpPackageName=com.mrousavy.camera.example appContext.mBasePackageName=com.mrousavy.camera.example appContext.mPackageInfo=android.app.LoadedApk@3c459a2
2021-07-07 12:41:23.735 6996-6996/com.mrousavy.camera.example D/NetworkSecurityConfig: No Network Security Config specified, using platform default
2021-07-07 12:41:23.753 6996-6996/com.mrousavy.camera.example D/NetworkSecurityConfig: No Network Security Config specified, using platform default
2021-07-07 12:41:23.757 6996-6996/com.mrousavy.camera.example D/SoLoader: init start
2021-07-07 12:41:23.758 6996-6996/com.mrousavy.camera.example D/SoLoader: adding system library source: /vendor/lib64
2021-07-07 12:41:23.758 6996-6996/com.mrousavy.camera.example D/SoLoader: adding system library source: /system/lib64
2021-07-07 12:41:23.758 6996-6996/com.mrousavy.camera.example D/SoLoader: adding application source: com.facebook.soloader.DirectorySoSource[root = /data/app/~~-z_4R8Pzz5Mzgaf5qRYxYw==/com.mrousavy.camera.example-nw7VZFdPxP4wyn44OUVcTg==/lib/arm64 flags = 0]
2021-07-07 12:41:23.759 6996-6996/com.mrousavy.camera.example D/SoLoader: adding backup source from : com.facebook.soloader.ApkSoSource[root = /data/data/com.mrousavy.camera.example/lib-main flags = 1]
2021-07-07 12:41:23.759 6996-6996/com.mrousavy.camera.example D/SoLoader: Preparing SO source: com.facebook.soloader.DirectorySoSource[root = /system/lib64 flags = 2]
2021-07-07 12:41:23.759 6996-6996/com.mrousavy.camera.example D/SoLoader: Preparing SO source: com.facebook.soloader.DirectorySoSource[root = /vendor/lib64 flags = 2]
2021-07-07 12:41:23.759 6996-6996/com.mrousavy.camera.example D/SoLoader: Preparing SO source: com.facebook.soloader.DirectorySoSource[root = /data/app/~~-z_4R8Pzz5Mzgaf5qRYxYw==/com.mrousavy.camera.example-nw7VZFdPxP4wyn44OUVcTg==/lib/arm64 flags = 0]
2021-07-07 12:41:23.759 6996-6996/com.mrousavy.camera.example D/SoLoader: Preparing SO source: com.facebook.soloader.ApkSoSource[root = /data/data/com.mrousavy.camera.example/lib-main flags = 1]
2021-07-07 12:41:23.760 6996-6996/com.mrousavy.camera.example V/fb-UnpackingSoSource: locked dso store /data/user/0/com.mrousavy.camera.example/lib-main
2021-07-07 12:41:23.761 6996-6996/com.mrousavy.camera.example V/fb-UnpackingSoSource: deps mismatch on deps store: regenerating
2021-07-07 12:41:23.761 6996-6996/com.mrousavy.camera.example V/fb-UnpackingSoSource: so store dirty: regenerating
2021-07-07 12:41:23.810 6996-6996/com.mrousavy.camera.example D/ApkSoSource: not allowing consideration of lib/arm64-v8a/libVisionCamera.so: deferring to libdir
2021-07-07 12:41:23.810 6996-6996/com.mrousavy.camera.example D/ApkSoSource: not allowing consideration of lib/arm64-v8a/libc++_shared.so: deferring to libdir
2021-07-07 12:41:23.810 6996-6996/com.mrousavy.camera.example D/ApkSoSource: not allowing consideration of lib/arm64-v8a/libfb.so: deferring to libdir
2021-07-07 12:41:23.810 6996-6996/com.mrousavy.camera.example D/ApkSoSource: not allowing consideration of lib/arm64-v8a/libfbjni.so: deferring to libdir
2021-07-07 12:41:23.810 6996-6996/com.mrousavy.camera.example D/ApkSoSource: not allowing consideration of lib/arm64-v8a/libfolly_futures.so: deferring to libdir
2021-07-07 12:41:23.810 6996-6996/com.mrousavy.camera.example D/ApkSoSource: not allowing consideration of lib/arm64-v8a/libfolly_json.so: deferring to libdir
2021-07-07 12:41:23.810 6996-6996/com.mrousavy.camera.example D/ApkSoSource: not allowing consideration of lib/arm64-v8a/libglog.so: deferring to libdir
2021-07-07 12:41:23.810 6996-6996/com.mrousavy.camera.example D/ApkSoSource: not allowing consideration of lib/arm64-v8a/libglog_init.so: deferring to libdir
2021-07-07 12:41:23.810 6996-6996/com.mrousavy.camera.example D/ApkSoSource: not allowing consideration of lib/arm64-v8a/libhermes-executor-common-debug.so: deferring to libdir
[... cut ...]
2021-07-07 12:42:15.843 6996-6996/com.mrousavy.camera.example I/SurfaceControl: release : mNativeObject = 483197003296 - Surface(name=Surface(name=743ae47 StatusBar)/@0x5debcad - animation-leash)/@0xd665941 / android.view.-$$Lambda$Rl1VZmNJ0VZDLK0BAbaVGis0rrA.accept:2 android.view.InsetsSourceControl.release:170 android.view.InsetsSourceConsumer.setControl:202 android.view.InsetsController.onControlsChanged:833 
2021-07-07 12:42:15.843 6996-6996/com.mrousavy.camera.example I/SurfaceControl: nativeRelease nativeObject s[483197003296]
2021-07-07 12:42:15.843 6996-6996/com.mrousavy.camera.example I/SurfaceControl: nativeRelease nativeObject e[483197003296]
```

mrousavy, 2021-07-07:
Why did you not fill out the issue template I explicitly created for such cases? now I have to ask you for every detail. 

What VisionCamera version are you using?

Kypsis, 2021-07-07:
Well seems the code block formatting hid the part where I wrote that it's the most current example from react-native-vision-camera. And seeing as it is from the most current example any other info than logs seemed extraneous 🙂

mrousavy, 2021-07-07:
Oh I see. So in the stacktrace I can see that there is a NullPointerException being raised here: https://github.com/mrousavy/react-native-vision-camera/blob/0bb9ba1e8a49e945530b8a7d733ba26aa57b3279/android/src/main/java/com/mrousavy/camera/CameraViewModule.kt#L160

The documentation for `CONTROL_AVAILABLE_VIDEO_STABILIZATION_MODES` literally says that this key is available on all devices (and therefore not null), wtf?

mrousavy, 2021-07-07:
Can you try again with the current master (3007cd9) to double-check the line numbers in the stacktrace? If it's still `CameraViewModule line 160` then that's weird...

mrousavy, 2021-07-07:
let me know if https://github.com/mrousavy/react-native-vision-camera/pull/259 fixes the issue.

Kypsis, 2021-07-07:
Not crashing anymore just black camera preview screen. Pressed the camera flip button twice just in case as well. Logcat below:
```
2021-07-07 15:24:13.046 24388-24388/com.mrousavy.camera.example I/ViewRootImpl@859bef[MainActivity]: ViewPostIme pointer 0
2021-07-07 15:24:13.112 24388-24388/com.mrousavy.camera.example I/ViewRootImpl@859bef[MainActivity]: ViewPostIme pointer 1
2021-07-07 15:24:13.131 24388-26604/com.mrousavy.camera.example I/ReactNativeJS: Re-rendering camera page with active camera. Device: "back (0)" (2288x1080 @ 60fps)
2021-07-07 15:24:13.230 24388-24388/com.mrousavy.camera.example I/CameraView: Configuring session...
2021-07-07 15:24:13.231 24388-24388/com.mrousavy.camera.example I/CameraView: Configuring session with Camera ID 0 and custom format...
2021-07-07 15:24:13.236 24388-24388/com.mrousavy.camera.example I/CameraView: Using custom format - photo: 2288x1080, video: 2288x1080 @ 60 FPS
2021-07-07 15:24:13.237 24388-24388/com.mrousavy.camera.example I/CameraView: Setting AE_TARGET_FPS_RANGE to 60-60, and SENSOR_FRAME_DURATION to 0
2021-07-07 15:24:13.237 24388-24388/com.mrousavy.camera.example I/VideoCapture: stopRecording
2021-07-07 15:24:13.239 24388-26655/com.mrousavy.camera.example D/Camera2CameraImpl: {Camera@138919f[id=1]} Use case androidx.camera.core.VideoCapture@3ac860b UPDATED
2021-07-07 15:24:13.239 24388-24388/com.mrousavy.camera.example D/DeferrableSurface: surface closed,  useCount=1 closed=true androidx.camera.core.SurfaceRequest$2@9c89232
2021-07-07 15:24:13.239 24388-24388/com.mrousavy.camera.example I/VideoCapture: stopRecording
2021-07-07 15:24:13.239 24388-26655/com.mrousavy.camera.example D/UseCaseAttachState: Active and attached use case: [androidx.camera.core.VideoCapture-e542a522-2564-4243-8e68-88e72407c5ab61638155, androidx.camera.core.ImageCapture-7721d5c2-f63e-4dac-ad21-0a9e502caff8147758312, androidx.camera.core.Preview-8153a298-1947-43fb-b74e-9e24b76d854e109552090] for camera: 1
2021-07-07 15:24:13.240 24388-26765/com.mrousavy.camera.example I/CCodec: state->set(RELEASING)
2021-07-07 15:24:13.240 24388-26655/com.mrousavy.camera.example D/CaptureSession: Attempting to submit CaptureRequest after setting
2021-07-07 15:24:13.240 24388-26655/com.mrousavy.camera.example D/CaptureSession: Issuing request for session.
2021-07-07 15:24:13.242 24388-26840/com.mrousavy.camera.example I/CCodec: [c2.android.aac.encoder] release(1)
2021-07-07 15:24:13.243 24388-26840/com.mrousavy.camera.example I/CCodec: state->set(RELEASED)
2021-07-07 15:24:13.243 24388-26765/com.mrousavy.camera.example I/hw-BpHwBinder: onLastStrongRef automatically unlinking death recipients
2021-07-07 15:24:13.244 24388-26765/com.mrousavy.camera.example I/MediaCodec: Codec shutdown complete
2021-07-07 15:24:13.246 24388-24388/com.mrousavy.camera.example D/DeferrableSurface: surface closed,  useCount=1 closed=true androidx.camera.core.impl.ImmediateSurface@3057cdf
2021-07-07 15:24:13.247 24388-24388/com.mrousavy.camera.example D/DeferrableSurface: surface closed,  useCount=1 closed=true androidx.camera.core.impl.ImmediateSurface@1ef07fb
2021-07-07 15:24:13.253 24388-24388/com.mrousavy.camera.example I/CameraView: Attaching 2 use-cases...
2021-07-07 15:24:13.257 24388-26655/com.mrousavy.camera.example D/Camera2CameraImpl: {Camera@138919f[id=1]} Use cases [Preview:androidx.camera.core.Preview-8153a298-1947-43fb-b74e-9e24b76d854e, androidx.camera.core.VideoCapture@3ac860b, ImageCapture:androidx.camera.core.ImageCapture-7721d5c2-f63e-4dac-ad21-0a9e502caff8] now DETACHED for camera
2021-07-07 15:24:13.257 24388-26655/com.mrousavy.camera.example D/UseCaseAttachState: All use case: [] for camera: 1
2021-07-07 15:24:13.257 24388-26655/com.mrousavy.camera.example D/Camera2CameraImpl: {Camera@138919f[id=1]} Resetting Capture Session
2021-07-07 15:24:13.258 24388-26655/com.mrousavy.camera.example D/Camera2CameraImpl: {Camera@138919f[id=1]} Releasing session in state OPENED
2021-07-07 15:24:13.258 24388-26655/com.mrousavy.camera.example D/UseCaseAttachState: Active and attached use case: [] for camera: 1
2021-07-07 15:24:13.259 24388-26655/com.mrousavy.camera.example D/UseCaseAttachState: Active and attached use case: [] for camera: 1
2021-07-07 15:24:13.259 24388-26655/com.mrousavy.camera.example D/Camera2CameraImpl: {Camera@138919f[id=1]} Closing camera.
2021-07-07 15:24:13.259 24388-26655/com.mrousavy.camera.example D/Camera2CameraImpl: {Camera@138919f[id=1]} Transitioning camera internal state: OPENED --> CLOSING
2021-07-07 15:24:13.261 24388-26655/com.mrousavy.camera.example D/CameraStateRegistry: Recalculating open cameras:
    Camera                                       State                 
    -------------------------------------------------------------------
    Camera@b1901d8[id=2]                         UNKNOWN               
    Camera@bfcc76d[id=3]                         UNKNOWN               
    Camera@138919f[id=1]                         CLOSING               
    Camera@409287[id=4]                          UNKNOWN               
    Camera@9ea3966[id=0]                         CLOSED                
    -------------------------------------------------------------------
    Open count: 1 (Max allowed: 1)
2021-07-07 15:24:13.261 24388-26655/com.mrousavy.camera.example D/CameraStateMachine: New public camera state CameraState{type=CLOSING, error=null} from CLOSING and null
[... cut ...]
2021-07-07 15:26:13.894 24388-24388/com.mrousavy.camera.example I/SurfaceControl: release : mNativeObject = 483197287664 - Surface(name=Surface(name=743ae47 StatusBar)/@0x5debcad - animation-leash)/@0x6ef12 / android.view.-$$Lambda$Rl1VZmNJ0VZDLK0BAbaVGis0rrA.accept:2 android.view.InsetsSourceControl.release:170 android.view.InsetsSourceConsumer.setControl:202 android.view.InsetsController.onControlsChanged:833 
2021-07-07 15:26:13.894 24388-24388/com.mrousavy.camera.example I/SurfaceControl: nativeRelease nativeObject s[483197287664]
2021-07-07 15:26:13.894 24388-24388/com.mrousavy.camera.example I/SurfaceControl: nativeRelease nativeObject e[483197287664]
```

mrousavy, 2021-07-07:
Yeah this looks like a bug in the native CameraX library from Google. Can you create an issue for this [in their issuetracker](https://issuetracker.google.com/issues/new?component=618491&template=1257717) please?

Kypsis, 2021-07-07:
I'm sorry but I don't think it's wise for me (for many reasons) to file an issue on behalf of a third party library. But it looks like the issue has already been raised: [https://issuetracker.google.com/issues/188690069](https://issuetracker.google.com/issues/188690069)

thestretivibity, 2021-07-07:
Same problem here, note10+ with android11 black screen.

mrousavy, 2021-09-23:
Yeah so apparently the stabilization modes cause null pointer exceptions even though they should be available on all devices:

[image]

So the android docs are lying and I just fixed this in #449

closed 2021-09-23 by mrousavy

mrousavy, 2021-09-24:
I've fixed this in [**2.8.1**](https://github.com/mrousavy/react-native-vision-camera/releases/tag/v2.8.1) 🎉🚀

rusakovic, 2022-02-23:
> I've fixed this in [**2.8.1**](https://github.com/mrousavy/react-native-vision-camera/releases/tag/v2.8.1) 🎉🚀

Thank you very much for your efforts.
But caught the same issue on Samsung s9.
 "react-native-vision-camera": "^2.12.0",

Some info from sentry.

```
java.lang.NullPointerException: null
    at com.mrousavy.camera.CameraView_RecordVideoKt$startRecording$1.accept(CameraView+RecordVideo.kt:79)
    at com.mrousavy.camera.CameraView_RecordVideoKt$startRecording$1.accept(CameraView+RecordVideo.kt:52)
    at androidx.camera.video.Recorder$RecordingRecord.lambda$updateVideoRecordEvent$6$Recorder$RecordingRecord(Recorder.java:2611)
    at androidx.camera.video.-$$Lambda$Recorder$RecordingRecord$Uo40MCKPuma4aZ5df1-_fzufzHg.run
    at android.os.Handler.handleCallback(Handler.java:873)
    at android.os.Handler.dispatchMessage(Handler.java:99)
    at android.os.Looper.loop(Looper.java:214)
    at android.app.ActivityThread.main(ActivityThread.java:6981)
    at java.lang.reflect.Method.invoke(Method.java)
    at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:493)
    at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1445)

```

```
Model | SM-G960U (PPR1.180610.011)
Model Id | PPR1.180610.011
Name | Galaxy S9

Kernel Version | 4.9.112-15611914
Name | Android
Rooted | no
Version | 9

```

Could you, please, help with it?
Thank you!

rusakovic, 2022-03-09:
Hello @mrousavy

Do you have a possibility to check this issue?
The same issue as other users described - black screen when open the camera on Samsung.

Thank you.

mrousavy, 2022-03-10:
Hey!

> Do you have a possibility to check this issue?

Sorry, no. I do not have that specific test device and I cannot reproduce it on my device.

PR #449 (2021-09-23): fix: Fix Android Stabilization Modes

## What

On some Samsung devices, getting the Digital Lens Stabilization Modes returns null, even though the Android Docs say it cannot be null:

[image]

So the docs are lying.

## Changes

* Fixes Stabilization Modes being `null` on some Samsung devices (even though the Android docs say it cannot be null!)
* Always returns at least `off` for stabilization modes
* Returns `cinematic` for optical stabilization mode

## Tested on

## Related issues

* Fixes #438
* Fixes #238