margelo/react-native-vision-camera #279: 🐛

julianklumpers, 2021-07-15:
### What were you trying to do?

I take a picture with `const image = await cameraRef.current.takePhoto()` which works fine.

### What happened instead?

But the output data has something strange.
The width and height are swapped... the height should be the width and visa versa, see output: 
`{
   "height": 2160,
   "width": 3840,
  "isRawPhoto": false,
  "path": "/private/var/mobile/Containers/Data/Application/4C4B43F6-0476-4E97-BECF-4C3CE1E28DAB/tmp/ReactNative/A538B243-30FB-42B1-B31D-E4C5B9E0326F.jpeg",
  "thumbnail": null,
...more
}`

From the documentation i read this lib only supports portrait mode for now, so i assume it has not something to do with landscape/portrait mode.

When i test the output with: `Image.getSize(image.path, console.log);` i get the correct width and height of the image.

### Relevant log output

```shell
LOG  Camera image output:  3840x2160
 LOG  Measured image with Image.getSize:  2160x3840
```

### Device

iPhone 12 PRO (ios 14.x)

### VisionCamera Version

2.4.1

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2021-07-15:
Actually the longer side is **always** the width unless you explicitly rotate it, that's how camera sensors work (at least on iOS).
I believe if I add rotation support I can swap width and height to be more logical

julianklumpers, 2021-07-15:
Hi @mrousavy,

Thanks for your fast response. I see, this makes sense. It was confusing for me at first, but after you mention this it makes sense. Rotation support will be awesome. I think this takes confusing away from users that needs to deal with the size of the image.

closed 2021-07-15 by julianklumpers