margelo/react-native-vision-camera #281: 🐛 Android: incorrect resolution/size photo output.

julianklumpers, 2021-07-17:
### What were you trying to do?

On android, I get a max photo output of 1080x1920 resolution. While the camera is capable of taking pictures of 3000x4000. It looks like the camera is zoomed in somehow, but i have zoom on 0.

I tried the expo-camera/react-native-camera library and their output size is correct.

### What happened instead?

Photo output is max 1080x1920 while the camera is capable of 3000x4000

### Device

iPhone 12 pro (ios 14.x)

### VisionCamera Version

2.4.1

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2021-07-24:
Please share the code of the camera component. Are you using `format={...}`? What `device={...}` are you using? expo-camera and react-native-camera both do not support custom/multiple devices selection nor format selection

julianklumpers, 2021-07-25:
Hi Marc,

Thank you for your reply.
The device I'm using is a Samsung A41.

**1. Did you use format={...}**
Yes, I did use format={...} no luck either. i used the sort format function from the documentation to get the highest resolution
https://mrousavy.github.io/react-native-vision-camera/docs/guides/formats#sort
output: 
```{"autoFocusSystem": "none", "colorSpaces": ["jpeg"], "fieldOfView": 69.64898258937245, "frameRateRanges": [{"maxFrameRate": 30, "minFrameRate": 1}, {"maxFrameRate": 10, "minFrameRate": 10}, {"maxFrameRate": 15, "minFrameRate": 15}, {"maxFrameRate": 20, "minFrameRate": 15}, {"maxFrameRate": 20, "minFrameRate": 20}, {"maxFrameRate": 30, "minFrameRate": 5}, {"maxFrameRate": 30, "minFrameRate": 30}], "isHighestPhotoQualitySupported": true, "maxISO": 1600, "maxZoom": 4, "minISO": 100, "photoHeight": 1440, "photoWidth": 2560, "supportsPhotoHDR": false, "supportsVideoHDR": false, "videoHeight": 1440, "videoStabilizationModes": ["off", "auto", "standard"], "videoWidth": 2560}```

as you can see the max resolution is 1440x2560

Output from the devices is only 1: `["wide-angle-camera"]` while the phone has 3 cameras.

Both expo-camera en react-native-camera gives me a resolution of 2250x4000

When i take a photo, the output is: `{"height": 1080, "isRawPhoto": false, "metadata": null, "path": "/data/user/0/com.cameratest/cache/mrousavy99938243064036335.jpg", "width": 1440}`

Code from camera component.
```
export default function App() {
  const [hasPermission, setHasPermission] = React.useState(null);

  const cameraRef = React.useRef();
  const devices = useCameraDevices();
  const device = devices.back;

  const formats = React.useMemo(
    () => device?.formats.sort(sortFormats),
    [device?.formats],
  );

  React.useEffect(() => {
    (async () => {
      const cameraPermission = await Camera.getCameraPermissionStatus();
      const microphonePermission = await Camera.getMicrophonePermissionStatus();

      if (microphonePermission === 'not-determined') {
        await Camera.requestMicrophonePermission();
      }
      if (cameraPermission === 'not-determined') {
        await Camera.requestCameraPermission();
      }

      setHasPermission(cameraPermission === 'authorized');
    })();
  }, []);

  const handleCaputre = async () => {
    console.log('asdasd');
    const data = await cameraRef.current.takePhoto({
      enableAutoStabilization: true,
      skipMetadata: true,
      photoCodec: 'jpeg',
      quality: 100,
    });

    console.log(data);
  };

  if (hasPermission === null) {
    return <View />;
  }

  if (hasPermission === false) {
    return <Text>No access to camera</Text>;
  }

  return (
    <SafeAreaView style={styles.container}>
      <View
        style={{
          ...StyleSheet.absoluteFillObject,
          display: 'flex',
          justifyContent: 'center',
          alignItems: 'center',
        }}>
        <Camera
          ref={cameraRef}
          style={{...StyleSheet.absoluteFillObject}}
          device={device}
          format={formats[0]}
          zoom={0}
          isActive={true}
          photo={true}
          video={true}
          audio={false}
        />
        <Pressable
          style={{...StyleSheet.absoluteFillObject, zIndex: 999999}}
          onPress={handleCaputre}
        />
      </View>
    </SafeAreaView>
  );
}
```

julianklumpers, 2021-08-08:
Any update on this? The issue still persist in version 2.5.0. 
I'm not able to get a high resolution output (tested it on 3 different android devices so far). The max resolution i'm getting is 2560x1440. While react-native-camera gives me 4000x3000, using the same camera.

Please clarify this for me? In case I'm missing something or this is an actual bug.

Thanks

mrousavy, 2021-08-08:
CameraX does not expose an API that provides all of the information we need for formats. I'm patching this information together and apparently this returns wrong sizes (using CameraCharacteristics), I can take a look soon

julianklumpers, 2021-08-08:
Hi @mrousavy,

Thanks for your reply. Glad you want to take a look at it. Other than that, this library is really great and has a lot of potential for a go-to camera library for React native.

I will try to help where i can (as my java/kotlin knowledge is almost zero). i can do some research.

julianklumpers, 2021-08-08:
Maybe this is something to look after?

"You can always use camera2 api StreamConfigurationMap.getOutputSizes() to get supported resolution even in CameraX. As CameraX is build on top of camera2 it shouldn't matter"

source: [https://stackoverflow.com/questions/60901831/how-to-get-available-resolutions-using-camerax-on-android](https://stackoverflow.com/questions/60901831/how-to-get-available-resolutions-using-camerax-on-android)

mrousavy, 2021-08-08:
Yes, I know, that's what I'm doing https://github.com/mrousavy/react-native-vision-camera/blob/ee762ce5f5f6d6738d3e62b9042c90322f0b09ef/android/src/main/java/com/mrousavy/camera/CameraViewModule.kt#L207

You could help debug this code to find out why it isn't working

julianklumpers, 2021-08-08:
Thanks for pointing me in the right direction. I will help debug this and come back to you.

jeremybarbet, 2021-08-14:
I noticed that if you don't use a custom format and let cameraX determine the configuration, the output photo is not zoomed in.

So I just created a patch to use `setTargetAspectRatio` based on the `format.photoHeight` and `format.photoWidth` instead of the `setTargetResolution` with the `format.photoSize`, it does the job for me: https://gist.github.com/jeremybarbet/c89698df5ddd8f312de04aed4f2dc555

I tried to see why `setTargetResolution` is not successfully using the resolution passed from the format but I can't see from where the issue is coming from.

mrousavy, 2021-08-16:
Nice job @jeremybarbet!

This always chooses the best resolution available though... If you find a solution that also chooses the correct resolution, please let me know/create a PR! Maybe the rotation is messed up?

ldstein, 2021-09-16:
Maybe related, but I noticed when hacking the FrameProcessor to capture a specific resolution, I had to reverse the width and height in order for CameraX to select the exact resolution described:

```
        // Hack! Ensure Preview and imageAnalysisBuilder are seeing them same thing
        var resolutionFlipped = Size(format.videoSize.height, format.videoSize.width)
        previewBuilder.setTargetResolution(resolutionFlipped)
        imageAnalysisBuilder.setTargetResolution(resolutionFlipped)
```

If I didn't reverse it, AndroidX is choosing a best fit on the wrong aspect ratio.

mrousavy, 2021-09-16:
Good catch @ldstein! Camera sizes are specified at landscape orientation, maybe CameraX is different in that case? Maybe CameraX changes that representation depending on Phone Orientation? 
If you fixed the issue, could you shoot me a PR?

ldstein, 2021-09-17:
Looking into it. https://github.com/ldstein/react-native-vision-camera/commit/a63c2924bb542e8c4f4ef0568ff3ffe96284f67b

Not confident yet this is consistent across all Android devices.

mrousavy, 2021-09-17:
@ldstein  this might be interesting:

[image]

Source: [CameraX Configuration Options (Specify Resolution)](https://developer.android.com/training/camerax/configuration#specify-resolution)

GaspardC, 2021-11-22:
@julianklumpers did you manage to solve this resolution problem ? 

I also get a max resolution of 2560x1440 on my Samsung A41 while `react-native-camera` gives me 4000x3000.

viljark, 2021-11-22:
For Android devices I had to swap the format image and video sizes before setting it for the Camera component.

```
if (format && Platform.OS === 'android') {
    ;[format.photoHeight, format.photoWidth] = [format.photoWidth, format.photoHeight]
    ;[format.videoHeight, format.videoWidth] = [format.videoWidth, format.videoHeight]

    setFormat(format)
}
```

this way I got correct video preview size and correct camera output.

mrousavy, 2021-11-22:
lol that's gotta be one of the weirdest JS code snippets I've ever seen

mrousavy, 2021-11-22:
Okay but seems like rotation is applied incorrectly in native

julianklumpers, 2021-11-22:
@GaspardC No did not solve the resolution problem yet unfortunately. I can try @viljark solution and hope it works for me to. But all of this seems like a temporary hack/fix. I did not have the time and knowledge to look into the native Java/Kotlin code to help solve it. So I hope @mrousavy could help with this because it is a real blocker for us right now for our product to use this library at the moment.

mrousavy, 2021-11-23:
I don't have any free time right now, been busy with the app development agency I'm running.
I'll take a look when I have free time :)

viljark, 2021-11-24:
> For Android devices I had to swap the format's image and video sizes before setting it for the Camera component.
> 
> ```
> if (format && Platform.OS === 'android') {
>     ;[format.photoHeight, format.photoWidth] = [format.photoWidth, format.photoHeight]
>     ;[format.videoHeight, format.videoWidth] = [format.videoWidth, format.videoHeight]
> 
>     setFormat(format)
> }
> ```
> 
> this way I got correct video preview size and correct camera output.

So this trick works for all of my test phones (Android 9-10), but for my test tablet (Lenovo TB-X505L, Android 10) I don't need to switch the values.

julianklumpers, 2021-11-24:
@viljark 
Thank you for your workaround. I was finally able to get the resolution i was looking for.
There is only one problem, the width and height are now switched.

`{"photo": {"height": 3024, "isRawPhoto": false, "metadata": {"Orientation": 6, "{Exif}": [Object], "{TIFF}": [Object]}, "path": "/data/user/0/com.cameratest/cache/mrousavy5629928989675797420.jpg", "width": 4032}}`

I would expect the height to be `4032` and the width  to be `3024`. I think this has something to do with rotation?
Did you also have this output?

ldstein, 2021-11-24:
I noticed the same issue as @viljark when I first investigated this in September. Seems like some camera sensors are mounted in portrait orientation and some in landscape orientation.  

Unfortunately, I was unable to work out how to coax this info from CameraX to intelligently flip format dimensions and had to move on to other projects.

julianklumpers, 2021-11-25:
i did some research and find out that the orientation number that comes back from the captured photo metadata, in my case `{"Orientation": 6, ...` is related to the image below.

But when switching the format width and height as @viljark suggested, the camera finally give back the correct resolution, but the metadata is not updated.

[image]
source: [https://www.impulseadventure.com/photo/exif-orientation.html](https://www.impulseadventure.com/photo/exif-orientation.html)

viljark, 2021-11-25:
@julianklumpers from my testing I have seen that you will always get the width > height (on android phones at least) and the rotation is then specified by the rotation option on the exif info (so orientation flag 6 === Rotate 90 CW). Most of the places (RN Image, browsers, file viewers) show the image correctly and take the rotation flag into account.

Since some of our logic required the image widht and height be correct without the orientation flag, I created a workaround by using expo-image-manipulator to process the image and then rewriting the image width and height.

```
import { manipulateAsync, SaveFormat } from 'expo-image-manipulator'
const photo = await cameraRef.current!.takePhoto({
	qualityPrioritization: 'speed',
	enableAutoStabilization: true,
})

const result = await manipulateAsync(photo.path, [], {
	format: SaveFormat.JPEG,
	compress: MAX_IMAGE_QUALITY,
})
photo.path = result.uri
photo.width = result.width
photo.height = result.height

```

viljark, 2021-12-14:
> So this trick works for all of my test phones (Android 9-10), but for my test tablet (Lenovo TB-X505L, Android 10) I don't need to switch the values.

Some additional testing revealed, that there is no need to swap the width and height if the camera device ise set while the phone is in landscape. 🤷

xulihang, 2022-02-17:
The default camera preview resolution I get is `864 x 480`.

I want to set the resolution to `1920 x 1080` with the following code:

```js
  const format = React.useMemo(() => {
    const desiredWidth = 1920;
    const desiredHeight = 1080;
    if (device){
      for (let index = 0; index < device.formats.length; index++) {
        const format = device.formats[index];
        console.log("h: "+format.videoHeight);
        console.log("w: "+format.videoWidth);
        if (format.videoWidth == desiredWidth && format.videoHeight == desiredHeight){
          console.log("select format: "+format);
          return format;
        }
      };
      
    }
    return undefined;
  }, [device?.formats])
```

But I only get `640x480`.

siddharth-kt, 2022-03-06:
> The default camera preview resolution I get is `864 x 480`.
>
> I want to set the resolution to `1920 x 1080` with the following code:
>
> ```js
>   const format = React.useMemo(() => {
>     const [...]

Did you got solution for this i am also facing the same issue ?

yestay90, 2022-12-29:
Just wanted to mention that switching metadata width and height is also happening on iPhone 11. Plus orientation is also 6 when taking portrait picture.

garmoshka-mo, 2023-04-07:
Photo & video sizes not configurable neither with presets nor with format props. (With format you can actually change size - but it will be random, not even close to width/height of specified format).

I had to hardcode desired resolution with patch: https://github.com/mrousavy/react-native-vision-camera/files/11178551/react-native-vision-camera%2B2.15.4.patch

mrousavy, 2023-04-07:
Hey @garmoshka-mo! That patch on iOS is not safe and might crash on a few devices.

On Android; yes, I left a few comments in the codebase warning that this is not accurate on Android due to CameraX's design. I will rewrite the android part to Camera2 in the [V3 PR](https://github.com/mrousavy/react-native-vision-camera/issues/1376), sponsor me on GitHub to support this effort. I need a lot of free time to do that.

mrousavy, 2023-08-18:
Hey! I've rewritten the entire Android codebase of VisionCamera from CameraX to Camera2 in the efforts of [✨ VisionCamera V3](https://github.com/mrousavy/react-native-vision-camera/issues/1376).

I just now completed the Camera2 rewrite and I believe the core structure is running, but there might be some edge cases to iron out. Can you try and test the PR #1674 for me to see if you can still reproduce this issue here?

Here's an instruction on how you can test that: https://github.com/mrousavy/react-native-vision-camera/pull/1674#issuecomment-1684104217

If the issue cannot be reproduced with that version/PR anymore, then hoorayy, I fixed it! 🎉 
Otherwise please let me know and I'll keep this issue open to keep track of it.

Thank you!

closed 2023-09-01 by mrousavy

mrousavy, 2023-09-30:
Hey - I'm tracking Orientation in this feature request/issue now: https://github.com/mrousavy/react-native-vision-camera/issues/1891 ✨ 

Make sure to upvote or sponsor to support this feature, and leave a comment if you have any additional thoughts/ideas.

nica0012, 2023-10-04:
@mrousavy can one go straight from v2 to v3?

mrousavy, 2023-10-05:
I think you might need to change the way you get a CameraDevice, other than that yes