margelo/react-native-vision-camera #287: 🐛 EXIF orientation

axeldelafosse, 2021-07-20:
### What were you trying to do?

I wanted to upload a picture to S3 as a blob.

### What happened instead?

The blob doesn't keep the EXIF orientation so the rotation isn't correct. 

For example:
```
const blob = await (await fetch(photo.uri)).blob();
await fetch(url, {
  method: 'PUT',
  body: blob
});
```

Note that this is working correctly when getting a picture from `expo-image-picker` for example.

### Device

iPhone 11 Pro, OnePlus 9

### VisionCamera Version

2.4.1

### Additional information

- [X] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2021-07-20:
Hey Axel 👋
So this happens both on iOS and Android?

axeldelafosse, 2021-07-20:
Hey Marc! Yes!

PR #301 (2021-07-26): feat: Support rotation

## What

* Supports phone rotation for the Preview View
* Supports phone rotation for taking pictures
* Supports phone rotation for recording views
* Supports phone rotation for the Frame Processor

## Changes

## Tested on

## Related issues

* Fixes #287 
* Fixes #213 
* Fixes #210
closed 2023-09-01 by mrousavy