margelo/react-native-vision-camera #293: ❓ How to migrate from react-native-camera where I need to read barcodes?

jacargentina, 2021-07-22:
### Question

react-native-camera says this project is the way to go, but I don't see any examples on how to get barcodes, (QR and EAN13) 

### VisionCamera Version

latest

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

chiubaka, 2021-07-23:
I'm in the same boat! Lots of the `react-native-camera` functionality seemed unreliable and buggy... the API is also a mess.

My impression is that `react-native-vision-camera` isn't quite where you're hoping for it to be yet. The way to do what you want seems to be through [Frame Processors](https://mrousavy.github.io/react-native-vision-camera/docs/guides/frame-processors/) and Frame Processor Plugins, which allow you to run arbitrary code on frames from the camera.

Frame Processors are still only available in the beta releases for Android, and as far as I can tell nobody has written a frame processor plugin for barcode scanning yet. This is a potentially urgent need for me, so I'm starting to think about implementing these myself.

chiubaka, 2021-07-23:
Frame Processors are already generally available for iOS though*. Comment about lack of a barcode scanning plugin still applies.

anhtuank7c, 2021-07-24:
@chiubaka We have to write our own logical to recognize barcode, qr code from [ImageProxy](https://github.com/mrousavy/react-native-vision-camera/blob/main/example/android/app/src/main/java/com/mrousavy/camera/example/ExampleFrameProcessorPlugin.java#L12) right?

chiubaka, 2021-07-24:
Yes, though I'd probably use one of the existing ML libraries from e.g. Google to do this.

mrousavy, 2021-07-28:
## Why it's not included in VisionCamera

Generally I want to avoid to have a `onBarcodeScanned` prop in VisionCamera as it adds a lot of added complexity and will be _slower_ than a Frame Processor because it would go over the bridge. 
Also, I'd have to merge a lot of implementation into the Android part (CameraX) since on native Android you would achieve the barcode reading the same way as you can now with VisionCamera - you add a "frame processor" (called "Image Analyser" in CameraX) and run a Barcode scanning algorithm on an `ImageProxy` object. That's the same API that VisionCamera provides, except that VisionCamera has the benefit that you can run this from the JS side, so it is more flexible.

For iOS I'm not sure if the Barcode scanner can be added **in addition to** the `video`, `photo` and `frameProcessor` use-cases, and I'd hate to make the barcode scanner a separate use-case which does not work simultaneously (so e.g. you can only use `onBarcodeScanned` and `video` but then you can't use `photo` anymore - that would suck.)

Either way, it's _really easy_ to implement a QR code scanner yourself, and it provides a LOT more flexibility and customizability (e.g. custom codes such as Snapchat SnapCodes or Apple AppClips):

## Implementing a code scanner yourself

You just need to follow the [**Creating Frame Processor Plugins**](https://mrousavy.github.io/react-native-vision-camera/docs/guides/frame-processors-plugins-overview) guide which is really small so you'll have your native frame processor plugin up and running _in a matter of a few minutes_.

So once you have the native plugin created (an empty function that just returns some value like `5` for now) and added it to `babel.config.js`, you can try calling that and use `_log` (`console.log` does not exist yet, will exist when https://github.com/mrousavy/react-native-vision-camera/pull/297 gets merged) to log the result. Unlike `console.log`, `_log` actually logs the value to the native console (Xcode Debug Area / Android Studio Logcat) - so you'll not find anything in Metro Logs (for now!).

If that works, you can start choosing your QR code scanner implementation. Here are a few choices:

### iOS

* [Apple's own **Vision** Framework](https://developer.apple.com/documentation/vision/detecting_objects_in_still_images)
* [Google's **MLKit**](https://developers.google.com/ml-kit/vision/barcode-scanning/ios)

### Android

* [**ZXing**](https://github.com/zxing/zxing/wiki/Getting-Started-Developing)
* [Google's **MLKit**](https://developers.google.com/ml-kit/vision/barcode-scanning/android)

I personally prefer MLKit because it's really easy to get started and I don't know if Apple's Vision framework provides a _synchronous_ "scan" function like Google's MLKit.

## Other

1. I consider Frame Processors _stable_ on both platforms - iOS and Android - but since a few people experience build issues with the latest beta, I still haven't promoted it to stable yet. So to use Frame Processors on Android you have to install react-native-vision-camera `2.4.2-beta.11` for now. I'll fix the build errors very soon.
2. `_log` is the only way to log something for now, `console.log` will be supported once I get https://github.com/mrousavy/react-native-vision-camera/pull/297 merged. (soon)
3. Let me know if you run into any issues, that's the only way I can collect feedback and promote the beta to stable sooner.

closed 2021-08-06 by mrousavy