margelo/react-native-vision-camera #294: FrameProcessorPlugin class not found❓

anhtuank7c, 2021-07-23:
### Question

I couldn't find the `frameprocessor` package from the module.
Do I miss something?

[image]

[image]

### VisionCamera Version

^2.4.1

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

mrousavy, 2021-07-23:
Hey!
You need to use the latest "beta" tag from npm instead of 2.4.1.
I'll promote it to stable soon.

anhtuank7c, 2021-07-23:
I couldn't found the note about version in the guide. I thought it was contains in the stable one.
@mrousavy Thank you so much.

closed 2021-07-23 by anhtuank7c