margelo/react-native-vision-camera #298: Undefined symbols for architecture x86_64:   "(extension in CoreMedia):__C.CMFormatDescriptionRef.presentationDimensions(usePixelAspectRatio: Swift.Bool, useCleanAperture: Swift.Bool) -> __C.CGSize", referenced from🐛

IamNotHuman, 2021-07-24:
### What were you trying to do?

Run the package 

### What happened instead?

Crashes on build

### Relevant log output

```shell
Undefined symbols for architecture x86_64:
  "(extension in CoreMedia):__C.CMFormatDescriptionRef.presentationDimensions(usePixelAspectRatio: Swift.Bool, useCleanAperture: Swift.Bool) -> __C.CGSize", referenced from:
      (extension in VisionCamera):__C.AVCaptureDeviceFormat.isBetterThan(__C.AVCaptureDeviceFormat) -> Swift.Bool in libVisionCamera.a(AVCaptureDevice.Format+isBetterThan.o)
      (extension in VisionCamera):__C.AVCaptureDeviceFormat.matchesFilter(__C.NSDictionary) -> Swift.Bool in libVisionCamera.a(AVCaptureDevice.Format+matchesFilter.o)
      (extension in VisionCamera):__C.AVCaptureDeviceFormat.toDictionary() -> [Swift.String : Any] in libVisionCamera.a(AVCaptureDevice.Format+toDictionary.o)
     (maybe you meant: default argument 1 of (extension in CoreMedia):__C.CMFormatDescriptionRef.presentationDimensions(usePixelAspectRatio: Swift.Bool, useCleanAperture: Swift.Bool) -> __C.CGSize, default argument 0 of (extension in CoreMedia):__C.CMFormatDescriptionRef.presentationDimensions(usePixelAspectRatio: Swift.Bool, useCleanAperture: Swift.Bool) -> __C.CGSize )
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
```

### Device

Iphone 11 pro (IOS 14.6), Simulator Iphone 11

### VisionCamera Version

^2.4.1

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2021-07-24:
Did you follow step 3 in the troubleshooting guide?

IamNotHuman, 2021-07-24:
Thanks for the quick response did not realize I had swift 5.0. Thanks

mrousavy, 2021-07-26:
Yes you need Swift 5.2 to build. I also often get confused when I see that build error 😅

closed 2021-07-26 by mrousavy