margelo/react-native-vision-camera #311: How do I enable/disable the shutter sound❓

jzxchiang1, 2021-07-31:
### Question

How do I enable or disable the shutter sound when taking a photo?

For some reason, when testing on my physical iPhone device, the sound plays when flash is off but doesn't play when flash is on.

Is this customizable? Thanks!

### VisionCamera Version

2.4.1

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

mrousavy, 2021-07-31:
The shutter sound is always played in the image capture pipeline:

[image]

But it looks like this actually can be customized by disposing the sound (see [this SO post](https://stackoverflow.com/a/55235949/5281431)). I'll mark this as a feature request and get to this soon

mrousavy, 2022-01-04:
yeah it's pretty easily possible - will take a look soon. Does the shutter sound also play on Android?

PABourdais, 2022-01-04:
On android native camera there is a shutter sound

siddharth-kt, 2022-01-26:
any update on this guys ?

JurianBoots, 2022-02-04:
Hey @mrousavy, any update? This feature would be awesome! :)

mrousavy, 2022-02-04:
Hey @JurianBoots, unfortunately not yet. I've had higher priorities in the past few days, such as my [agency Margelo](https://margelo.io), as well as other issues/PRs in VisionCamera/MMKV created by [my GitHub sponsors](https://github.com/sponsors/mrousavy).

wassgha, 2022-05-25:
Could we please prioritize either this, or taking snapshots on iOS? I'm trying to make blurry previews of the camera, ideally this would be a low-quality snapshot but since that's not available on iOS I resorted to taking photos which plays the shutter sound even though the user did not trigger the photo.

mrousavy, 2022-05-27:
> Could we please prioritize either this

Sure, you can prioritize this. Shoot me a PR if you have a working change.

wilav-dev, 2022-07-26:
Hello, any update?

zymronak, 2022-10-04:
I found a way to disable the shutter should

1. open PhotoCaptureDelegate
2. find func photoOutput
3. add given line into the function and rebuild the app ```
// dispose system shutter sound
    AudioServicesDisposeSystemSoundID(1108)
```

RONAKDHOLARIYA, 2022-10-04:
I found a way to disable the shutter should 1. open PhotoCaptureDelegate 2. find func photoOutput 3. add given line into the function and rebuild the app ``` // dispose system shutter sound AudioServicesDisposeSystemSoundID(1108) ```

mrousavy, 2023-08-21:
Hey! I just implemented this in V3! :) https://github.com/mrousavy/react-native-vision-camera/pull/1702

closed 2023-09-01 by mrousavy

jagdeepChicmic, 2023-09-21:
file shoes merged but code not reflected in 2.16.1 version(for ios)

mrousavy, 2023-12-05:
Hey - no, this is a V3 feature. I'm not adding any more features to V2, as I already spend a lot of time on V3 and I don't have any more free time to dedicate to V2.