margelo/react-native-vision-camera #312: 🐛  Stale closure when using onRecordingFinished

Kypsis, 2021-07-31:
### What were you trying to do?

When the recording is stopped with `camera.current.stopRecording();` the `onRecordingFinished` triggers that was declared in `camera.current.startRecording({onRecordingFinished: (video: VideoFile) => void;})` but the closure of that function is from when the recording started, not when it finished. Thus making the usage of `onRecordingFinished` much more difficult for post recording actions with the current state. 

### What happened instead?

Using the familiar counter example, if `count` was 1 when recording started and 6 when recording stopped, then on triggering `onRecordingFinished` it will be 1 instead of 6.

### Device

Pixel 4a

### VisionCamera Version

Current master of react-native-vision-camera and the example

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2021-07-31:
I see what the problem is, I'm just not sure how an alternative API would look like. You need to pass the `onRecordingFinished` callback in there to catch any capture errors.

You could instead use a ref (`useRef`) and access `.current` inside the `onRecordingFinished` closure.

Kypsis, 2021-07-31:
Good point about using `useRef` and that solves the staleness albeit in a roundabout way (imo). Still running into a stale closure in a library code was unexpected and took more time to pinpoint than I care to admit (first instinct is to suspect your own code, not external code).

I think the issue can be closed if there is a notice somewhere warning about the staleness because it's not an obvious thing to expect. Or a better solution comes up. 🙂

mrousavy, 2021-08-01:
I agree this can be confusing, however this is just how the JavaScript language works (and a side-effect of how the React Hooks/Function component API works) so I'm not going to mention this in VisionCamera docs.

Nice that this fixed your problem! 👍

closed 2021-08-01 by mrousavy