margelo/react-native-vision-camera #313: ✨ OCR support

404-html, 2021-08-01:
### What feature or enhancement are you suggesting?

VisionCamera can be more vision with [optical character recognition](https://en.wikipedia.org/wiki/Optical_character_recognition).
Probably the best is to do this as a [Frame Processor Plugin](https://mrousavy.github.io/react-native-vision-camera/docs/guides/frame-processors-plugins-overview). What I would like to get is an array of `blocks` of recognized text with strings and coordinates, e.g. `[{ value: 'recognized text', coords: { x: 100, y: 100 } }, ...]`.

### What Platforms whould this feature/enhancement affect?

iOS, Android

### Alternatives/Workarounds

Currently using [RNCamera](https://react-native-camera.github.io/react-native-camera/docs/rncamera#ontextrecognized) for that.

### Additional information

- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar feature requests in this repository](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22✨+enhancement%22) and found none.

mrousavy, 2021-08-01:
As you mentioned in the issue description - this can be done as a [Frame Processor Plugin](https://mrousavy.github.io/react-native-vision-camera/docs/guides/frame-processors-plugins-overview/). If you need any help creating frame processor plugins, feel free to ask ahead! It's a really easy process. 👍

closed 2021-08-01 by mrousavy

Noitidart, 2021-08-29:
I'm curious if you get this done @404-html - I couldn't figure it out so I used https://github.com/agoldis/react-native-mlkit-ocr

404-html, 2021-08-29:
Sorry @Noitidart, no progress from my side, I might give it a go when I will have some time but hard to say when it will be.
I believe this could be done the same way as in https://github.com/react-native-camera/react-native-camera - using Firebase ML kit for OCR.

rodgomesc, 2021-08-30:
This is very easy to implement I will try make a functional version until
the end of this week, please keep following this thread comments

Noitidart, 2021-08-30:
Thank you @rodgomesc!!

tushgup, 2021-09-12:
@rodgomesc Thank you for your work on adding OCR support to this library!! I was wondering if you were planning on leveraging the live text API launched in iOS 15 in your implementation?

rodgomesc, 2021-09-14:
sorry for the delay to implement this, i'm very busy with other projects but i want to implement ocr for android/ios as soon as possible

rodgomesc, 2021-09-14:
@tushgup  sure, android 21/30 and ios 9.0/15

jolic, 2022-03-25:
Is this feature implemented?