margelo/react-native-vision-camera #315: 🐛  device?.supportsParallelVideoProcessing is true but library fails to init frameProcessor

Kypsis, 2021-08-02:
### What were you trying to do?

I wanted to use frameprocessor form the current version of react-native-vision-camera/example. 

### What happened instead?

`device?.supportsParallelVideoProcessing` returns true but the frameProcessor fails to initialize with the following error:
`[device/parallel-video-processing-not-supported: [device/parallel-video-processing-not-supported] The given LEGACY Camera device does not support parallel video processing (`video={true}` + `frameProcessor={...}`). Disable either `video` or `frameProcessor`. To find out if a device supports parallel video processing, check the `supportsParallelVideoProcessing` property on the CameraDevice. See https://mrousavy.github.io/react-native-vision-camera/docs/guides/lifecycle#the-supportsparallelvideoprocessing-prop for more information.]`

### Relevant log output

```shell
2021-08-02 12:10:51.419 2057-2210/com.mrousavy.camera.example I/VisionCamera: Initializing FrameProcessorRuntimeManager...
2021-08-02 12:10:51.419 2057-2210/com.mrousavy.camera.example I/VisionCamera: Initializing Vision JS-Runtime...
2021-08-02 12:10:51.422 2057-2210/com.mrousavy.camera.example I/VisionCamera: Initialized Vision JS-Runtime!
2021-08-02 12:10:51.422 2057-2210/com.mrousavy.camera.example I/FrameProcessorRuntime: Installing Frame Processor Plugins...
2021-08-02 12:10:51.422 2057-2210/com.mrousavy.camera.example I/VisionCamera: Installing Frame Processor Plugin "__example_plugin"...
2021-08-02 12:10:51.422 2057-2210/com.mrousavy.camera.example I/FrameProcessorRuntime: Successfully installed 1 Frame Processor Plugins!
2021-08-02 12:10:51.552 2057-2205/com.mrousavy.camera.example I/ReactNativeJNI: received didPause for reason: 0 in state: RunningDetached
2021-08-02 12:10:51.653 2057-2205/com.mrousavy.camera.example D/SoLoader: About to load: libreactnativeblob.so
2021-08-02 12:10:51.654 2057-2205/com.mrousavy.camera.example D/SoLoader: libreactnativeblob.so not found on /data/data/com.mrousavy.camera.example/lib-main
2021-08-02 12:10:51.654 2057-2205/com.mrousavy.camera.example D/SoLoader: libreactnativeblob.so found on /data/app/~~Yf7ywxPn_Ow2NM3XralC6Q==/com.mrousavy.camera.example-RTQSI8aRZxUVTrUL7tpKWg==/lib/arm64
2021-08-02 12:10:51.654 2057-2205/com.mrousavy.camera.example D/SoLoader: Not resolving dependencies for libreactnativeblob.so
2021-08-02 12:10:51.655 2057-2205/com.mrousavy.camera.example D/SoLoader: Loaded: libreactnativeblob.so
2021-08-02 12:10:51.684 2057-2206/com.mrousavy.camera.example I/WebViewFactory: Loading com.google.android.webview version 92.0.4515.115 (code 451511533)
2021-08-02 12:10:51.706 2057-2206/com.mrousavy.camera.example W/.camera.exampl: Accessing hidden method Landroid/os/Trace;->isTagEnabled(J)Z (greylist, reflection, allowed)
2021-08-02 12:10:51.706 2057-2206/com.mrousavy.camera.example W/.camera.exampl: Accessing hidden method Landroid/os/Trace;->traceBegin(JLjava/lang/String;)V (greylist, reflection, allowed)
2021-08-02 12:10:51.706 2057-2206/com.mrousavy.camera.example W/.camera.exampl: Accessing hidden method Landroid/os/Trace;->traceEnd(J)V (greylist, reflection, allowed)
2021-08-02 12:10:51.706 2057-2206/com.mrousavy.camera.example W/.camera.exampl: Accessing hidden method Landroid/os/Trace;->asyncTraceBegin(JLjava/lang/String;I)V (greylist, reflection, allowed)
2021-08-02 12:10:51.706 2057-2206/com.mrousavy.camera.example W/.camera.exampl: Accessing hidden method Landroid/os/Trace;->asyncTraceEnd(JLjava/lang/String;I)V (greylist, reflection, allowed)
2021-08-02 12:10:51.709 2057-2206/com.mrousavy.camera.example I/cr_WVCFactoryProvider: Loaded version=92.0.4515.115 minSdkVersion=29 isBundle=true multiprocess=true packageId=2
2021-08-02 12:10:51.724 2057-2206/com.mrousavy.camera.example I/cr_LibraryLoader: Successfully loaded native library
2021-08-02 12:10:51.725 2057-2206/com.mrousavy.camera.example I/cr_CachingUmaRecorder: Flushed 7 samples from 7 histograms.
2021-08-02 12:10:52.007 2057-2205/com.mrousavy.camera.example I/VisionCamera: Installing JSI bindings...
2021-08-02 12:10:52.007 2057-2205/com.mrousavy.camera.example I/VisionCamera: Finished installing JSI bindings!
2021-08-02 12:10:52.050 1753-2127/? D/ConnectivityService: NetReassign [no changes]
2021-08-02 12:10:52.431 2057-2057/com.mrousavy.camera.example D/SoLoader: About to load: libyoga.so
2021-08-02 12:10:52.432 2057-2057/com.mrousavy.camera.example D/SoLoader: libyoga.so not found on /data/data/com.mrousavy.camera.example/lib-main
2021-08-02 12:10:52.432 2057-2057/com.mrousavy.camera.example D/SoLoader: libyoga.so found on /data/app/~~Yf7ywxPn_Ow2NM3XralC6Q==/com.mrousavy.camera.example-RTQSI8aRZxUVTrUL7tpKWg==/lib/arm64
2021-08-02 12:10:52.432 2057-2057/com.mrousavy.camera.example D/SoLoader: Not resolving dependencies for libyoga.so
2021-08-02 12:10:52.433 2057-2057/com.mrousavy.camera.example D/SoLoader: Loaded: libyoga.so
2021-08-02 12:10:52.438 2057-2205/com.mrousavy.camera.example I/ReactNativeJS: Running "CameraPage" with {"initialProps":{"componentId":"Component1"},"rootTag":1}
2021-08-02 12:10:52.562 1753-1753/? W/WindowManager: removeWindowToken: Attempted to remove non-existing token: android.os.Binder@a49ed1e
2021-08-02 12:10:52.640 2057-2205/com.mrousavy.camera.example I/ReactNativeJS: re-rendering camera page without active camera
2021-08-02 12:10:52.824 2057-2057/com.mrousavy.camera.example I/ReactNative: [GESTURE HANDLER] Initialize gesture handler for root view com.reactnativenavigation.react.ReactView{90fcc62 V.E...... ......ID 0,0-1080,2296 #1}
2021-08-02 12:10:52.841 2057-2057/com.mrousavy.camera.example D/ExtenderVersion: No versioning extender found. Falling back to default.
2021-08-02 12:10:52.859 2057-2228/com.mrousavy.camera.example I/CameraManagerGlobal: Connecting to camera service
2021-08-02 12:10:52.861 1151-2072/? W/ServiceManager: Permission failure: android.permission.CAMERA_OPEN_CLOSE_LISTENER from uid=10378 pid=2057
2021-08-02 12:10:52.875 2057-2228/com.mrousavy.camera.example D/CameraRepository: Added camera: 0
2021-08-02 12:10:52.891 2057-2228/com.mrousavy.camera.example I/Camera2CameraInfo: Device Level: INFO_SUPPORTED_HARDWARE_LEVEL_FULL
2021-08-02 12:10:52.892 2057-2228/com.mrousavy.camera.example D/CameraRepository: Added camera: 1
2021-08-02 12:10:52.893 2057-2228/com.mrousavy.camera.example I/Camera2CameraInfo: Device Level: INFO_SUPPORTED_HARDWARE_LEVEL_FULL
2021-08-02 12:10:52.893 2057-2228/com.mrousavy.camera.example D/CameraValidator: Verifying camera lens facing on sunfish, lensFacingInteger: null
2021-08-02 12:10:53.014 2057-2057/com.mrousavy.camera.example W/CameraView: CameraViewModule::getAvailableCameraDevices took: 226 ms
2021-08-02 12:10:53.060 2057-2205/com.mrousavy.camera.example I/ReactNativeJS: Re-rendering camera page with active camera. Device: "back (0)" (4000x2000 @ 30fps)
2021-08-02 12:10:53.094 1003-1003/? I/sensors-hal: [hal]flush_physical_sensor:320, android.sensor.accelerometer/11
2021-08-02 12:10:53.094 1003-1003/? I/sensors-hal: [hal]flush_physical_sensor:326, android.sensor.accelerometer/11 completed
2021-08-02 12:10:53.095 2057-2057/com.mrousavy.camera.example D/CameraView: Lifecycle went from INITIALIZED -> CREATED (isActive: true | isAttachedToWindow: false)
2021-08-02 12:10:53.129 2057-2205/com.mrousavy.camera.example I/ReactNativeJS: Re-rendering camera page with active camera. Device: "back (0)" (4000x2000 @ 30fps)
2021-08-02 12:10:53.165 2057-2057/com.mrousavy.camera.example D/CameraView: Lifecycle went from CREATED -> RESUMED (isActive: true | isAttachedToWindow: true)
2021-08-02 12:10:53.176 3308-3332/? D/DeviceStateHelper: Audo mode: 0
2021-08-02 12:10:53.178 3308-3332/? D/DeviceStateHelper: Audo mode: 0
2021-08-02 12:10:53.185 2057-2057/com.mrousavy.camera.example D/CameraView: Lifecycle went from RESUMED -> RESUMED (isActive: true | isAttachedToWindow: true)
2021-08-02 12:10:53.185 2057-2057/com.mrousavy.camera.example I/CameraView: Configuring session...
2021-08-02 12:10:53.186 2057-2057/com.mrousavy.camera.example I/CameraView: Configuring session with Camera ID 0 and custom format...
2021-08-02 12:10:53.192 2057-2057/com.mrousavy.camera.example I/CameraView: Using custom format - photo: 1920x1080, video: 1920x1080 @ 30 FPS
2021-08-02 12:10:53.192 2057-2057/com.mrousavy.camera.example I/CameraView: Setting AE_TARGET_FPS_RANGE to 30-30, and SENSOR_FRAME_DURATION to 0
2021-08-02 12:10:53.192 2057-2057/com.mrousavy.camera.example I/CameraView: Adding VideoCapture use-case...
[... cut ...]
Pixel 4a

### VisionCamera Version

latest master

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2021-08-02:
Hi! Thanks for the bug report. Could you try opening Android Studio and place a breakpoint in that line:

https://github.com/mrousavy/react-native-vision-camera/blob/43f5a735a5459c7dad70ee8f5bc6eada0efc6cd2/android/src/main/java/com/mrousavy/camera/CameraView.kt#L138

Then as soon as the breakpoint gets hit, please tell me what value the `hardwareLevel` value has.

mrousavy, 2021-08-02:
You're just running the `example/` right? Everything works fine if you disable `photo` and/or `video`?

Kypsis, 2021-08-02:
[image]

I'm indeed only running the example and it only works if I disable `video`. Disabling `photo` or `audio` or any combination not `video` does not work.

naqibhishamuddin, 2021-09-14:
Hi, @mrousavy @Kypsis are there any updates on this issue 🐛 ? Thank you 🙌🏻

mrousavy, 2021-09-14:
Hi! Sorry, no update from my side, I've been really busy with contracting. I'd love to review and merge a PR that fixes this, not entirely sure what's causing this but I can give pointers if you want to debug this.

HorbachAndrii, 2022-02-22:
I have the same issue for Android 11 Xiaomi Redmi Note 8 Pro

HorbachAndrii, 2022-02-22:
@Kypsis did you resolve the problem?

TuJobs, 2022-06-02:
+1 same issue for Oppo A1k

TuJobs, 2022-06-04:
+1 samsung s22 ultral
+1 samsung s22+
i think samsung device have this problem.

HorbachAndrii, 2022-06-04:
@TuJobs what react-native version are you using?

TuJobs, 2022-06-06:
@HorbachAndrii it's react-native": "0.68.2

TensionCoding, 2022-10-18:
I am running into this same problem on a Google Pixel 6a with the example app. Has there been any movement on this?

venura-atlasopen, 2023-01-02:
@TuJobs @TensionCoding did you resolve the problem? I am having it running the example app.

mrousavy, 2023-08-18:
Hey! I've rewritten the entire Android codebase of VisionCamera from CameraX to Camera2 in the efforts of [✨ VisionCamera V3](https://github.com/mrousavy/react-native-vision-camera/issues/1376).

I just now completed the Camera2 rewrite and I believe the core structure is running, but there might be some edge cases to iron out. Can you try and test the PR #1674 for me to see if you can still reproduce this issue here?

Here's an instruction on how you can test that: https://github.com/mrousavy/react-native-vision-camera/pull/1674#issuecomment-1684104217

If the issue cannot be reproduced with that version/PR anymore, then hoorayy, I fixed it! 🎉 
Otherwise please let me know and I'll keep this issue open to keep track of it.

Thank you!

closed 2023-09-01 by mrousavy