margelo/react-native-vision-camera #323: 🐛  Preset prop has no effect on Android

Kypsis, 2021-08-04:
### What were you trying to do?

Setting the preset has no effect on recording quality on Android. Whatever preset is chosen, recorded quality is always 1920*1080 FullHD. This also somewhat relatedly raises the question is the frameBuffer, surface(preview) and recording the same size. Sidenote, I remember being able to change each of those to be independent in Flutter with a library hack (preview on Android was capped to 1280*720 max for whatever reason).

### What happened instead?

Nothing happens :(

### Device

Pixel 4a

### VisionCamera Version

latest master

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2021-08-04:
`preset` is iOS only. To change the recording size, use a custom `format`

Kypsis, 2021-08-05:
The format works, kinda, even if it's somewhat clunky to use. There is an issue with format though that no matter the resolution picked the recorded file size always ends up at ca 30MB. Also even though the preview seems to respect higher resolutions (gets that zoomed in effect if using 4:3 formats) the recording itself seems to be capped at FullHD (1920x1080). Using 30 fps and yuv in all of the tests.

mrousavy, 2021-08-05:
> even if it's somewhat clunky to use

why?

> recording itself seems to be capped at FullHD

I think that's an issue by the native CameraX library from Google, maybe the recording doesn't support changing resolution yet? You could play around with the following code:

https://github.com/mrousavy/react-native-vision-camera/blob/1c32726ef6e38fed1dfeca2141e8c18f5bd1e23a/android/src/main/java/com/mrousavy/camera/CameraView.kt#L359-L362

mrousavy, 2021-08-05:
working on a fix here: https://github.com/mrousavy/react-native-vision-camera/pull/325

Kypsis, 2021-08-05:
That actually works to unlock proper aspect ratios for recordings. Still this:
```js
format={
              device?.formats?.filter(
                format =>
                  format.videoWidth === 4032 &&
                  format.videoHeight === 3024 &&
                  format.colorSpaces.includes("yuv"),
              )[0]
            }
```
records at 1600x1200 max so might indeed be CameraX issue. As for presets on android vs just only format, ease of use and sensible defaults are always compelling use cases🙂

mrousavy, 2021-08-05:
> As for presets on android vs just only format, ease of use and sensible defaults are always compelling use cases🙂

`preset` is a property of the native iOS camera library only. Android's CameraX library does not have this feature, if you want support for this feature you can implement some sort of preset resolver/mapper for Android yourself, VisionCamera is an open source project I work on in my free time

mrousavy, 2021-08-05:
I'll try to find out why CameraX is locked to 1920x1080 🤔

Kypsis, 2021-08-05:
> `preset` is a property of the native iOS camera library only. Android's CameraX library does not have this feature, if you want support for this feature you can implement some sort of preset [...]

I was kinda afraid that is the case. Still it's not a showstopper or anything just a nice convenience feature. 

As for the other, it is not locked to 1920x1080 anymore. Just the upper range got capped to 1920x1080 previously (now the upper range at least follows aspect ratio of the given format, so 4032x3024 is 1600x 1200). You could still go lower (even before the proposed change) like 1280x720 and that would work.

mrousavy, 2021-08-05:
btw instead of `filter(x => true)[0]` you can use `find(x => true)`, which is less code, cleaner to read and faster. I'd still use `useMemo` to memoize this and avoid running this filtering on every re-render.

Oh btw you're running VisionCamera from master right? Could you please go into `CameraPage.tsx` and disable `photo` and `video` and always enable the `frameProcessor` and let me know if the app works and the Frame Processor logs to the Metro Bundler console?

Kypsis, 2021-08-05:
I'm actually running vision camera on a feature branch of a production app and both the VisionCamera master. But will try asap.

Kypsis, 2021-08-05:
[image]
Seems to work okay.

mrousavy, 2021-08-05:
wow, amazing. Think I can promote the `beta` to a stable release now?

I also think it's amazing that your camera says 60 FPS since I thought CameraX only supports 30 FPS? 🤨

Kypsis, 2021-08-05:
What format is selected and what actually gets recorded is kinda weird though. 320x240 format => recording 352x288, 640x480 => 1024x768, 1280x720 => 1600x1200, 1920x1080 => 1600x1200, no format(default) = >1920x1080. All formats yuv. 
Also quite the data rate difference:
[image]
[image]

mrousavy, 2021-08-05:
oh yeah I see - that's because CameraX doesn't expose the video width/height in a format - everything's loosely coupled. I'll try to open a feature request for CameraX, but this is out of my hands

Kypsis, 2021-08-05:
I reverted the [#325](https://github.com/mrousavy/react-native-vision-camera/pull/325) changes just to verify, but without #325 changes the 1280x720 format => 
[image]
Which is interesting because after #325 you get 1600x1200. CameraX certainly has bugs a plenty.

mrousavy, 2023-08-18:
Hey! I've rewritten the entire Android codebase of VisionCamera from CameraX to Camera2 in the efforts of [✨ VisionCamera V3](https://github.com/mrousavy/react-native-vision-camera/issues/1376).

I just now completed the Camera2 rewrite and I believe the core structure is running, but there might be some edge cases to iron out. Can you try and test the PR #1674 for me to see if you can still reproduce this issue here?

Here's an instruction on how you can test that: https://github.com/mrousavy/react-native-vision-camera/pull/1674#issuecomment-1684104217

If the issue cannot be reproduced with that version/PR anymore, then hoorayy, I fixed it! 🎉 
Otherwise please let me know and I'll keep this issue open to keep track of it.

Thank you!

PR #325 (2021-08-05): fix: Fix `format` Preview and Video Capture resolution setting

## What

## Changes

## Tested on

## Related issues

* _Maybe_ fixes #323
closed 2023-09-01 by mrousavy