margelo/react-native-vision-camera #338: 🐛  Android build fails on CI

Kypsis, 2021-08-06:
### What were you trying to do?

Tried to have the 2.5.0 build Android on CI (Bitrise). Have to mention this is the first time trying to build react-native-vision-camera on CI. IOS builds went fine.

### What happened instead?

Build fails with:
`ninja: error: '../../../../build/third-party-ndk/hermes/jni/armeabi-v7a/libhermes.so', needed by '../../../../build/intermediates/cmake/release/obj/armeabi-v7a/libVisionCamera.so', missing and no known rule to make it`

Doesn't like relative paths?

### Relevant log output

```shell
> Task :react-native-vision-camera:generateJsonModelRelease
/Users/vagrant/git/node_modules/react-native-vision-camera/android/CMakeLists.txt : C/C++ release[REDACTED]armeabi-v7a : CMake Warning at /Users/vagrant/git/node_modules/react-native-vision-camera/android/CMakeLists.txt:122 (message):
  VisionCamera linking: FOR_HERMES=true
/Users/vagrant/git/node_modules/react-native-vision-camera/android/CMakeLists.txt : C/C++ release[REDACTED]armeabi-v7a : CMake Warning at /Users/vagrant/git/node_modules/react-native-vision-camera/android/CMakeLists.txt:122 (message):
  VisionCamera linking: FOR_HERMES=true
/Users/vagrant/git/node_modules/react-native-vision-camera/android/CMakeLists.txt : C/C++ release[REDACTED]arm64-v8a : CMake Warning at /Users/vagrant/git/node_modules/react-native-vision-camera/android/CMakeLists.txt:122 (message):
  VisionCamera linking: FOR_HERMES=true
/Users/vagrant/git/node_modules/react-native-vision-camera/android/CMakeLists.txt : C/C++ release[REDACTED]arm64-v8a : CMake Warning at /Users/vagrant/git/node_modules/react-native-vision-camera/android/CMakeLists.txt:122 (message):
  VisionCamera linking: FOR_HERMES=true
/Users/vagrant/git/node_modules/react-native-vision-camera/android/CMakeLists.txt : C/C++ release[REDACTED]x86 : CMake Warning at /Users/vagrant/git/node_modules/react-native-vision-camera/android/CMakeLists.txt:122 (message):
  VisionCamera linking: FOR_HERMES=true
/Users/vagrant/git/node_modules/react-native-vision-camera/android/CMakeLists.txt : C/C++ release[REDACTED]x86 : CMake Warning at /Users/vagrant/git/node_modules/react-native-vision-camera/android/CMakeLists.txt:122 (message):
  VisionCamera linking: FOR_HERMES=true
/Users/vagrant/git/node_modules/react-native-vision-camera/android/CMakeLists.txt : C/C++ release[REDACTED]x86_64 : CMake Warning at /Users/vagrant/git/node_modules/react-native-vision-camera/android/CMakeLists.txt:122 (message):
  VisionCamera linking: FOR_HERMES=true
/Users/vagrant/git/node_modules/react-native-vision-camera/android/CMakeLists.txt : C/C++ release[REDACTED]x86_64 : CMake Warning at /Users/vagrant/git/node_modules/react-native-vision-camera/android/CMakeLists.txt:122 (message):
  VisionCamera linking: FOR_HERMES=true
> Task :react-native-vision-camera:prepareHermes UP-TO-DATE
> Task :react-native-vision-camera:prepareJSC
> Task :react-native-vision-camera:createNativeDepsDirectories UP-TO-DATE
> Task :react-native-vision-camera:downloadBoost
Download https://github.com/react-native-community/boost-for-react-native/releases/download/v1.63.0-0/boost_1_63_0.tar.gz
> Task :react-native-vision-camera:prepareBoost
> Task :react-native-vision-camera:downloadDoubleConversion
Download https://github.com/google/double-conversion/archive/v1.1.6.tar.gz
> Task :react-native-vision-camera:prepareDoubleConversion
> Task :react-native-vision-camera:downloadFolly
Download https://github.com/facebook/folly/archive/v2020.01.13.00.tar.gz
> Task :react-native-vision-camera:prepareFolly
> Task :react-native-vision-camera:downloadGlog
Download https://github.com/google/glog/archive/v0.3.5.tar.gz
> Task :react-native-vision-camera:prepareGlog
> Task :react-native-vision-camera:prepareThirdPartyNdkHeaders
> Task :react-native-vision-camera:externalNativeBuildRelease FAILED
Build VisionCamera_armeabi-v7a
ninja: Entering directory `/Users/vagrant/git/node_modules/react-native-vision-camera/android/.cxx/cmake/release/armeabi-v7a'
ninja: error: '../../../../build/third-party-ndk/hermes/jni/armeabi-v7a/libhermes.so', needed by '../../../../build/intermediates/cmake/release/obj/armeabi-v7a/libVisionCamera.so', missing and no known rule to make it
FAILURE: Build failed with an exception.
* What went wrong:
Execution failed for task ':react-native-vision-camera:externalNativeBuildRelease'.
> Build command failed.
  Error while executing process /usr/local/share/android-sdk/cmake/3.10.2.4988404/bin/ninja with arguments {-C /Users/vagrant/git/node_modules/react-native-vision-camera/android/.cxx/cmake/release/armeabi-v7a VisionCamera}
  ninja: Entering directory `/Users/vagrant/git/node_modules/react-native-vision-camera/android/.cxx/cmake/release/armeabi-v7a'
  
  ninja: error: '../../../../build/third-party-ndk/hermes/jni/armeabi-v7a/libhermes.so', needed by '../../../../build/intermediates/cmake/release/obj/armeabi-v7a/libVisionCamera.so', missing and no known rule to make it
* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.
* Get more help at https://help.gradle.org
Deprecated Gradle features were used in this build, making it incompatible with Gradle 7.0.
Use '--warning-mode all' to show the individual deprecation warnings.
See https://docs.gradle.org/6.7.1/userguide/command_line_interface.html#sec:command_line_warnings
BUILD FAILED in 11m 55s
651 actionable tasks: 651 executed
Build task failed, error: exit status 1
```

### Device

Default Bitrise build (Xcode 12.4.x, on macOS 10.15.7 (Catalina))

### VisionCamera Version

2.5.0

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2021-08-06:
I believe the build works if you run it twice, I just wonder why it doesn't work the first time 🤨

Kypsis, 2021-08-06:
That actually works 🤨  How did you even come up with that solution...

jehartzog, 2021-08-13:
Seeing this as well with `2.5.0` on our CircleCI build. Also saw it locally, had to run twice to get through successfully. I tried to do some hackyness with fastlane to get it to build twice:

```rb
  lane :staging do
    begin
      gradle(
        task: "app:assembleStagingrelease"
      )
    rescue
      gradle(
        task: "app:assembleStagingrelease"
      )
    end
```

The build made it further, but then I ran into a new error I couldn't find a way around:

```
More than one file was found with OS independent path 'lib/x86/libfbjni.so'
```

For now we're doing manual Android builds. Happy to help assist with debugging this 😀 .

mrousavy, 2021-08-14:
@jehartzog as for the `More than one file was found with OS independent path ...` error, take a look at [this stackoverflow post](https://stackoverflow.com/questions/44342455/more-than-one-file-was-found-with-os-independent-path-meta-inf-license). This is not an issue with VisionCamera, but with how Android and Gradle works.

 Add this to your `build.gradle`:

```groovy
android {

    ...

    packagingOptions {
        pickFirst '**/*.so'
    }
}
```

I still wonder why the build has to be ran twice

damjtoh, 2021-08-15:
I'm having the same problem 
```
  ninja: error: '../../../../build/third-party-ndk/hermes/jni/armeabi-v7a/libhermes.so', needed by '../../../../build/intermediates/cmake/release/obj/armeabi-v7a/libVisionCamera.so', missing and no known rule to make it
```
But Bitrise is not letting me re-run the step. So I'm at a dead-end for now.

jehartzog, 2021-08-15:
@damjtoh I had the same issue w/ CircleCI. If you use fastlane, try [this ugly hack](https://github.com/mrousavy/react-native-vision-camera/issues/338) to attempt the build twice on the same instance. I also ended up needing [this fix](https://github.com/mrousavy/react-native-vision-camera/issues/338#issuecomment-898867699), but at least our build is back up and running for now.

RayVR, 2021-08-16:
> @jehartzog as for the More than one file was found with OS independent path ... error, take a look at this stackoverflow post. This is not an issue with VisionCamera, but with how Android and Gradle works.

@mrousavy I'm curious if there is a better solution since react-native-vision-camera did not package all of these .so files prior to 2.5.0. With RNVC@2.4.1 I do no need to worry about all of these conflicting .so files and no other dependency in my project (of which there are many) distributes their own .so files for these libs.

mrousavy, 2021-08-16:
@damjtoh 

> So I'm at a dead-end for now.

You could take a look at the `build.gradle`, play around with configurations and when you fix the issue create a PR. Then the problem is fixed.

mrousavy, 2021-08-16:
@RayVR 

> I'm curious if there is a better solution since react-native-vision-camera did not package all of these .so files prior to 2.5.0. With RNVC@2.4.1 I do no need to worry about all of these conflicting [...]

yes, because 2.4.1 didn't have any C++ code in the Android codebase because that didn't support Frame Processors yet. 2.5.0 comes with support for Frame Processors on Android, which requires C++ code and therefore I need to extract and link the `.so` files from react-native. You can take a look at the `build.gradle` of VisionCamera and if you find a better solution create a PR and I will happily merge this 🤗

wanderSX, 2021-08-16:
Build also fails on AppCenter

mrousavy, 2021-08-16:
> Build also fails on AppCenter

okay

ribeirogabriel, 2021-08-20:
@Kypsis A solution that I found, more like a workaround was putting all the gradle tasks related to the building process of vision camera into a single script on package.json and running it before the Ci build, like so:

`"android:prepareNativeDeps": "cd android && ./gradlew :react-native-vision-camera:prepareJSC && ./gradlew :react-native-vision-camera:prepareHermes && ./gradlew :react-native-vision-camera:extractJNIFiles && ./gradlew :ribeirogabriel_react-native-vision-camera:extractAARHeaders && ./gradlew :eact-native-vision-camera:prepareThirdPartyNdkHeaders"`

This solves the issue of the libhermes.so file not being available at the time when libCameraVision.so needs it. 

We still need to see what we can do on the gradle side to fix it, I'll try to take a look on it in the following days

mrousavy, 2021-08-20:
Nice workaround! It surely must be possible to run those preparations in the build.gradle file before the native side is being built though, thanks for taking time to investigate this @ribeirogabriel ! 🖤

mikaoelitiana, 2021-09-01:
@ribeirogabriel there is a small typo on the script `&& ./gradlew :eact-native-vision-camera:prepareThirdPartyNdkHeaders"` should be `&& ./gradlew :react-native-vision-camera:prepareThirdPartyNdkHeaders"`

mikaoelitiana, 2021-09-01:
@wanderSX  did you find a workaround for appcenter?

wanderSX, 2021-09-01:
@mikaoelitiana not yet. Adding `yarn android:prepareNativeDeps` to pre-build script on appCenter helped with 
`ninja: error: '../../../../build/third-party-ndk/hermes/jni/armeabi-v7a/libhermes.so', needed by '../../../../build/intermediates/cmake/release/obj/armeabi-v7a/libVisionCamera.so', missing and no known rule to make it` from OP. But now i'm stuck with this issue https://github.com/mrousavy/react-native-vision-camera/issues/388

mrousavy, 2021-09-04:
@bear-junior created a PR to fix this here: #407 🎉 
can anyone here confirm that the PR (#407) fixes this issue?

bear-junior, 2021-09-05:
definitely this pre-build script is not necessary, because if you run this as a pre-build, then after that the build will start with `clean` command on the CI and will remove everything from the **build** folder (location for all of those tasks). In this case libHermes.so is missing, because prepareHermes tries to add file to **build/third-party-ndk** which probably doesn't exist yet (I guess that's the reason why you have to run build twice). Order of execution: createNativeDepsDirectories, prepareThirdPartyNdkHeaders if needed, prepareHermes and prepareJSC, extractAARHeaders and extractJNIFiles 🙂 

#407  tested on AppCenter

euri10, 2021-10-09:
I have same issue with 
```
  "dependencies": {
    "@xstate/react": "^1.6.1",
    "axios": "^0.22.0",
    "react": "17.0.2",
    "react-native": "^0.65.1",
    "react-native-app-auth": "^6.4.0",
    "react-native-reanimated": "^2.2.2",
    "react-native-vision-camera": "^2.9.0",
    "react-router-native": "^5.2.1",
    "xstate": "^4.25.0"
  },
```
<details>

 <summary>log</summary>

```
❯ npx react-native run-android --verbose
info Running jetifier to migrate libraries to AndroidX. You can disable it using "--no-jetifier" flag.
Jetifier found 937 file(s) to forward-jetify. Using 48 workers...
info Starting JS server...
info Installing the app...
debug Running command "cd android && ./gradlew app:installDebug -PreactNativeDevServerPort=8081"

> Configure project :react-native-vision-camera
Warning: The 'kotlin-android-extensions' Gradle plugin is deprecated. Please use this migration guide (https://goo.gle/kotlin-android-extensions-deprecation) to start working with View Binding (https://developer.android.com/topic/libraries/view-binding) and the 'kotlin-parcelize' plugin.

> Task :react-native-vision-camera:generateJsonModelDebug FAILED
/home/lotso/PycharmProjects/mmc/mmc_frontend_native/node_modules/react-native-vision-camera/android/CMakeLists.txt : C/C++ debug|armeabi-v7a : CMake Warning at /home/lotso/PycharmProjects/mmc/mmc_frontend_native/node_modules/react-native-vision-camera/android/CMakeLists.txt:124 (message):
  VisionCamera linking: FOR_HERMES=false

/home/lotso/PycharmProjects/mmc/mmc_frontend_native/node_modules/react-native-vision-camera/android/CMakeLists.txt : C/C++ debug|armeabi-v7a : CMake Warning at /home/lotso/PycharmProjects/mmc/mmc_frontend_native/node_modules/react-native-vision-camera/android/CMakeLists.txt:124 (message):
  VisionCamera linking: FOR_HERMES=false

41 actionable tasks: 9 executed, 32 up-to-date

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':react-native-vision-camera:generateJsonModelDebug'.
> /home/lotso/PycharmProjects/mmc/mmc_frontend_native/node_modules/react-native-vision-camera/android/CMakeLists.txt : C/C++ debug|armeabi-v7a : CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
  Please set them or make sure they are set and tested correctly in the CMake files:
  JSI_LIB
      linked by target "VisionCamera" in directory /home/lotso/PycharmProjects/mmc/mmc_frontend_native/node_modules/react-native-vision-camera/android

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.

* Get more help at https://help.gradle.org

BUILD FAILED in 24s

error Failed to install the app. Make sure you have the Android development environment set up: https://reactnative.dev/docs/environment-setup.
Error: Command failed: ./gradlew app:installDebug -PreactNativeDevServerPort=8081

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':react-native-vision-camera:generateJsonModelDebug'.
> /home/lotso/PycharmProjects/mmc/mmc_frontend_native/node_modules/react-native-vision-camera/android/CMakeLists.txt : C/C++ debug|armeabi-v7a : CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
  Please set them or make sure they are set and tested correctly in the CMake files:
  JSI_LIB
      linked by target "VisionCamera" in directory /home/lotso/PycharmProjects/mmc/mmc_frontend_native/node_modules/react-native-vision-camera/android

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.

* Get more help at https://help.gradle.org

BUILD FAILED in 24s

    at makeError (/home/lotso/PycharmProjects/mmc/mmc_frontend_native/node_modules/execa/index.js:174:9)
    at /home/lotso/PycharmProjects/mmc/mmc_frontend_native/node_modules/execa/index.js:278:16
    at processTicksAndRejections (internal/process/task_queues.js:95:5)
    at async runOnAllDevices (/home/lotso/PycharmProjects/mmc/mmc_frontend_native/node_modules/@react-native-community/cli-platform-android/build/commands/runAndroid/runOnAllDevices.js:106:5)
    at async Command.handleAction (/home/lotso/PycharmProjects/mmc/mmc_frontend_native/node_modules/react-native/node_modules/@react-native-community/cli/build/index.js:192:9)
```

</details>

mrousavy, 2021-10-09:
Oh yeah, 2.9.0 currently only works on RN 0.66. Will fix this to also work on older RN versions when I find some time on the weekend!

wcandillon, 2021-11-08:
@mrousavy I'm affected by the same issue where I need to run the command twice. I'm using RN v0.64, could that be the issue?

PR #407 (2021-09-06): fix: Fix Android tasks execution
## What
This PR fixes Android tasks execution.

## Changes
This PR changes the tasks order and prepare everything correctly to be executed by task externalNativeBuild.

## Tested on
Local machine for example project and on AppCenter CI.

## Related issues
    * Fixes #338 
    * Fixes #388
PR #694 (2022-01-02): fix: Fix Android build on React Native 0.65 and older

## What

## Changes

## Tested on

## Related issues

* Fixes https://github.com/mrousavy/react-native-vision-camera/issues/654
* Fixes https://github.com/mrousavy/react-native-vision-camera/issues/588
* Fixes #546 
* Fixes #513 
* Fixes #422 
* Fixes #390 
* Fixes #388 
* Fixes #369 
* Fixes #365 
* Fixes #341 
* Fixes #338 
* Fixes #269 
* Overrides #370 
* Overrides #676 
* Resolves https://github.com/mrousavy/react-native-vision-camera/issues/663
closed 2022-01-02 by mrousavy