margelo/react-native-vision-camera #339: 🐛 [Android] Permission issues on API level 21

xympa, 2021-08-06:
### What were you trying to do?

I was trying to request permissions in order to instantiate the `<Camera />` component according to the example application of the plugin` vision-camera-image-labeler` as found [here](https://github.com/mrousavy/vision-camera-image-labeler/blob/main/example/src/App.tsx).

### What happened instead?

The app crashes on the native side when calling the `requestCameraPermission()` method on this device.

According to https://stackoverflow.com/a/32297203 the internal call should be done on a `ContextCompat` which is actually done correctly on the internal 'configure'. But it doesn't seem to be the case when the `requestCameraPermission()` method is invoked on the JS side.

Since on SDK 21 and (presumably < 23) the permissions are accepted at install time the permission should already be accepted if the manifest was correctly configured.

Given this fact, the crash can be easily avoided through a workaround on the JS side such as:

```ts
import {
  getApiLevelSync,
} from 'react-native-device-info';

//...

  React.useEffect(() => {
    (async () => {
//Excuse the repeated calls to the sync method
      if (
        Platform.OS !== 'android' ||
        (Platform.OS === 'android' && getApiLevelSync() >= 23)
      ) {
        const status = await Camera.requestCameraPermission();
        setHasPermission(status === 'authorized');
      } else if (Platform.OS === 'android' && getApiLevelSync() < 23) {
        setHasPermission(true);
      }
    })();
  }, []);
```

This causes the application to behave normally. If my assumptions are correct this should be added to the troubleshooting section of the documentation, while no native fix is implemented.

### Relevant log output

```shell
08-06 11:46:39.341 30888-30922/? E/AndroidRuntime: FATAL EXCEPTION: mqt_native_modules
    Process: com.example.visioncamerabarcodedetector, PID: 30888
    java.lang.RuntimeException: Could not invoke CameraView.requestCameraPermission
        at com.facebook.react.bridge.JavaMethodWrapper.invoke(JavaMethodWrapper.java:383)
        at com.facebook.react.bridge.JavaModuleWrapper.invoke(JavaModuleWrapper.java:151)
        at com.facebook.react.bridge.queue.NativeRunnable.run(Native Method)
        at android.os.Handler.handleCallback(Handler.java:739)
        at android.os.Handler.dispatchMessage(Handler.java:95)
        at com.facebook.react.bridge.queue.MessageQueueThreadHandler.dispatchMessage(MessageQueueThreadHandler.java:27)
        at android.os.Looper.loop(Looper.java:145)
        at com.facebook.react.bridge.queue.MessageQueueThreadImpl$4.run(MessageQueueThreadImpl.java:226)
        at java.lang.Thread.run(Thread.java:818)
     Caused by: java.lang.reflect.InvocationTargetException
        at java.lang.reflect.Method.invoke(Native Method)
        at java.lang.reflect.Method.invoke(Method.java:372)
        at com.facebook.react.bridge.JavaMethodWrapper.invoke(JavaMethodWrapper.java:372)
        at com.facebook.react.bridge.JavaModuleWrapper.invoke(JavaModuleWrapper.java:151) 
        at com.facebook.react.bridge.queue.NativeRunnable.run(Native Method) 
        at android.os.Handler.handleCallback(Handler.java:739) 
        at android.os.Handler.dispatchMessage(Handler.java:95) 
        at com.facebook.react.bridge.queue.MessageQueueThreadHandler.dispatchMessage(MessageQueueThreadHandler.java:27) 
        at android.os.Looper.loop(Looper.java:145) 
        at com.facebook.react.bridge.queue.MessageQueueThreadImpl$4.run(MessageQueueThreadImpl.java:226) 
        at java.lang.Thread.run(Thread.java:818) 
     Caused by: java.lang.NoSuchMethodError: No virtual method requestPermissions([Ljava/lang/String;I)V in class Landroid/app/Activity; or its super classes (declaration of 'android.app.Activity' appears in /system/framework/framework.jar:classes2.dex)
        at com.facebook.react.ReactActivityDelegate.requestPermissions(ReactActivityDelegate.java:161)
        at com.facebook.react.ReactActivity.requestPermissions(ReactActivity.java:108)
        at com.mrousavy.camera.CameraViewModule.requestCameraPermission(CameraViewModule.kt:301)
        at java.lang.reflect.Method.invoke(Native Method) 
        at java.lang.reflect.Method.invoke(Method.java:372) 
        at com.facebook.react.bridge.JavaMethodWrapper.invoke(JavaMethodWrapper.java:372) 
        at com.facebook.react.bridge.JavaModuleWrapper.invoke(JavaModuleWrapper.java:151) 
        at com.facebook.react.bridge.queue.NativeRunnable.run(Native Method) 
        at android.os.Handler.handleCallback(Handler.java:739) 
        at android.os.Handler.dispatchMessage(Handler.java:95) 
        at com.facebook.react.bridge.queue.MessageQueueThreadHandler.dispatchMessage(MessageQueueThreadHandler.java:27) 
        at android.os.Looper.loop(Looper.java:145) 
        at com.facebook.react.bridge.queue.MessageQueueThreadImpl$4.run(MessageQueueThreadImpl.java:226) 
        at java.lang.Thread.run(Thread.java:818)
```

### Device

Samsung Galaxy Tab 4, Android 5, API 21

### VisionCamera Version

2.5.0

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2021-08-09:
Interesting! Let me know if #344 fixes this for you

xympa, 2021-08-09:
Using the PR branch, the hook

```ts
 React.useEffect(() => {
    (async () => {
      const status = await Camera.requestCameraPermission();
      setHasPermission(status === 'authorized');
    })();
  }, []);
```

Works as intended on both the problematic device (Samsung Galaxy Tab 4 ,API 21) and on an Honor 6X (API 24), so pre and post Android M, with no native crashes.

Feel free to close the issue at your discretion, thank you.

PR #344 (2021-08-09): fix: Always return `authorized` for requestPermission on SDK 22 and below
…elow

## What

Apparently requesting permission using the `PermissionAwareActivity` fails when on SDK 22 and below. This PR checks the SDK version, and if it is below 23 we simply always return `authorized`.

## Changes

## Tested on

## Related issues

* Fixes #339
closed 2021-08-09 by mrousavy