margelo/react-native-vision-camera #340: 🐛 [Android] - Does not build within monorepo

batical, 2021-08-06:
### What were you trying to do?

Trying to build on android with a Monorepo with version 2.5.0

Previous version was working fine

### What happened instead?

```
Where:
Build file '[...]/node_modules/react-native-vision-camera/android/build.gradle' line: 220

* What went wrong:
A problem occurred evaluating project ':react-native-vision-camera'.
> Expected directory '[...]/packages/mobile/android/../node_modules/react-native/android' to contain exactly one file, however, it contains no files.
```

I changed the path on :
line 220, 221, 223, 229

but got C compilation error
fatal error: 'jsi/jsi.h' file not found

### Device

Android

### VisionCamera Version

2.5.0

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2021-08-06:
Interesting, does this work if you create a symlink to `node_modules` in the `mobile/` directory?

batical, 2021-08-07:
I will try to give a try next week. (for now I created a patch-package for my case)

zifahm, 2021-08-20:
@batical  what did you fix on patch package?

zifahm, 2021-09-01:
@batical thanks :+1:

batical, 2021-09-01:
working also with last versions 2.6.1 in debug or release mode

Hirbod, 2021-12-18:
@mrousavy I've encountered the same issue right now while trying to install it in my monorepo setup.
Maybe this one is helpful for you: https://github.com/software-mansion/react-native-gesture-handler/pull/1757/files

This is how react-native-gesture-handler fixed that issue for monorepos when trying to determine react-native-reanimated version

Alexispap, 2022-07-01:
I encountered this at RN 0.69.1

mrozkosz, 2022-07-05:
I have the same problem and still not resolved @mrousavy, @Simek, @hirbod, @developerantoniosousa,  @janicduplessis, @hannojg

hirbod, 2022-07-05:
Not sure what you are doing, but I have a monorepo setup (0.68.2) and it's working fine

mrozkosz, 2022-07-05:
As you can see when I build the application I get this error.

[image]

[image]

"react-native-vision-camera": "^2.13.5",
"react-native-reanimated": "^2.9.1"
"react-native": "0.69.1",

java version "16.0.1"

What is the problem? @hirbod, without react-native-reanimated I can build the application, but I need this...

hirbod, 2022-07-05:
I don't know. And I am not sure why you Tag so many random people which aren't even involved

closed 2023-09-01 by mrousavy