margelo/react-native-vision-camera #341: Project with a custom dev client is failing during build🐛

haibert, 2021-08-06:
### What were you trying to do?

I am trying to install the library in my custom dev client project. 
I followed the following instructions: 
1. expo install react-native-vision-camera
2. added the following config plugin to my app.json
"plugins": [
    [
      "react-native-vision-camera",
      {
        "cameraPermissionText": "$(PRODUCT_NAME) needs access to your Camera.",

        // optionally, if you want to record audio:
        "enableMicrophonePermission": true,
        "microphonePermissionText": "$(PRODUCT_NAME) needs access to your Microphone."
      }
    ]
  ]
3. rebuilt the binary using
      **eas build --profile development --platform all**

### What happened instead?

The build fails with the following error on Android  build from EAS build: 
```
[stderr] FAILURE: Build failed with an exception.
[stderr] * What went wrong:
[stderr] Execution failed for task ':react-native-vision-camera:externalNativeBuildDebug'.
[stderr] > Build command failed.
[stderr]   Error while executing process /opt/android/cmake/3.10.2.4988404/bin/ninja with arguments {-C /build/workingdir/build/node_modules/react-native-vision-camera/android/.cxx/cmake/debug/armeabi-v7a VisionCamera}
[stderr]   ninja: Entering directory `/build/workingdir/build/node_modules/react-native-vision-camera/android/.cxx/cmake/debug/armeabi-v7a'
[stderr]   
[stderr]   ninja: error: '../../../../build/third-party-ndk/hermes/jni/armeabi-v7a/libhermes.so', needed by '../../../../build/intermediates/cmake/debug/obj/armeabi-v7a/libVisionCamera.so', missing and no known rule to make it
[stderr] * Try:
[stderr] Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.
[stderr] * Get more help at https://help.gradle.org
[stderr] BUILD FAILED in 6m 35s
Deprecated Gradle features were used in this build, making it incompatible with Gradle 7.0.
Use '--warning-mode all' to show the individual deprecation warnings.
See https://docs.gradle.org/6.8/userguide/command_line_interface.html#sec:command_line_warnings
875 actionable tasks: 875 executed
Error: Gradle build failed with unknown error. Please see logs for the "Run gradlew" phase.
```

following error on iOS build from EAS build: 

```
❌  (ios/Pods/Headers/Public/ReactCommon/ReactCommon/TurboModuleUtils.h:13:10)

  11 | #include <string>
  12 | 
> 13 | #include <folly/Optional.h>
     |          ^ 'folly/Optional.h' file not found
  14 | #include <jsi/jsi.h>
  15 | 
  16 | #include <ReactCommon/CallInvoker.h>
› Compiling react-native-vision-camera Pods/VisionCamera » JSIUtils.mm
› Compiling react-native-vision-camera Pods/VisionCamera » JSConsoleHelper.mm
› Compiling react-native-vision-camera Pods/VisionCamera » FrameProcessorUtils.mm
› Compiling react-native-vision-camera Pods/VisionCamera » FrameProcessorRuntimeManager.mm
▸ ** ARCHIVE FAILED **
▸ The following build commands failed:
▸ 	CompileC /Users/expo/Library/Developer/Xcode/DerivedData/EventShare-cvvxuvgbosgptrfgtqzckczinmol/Build/Intermediates.noindex/ArchiveIntermediates/EventShare/IntermediateBuildFilesPath/Pods.build/Debug-iphoneos/VisionCamera.build/Objects-normal/arm64/RCTBridge+runOnJS.o /Users/expo/workingdir/build/node_modules/react-native-vision-camera/ios/React\ Utils/RCTBridge+runOnJS.mm normal arm64 objective-c++ com.apple.compilers.llvm.clang.1_0.compiler
▸ (1 failure)
** ARCHIVE FAILED **
The following build commands failed:
	CompileC /Users/expo/Library/Developer/Xcode/DerivedData/EventShare-cvvxuvgbosgptrfgtqzckczinmol/Build/Intermediates.noindex/ArchiveIntermediates/EventShare/IntermediateBuildFilesPath/Pods.build/Debug-iphoneos/VisionCamera.build/Objects-normal/arm64/RCTBridge+runOnJS.o /Users/expo/workingdir/build/node_modules/react-native-vision-camera/ios/React\ Utils/RCTBridge+runOnJS.mm normal arm64 objective-c++ com.apple.compilers.llvm.clang.1_0.compiler
(1 failure)
Exit status: 65
+-------------+-------------------------+
[... 315 lines cut ...]
to troubleshoot build and signing issues: https://docs.fastlane.tools/codesigning/getting-started/
Before submitting an issue on GitHub, please follow the guide above and make
sure your project is set up correctly.
fastlane uses `xcodebuild` commands to generate your binary, you can see the
the full commands printed out in yellow in the above log.
Make sure to inspect the output above, as usually you'll find more error information there
[stderr] [!] Error building the application - see the log above
Error: Fastlane build failed with unknown error. Please refer to the "Run fastlane" and "Xcode Logs" phases.
Fastlane errors in most cases are not printed at the end of the output, so you may not find any useful information in the last lines of output when looking for an error message.
```

### Device

iPhone 12

### VisionCamera Version

2.5.0

### Additional information

- [X] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

Kypsis, 2021-08-06:
Partially same issue as #338 and probably same solution at least for Android part.

haibert, 2021-08-06:
Hmmmm, I think I tried twice but i'll try a few times in a row just to make sure.

haibert, 2021-08-06:
I rebuilt 4 times but the issue persists :(

haibert, 2021-08-07:
[image]

Im not sure if this is relevant but after the build failed I came back to my app.json and it says it cannot find the module.. but it does exists inside my node_modules directory. I can provide a full repo if necessary..

haibert, 2021-08-08:
Update: I created a fresh expo project to get rid of all other potential issues and I still get the "folly" fail error. 

So its super easy to recreate this error: 
1. expo init vision-cam-test-project
2. yarn add expo-dev-client
3. eas build:configure
4. eas device:create
5. vision camera installation instructions

mrousavy, 2021-08-09:
I think the folly install error is happening because EAS still uses React Native 0.63, I tried working on a fix in #273 but didn't find one that works smoothly for every version - once the new Expo SDK releases this should work fine.

As for the Android build error - this must be happening in the `build.gradle` where something is requiring `libhermes.so`, but I added the `prepareHermes` step to the tasks and it still happens: https://github.com/mrousavy/react-native-vision-camera/blob/ee762ce5f5f6d6738d3e62b9042c90322f0b09ef/android/build.gradle#L458

I'd appreciate some help here as I have little experience with gradle...

Kypsis, 2021-08-09:
I managed to get into the same state of affairs on development mac (not CI) when specifying a version of gradle (`distributionUrl` in `gradle-wrapper.properties`) that is not yet on device.

haibert, 2021-08-09:
@mrousavy so does expo need to support RN 0.64 for us to be able to use this library? I'm not sure if they are going to add this in version SDK 43, maybe they will give you some information on that since you've joined the team?

haibert, 2021-08-09:
> I managed to get into the same state of affairs on development mac (not CI) when specifying a version of gradle (`distributionUrl` in `gradle-wrapper.properties`) that is not yet on device. 
> 
> 
> 
> And it is gone the second time after trying to build...

Did you do this by running expo prebuild ? Sounds like you're modifying the android files directly correct ? I'm trying to keep the project in a managed state.

mrousavy, 2021-08-09:
Android + C++ is hard 😓

Kypsis, 2021-08-09:
Should have specified that I don't use Expo and the retry building again is quite literally that in CI, it just retries the building Android step again without restarting whole CI process. I assumed due to Fastlane your build was failing in CI environment.

haibert, 2021-08-09:
@mrousavy just to make sure I understood correctly, expo 43 will update the version of RN used from .63 to .64 which might fix the issue ?

mrousavy, 2021-08-09:
> expo 43 will update the version of RN used from .63 to .64
I am not sure about that, either 43 or 44.

> which might fix the issue
yes, I believe this is a RN 0.63 issue. @axeldelafosse and @matheusmatos made some very interesting research here: https://github.com/mrousavy/react-native-vision-camera/issues/195 this could be used to fix the issue, but it's a bit of a complicated setup :/
you can either patch it to make it work (using patch-package) or if someone opens up a PR for this I can merge it for now and revert the change once an Expo SDK version with RN 0.64 is released.

haibert, 2021-08-09:
Just got confirmation from Evan Bacon that SDK 43 will have 0.64, it's been merged.  🙏🏻
https://github.com/expo/expo/pull/13793

mrousavy, 2021-08-09:
bacon's the best

- also that PR by @Kudo is huge 😳

mrousavy, 2021-08-09:
so yeah VisionCamera will work with Expo 43, I'll focus on that. Is it a thing that people don't upgrade their Expo SDKs or can I assume that it's an obvious requirement?

haibert, 2021-08-09:
It can definitely be an obvious requirement and most people update, because expo just gets better and better. Rarely any breaking changes anyway from what i've seen...

PR #694 (2022-01-02): fix: Fix Android build on React Native 0.65 and older

## What

## Changes

## Tested on

## Related issues

* Fixes https://github.com/mrousavy/react-native-vision-camera/issues/654
* Fixes https://github.com/mrousavy/react-native-vision-camera/issues/588
* Fixes #546 
* Fixes #513 
* Fixes #422 
* Fixes #390 
* Fixes #388 
* Fixes #369 
* Fixes #365 
* Fixes #341 
* Fixes #338 
* Fixes #269 
* Overrides #370 
* Overrides #676 
* Resolves https://github.com/mrousavy/react-native-vision-camera/issues/663
closed 2022-01-02 by mrousavy