margelo/react-native-vision-camera #349: ❓ Face detection with Frame Processors

xuraganx, 2021-08-10:
### Question

Hello! I would like to use VisionCamera to implement live face detection functionality (detect face, outline the face with a rectangle with real time updates and crop some images with face which has better score). It seems like I should use Frame Processors for this. So, is there implemented Frame Processors plugins for face detection? If not, is there way to implement frame processors plugin without writing native code? Thanks!

### VisionCamera Version

2.5.0

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

mrousavy, 2021-08-10:
Hi! Yes, this is a perfect case for Frame Processors. To show the rectangle in realtime (possibly even with smooth animations) you can assign the coordinates to a Reanimated Shared Value inside the Frame Processor which you can then use to display a rectangle. (See [the cat bounds example in the docs](https://mrousavy.github.io/react-native-vision-camera/docs/guides/frame-processors#interacting-with-frame-processors))

> So, is there implemented Frame Processors plugins for face detection?

No, there is none available _yet_ - but I played around with it a while ago and created one within 5 minutes but I deleted it already. It's very easy to create it yourself though, should be less than 10 lines of code.

> is there way to implement frame processors plugin without writing native code?

no, JavaScript is too slow to run these algorithms. You need to write a few (~10) lines of native code, but don't be afraid - Swift and Java/Kotlin is very easy!

xuraganx, 2021-08-10:
Thanks a lot!
If I will use MLKit do I need to extra setup something like firebase project or something else or it will similar to ImageLabeler plugin?

mrousavy, 2021-08-10:
If you use MLKit that will look almost exactly the same as the Image Labeler plugin, I don't think you need to set up firebase in your AppDelegate for this.. That should all be mentioned in their docs!

rodgomesc, 2021-08-13:
@xuraganx you dont need to setup firebase anymore, mlkit for firebase, and the new mlkit are two different things

> On June 3, 2020, we made some changes to ML Kit for Firebase to better distinguish the on-device APIs from cloud based APIs. The current set of APIs is now split into the following two products:
> 
> A new product, simply called ML Kit, which will contain all the on-device APIs
> 
> Firebase Machine Learning, focused on cloud-based APIs and custom model deployment.
> 
> This change will also make it easier to integrate ML Kit into your app if you only need an on-device solution. This document explains how to migrate your app from the Firebase ML Kit SDK to the new ML Kit SDK.

more info on https://developers.google.com/ml-kit/migration

rodgomesc, 2021-08-20:
@xuraganx  you can check a working example in android of face detection here https://github.com/mrousavy/react-native-vision-camera/issues/376

closed 2021-09-24 by mrousavy

CodingWithNobody, 2023-05-01:
Try this https://www.youtube.com/watch?v=mv4c2uTxR4U