margelo/react-native-vision-camera #354: 🐛 Android: Changing device front to back mid video record stops recording

jehartzog, 2021-08-13:
### What were you trying to do?

We wanted to allow users to swap between front/back devices

### Reproduceable Code

```tsx
const devices = useCameraDevices();
  const isFocused = useIsFocused();
  const [cameraType, setCameraType] = React.useState<'front' | 'back'>('front');

  const device = cameraType === 'front' ? devices.front : devices.back;
  const preset = 'hd-1280x720';

  if (!device) return null;

  return (
   <>
      <Camera
        style={StyleSheet.absoluteFill}
        device={device}
        preset={preset}
        isActive={isFocused}
        video
        audio
        ref={cameraRef}
      />
      <TouchableOpacity style={localStyles.flip} onPress={() => setCameraType(cameraType === 'front' ? 'back' : 'front' )} />
    </>
   );
```

### What happened instead?

iOS: Works great. A few frames of oddness when we swap, but nothing big
Android: It swaps the preview, but `onRecordingFinished` is called.

### Device

Pixel 2 XL, OnePlus 8T

### VisionCamera Version

2.5.0

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2021-08-14:
I don't think this is supported by CameraX. I've created an issue in Google's CameraX issue tracker [here](https://issuetracker.google.com/issues/177918195) but haven't received any responses.

You can easily take a look at this yourself by playing around with the code and shooting me a PR if you manage to get it working :)

jehartzog, 2021-08-14:
@mrousavy Awesome for the response, thanks. We were not actually sure this was possible but wanted to explore a bit. This lib allows us to do this swap using iOS. We did noticed most Android apps don't allow this, but did find that the Android Snapchat app supports it. I see that you actually have more info on how they do this.

I guess this is probably an enhancement request rather than a bug, and one likely that needs to come from underlying Android lib, so happy to close this issue. Thanks again!

mrousavy, 2021-08-14:
don't close, let's keep track of it here.

mrousavy, 2023-08-18:
Hey! I've rewritten the entire Android codebase of VisionCamera from CameraX to Camera2 in the efforts of [✨ VisionCamera V3](https://github.com/mrousavy/react-native-vision-camera/issues/1376).

I just now completed the Camera2 rewrite and I believe the core structure is running, but there might be some edge cases to iron out. Can you try and test the PR #1674 for me to see if you can still reproduce this issue here?

Here's an instruction on how you can test that: https://github.com/mrousavy/react-native-vision-camera/pull/1674#issuecomment-1684104217

If the issue cannot be reproduced with that version/PR anymore, then hoorayy, I fixed it! 🎉 
Otherwise please let me know and I'll keep this issue open to keep track of it.

Thank you!

closed 2023-09-01 by mrousavy