margelo/react-native-vision-camera #366: 🐛Android : Non-display rotation not supported with SurfaceView / PERFORMANCE mode.

Kypsis, 2021-08-18:
### What were you trying to do?

When starting an app in landscape mode and/or navigating to the camera view from another view while in landscape mode the preview is stretched out and the media gets captured sideways (as if the device is still in portrait mode). On rotation the preview corrects itself. About 20% of time when navigating between camera view and another view in landscape the bug reappears.

This bug can be reproduced 100% of the time if forcing the screen to be locked to landscape for the camera view or when quick refreshing repeatedly when in landscape camera view (might take upwards of 20 tries or might have 3 in row after 2 tries).

### What happened instead?

-

### Relevant log output

```shell
2021-08-18 13:10:48.645 4266-4266/com.mrousavy.camera.example I/VideoCapture: source: 5 audioSampleRate: 48000 channelConfig: 16 audioFormat: 2 bufferSize: 3840
2021-08-18 13:10:48.647 4266-4266/com.mrousavy.camera.example D/DeferrableSurface: Surface created[total_surfaces=6, used_surfaces=3](androidx.camera.core.impl.ImmediateSurface@ca5afa9}
2021-08-18 13:10:48.648 4266-4266/com.mrousavy.camera.example D/CameraOrientationUtil: getRelativeImageRotation: destRotationDegrees=0, sourceRotationDegrees=90, isOppositeFacing=true, result=90
2021-08-18 13:10:48.648 4266-4266/com.mrousavy.camera.example I/CameraView.performance: Session configured in 96 ms! Camera: androidx.camera.lifecycle.LifecycleCamera@77a9732
2021-08-18 13:10:48.648 4266-4266/com.mrousavy.camera.example I/CameraView: invokeOnInitialized()
2021-08-18 13:10:48.650 4266-4266/com.mrousavy.camera.example D/PreviewView: Surface requested by Preview.
2021-08-18 13:10:48.652 4266-4266/com.mrousavy.camera.example D/StreamStateObserver: Update Preview stream state to IDLE
2021-08-18 13:10:48.652 4266-4344/com.mrousavy.camera.example I/ReactNativeJS: Camera initialized!
2021-08-18 13:10:48.652 4266-4266/com.mrousavy.camera.example D/PreviewView: Preview transformation info updated. TransformationInfo{cropRect=Rect(0, 0 - 1920, 1080), rotationDegrees=90, targetRotation=0}
2021-08-18 13:10:48.652 4266-4266/com.mrousavy.camera.example D/PreviewTransform: Transformation info set: TransformationInfo{cropRect=Rect(0, 0 - 1920, 1080), rotationDegrees=90, targetRotation=0} 1920x1080 false
2021-08-18 13:10:48.655 4266-4344/com.mrousavy.camera.example I/ReactNativeJS: Re-rendering camera page with active camera. Device: "back (0)" (4000x2000 @ 60fps)
2021-08-18 13:10:48.663 4266-4266/com.mrousavy.camera.example D/SurfaceViewImpl: Surface created.
2021-08-18 13:10:48.663 4266-4266/com.mrousavy.camera.example D/SurfaceViewImpl: Surface changed. Size: 1920x1080
2021-08-18 13:10:48.665 4266-4266/com.mrousavy.camera.example D/SurfaceViewImpl: Surface set on Preview.
2021-08-18 13:10:48.665 4266-4266/com.mrousavy.camera.example E/PreviewTransform: Non-display rotation not supported with SurfaceView / PERFORMANCE mode.
2021-08-18 13:10:48.806 4266-4377/com.mrousavy.camera.example W/GraphicBufferSource: released unpopulated slots: [0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31, 32, 33, 34, 35, 36, 37, 38, 39, 40, 41, 42, 43, 44, 45, 46, 47, 48, 49, 50, 51, 52, 53, 54, 55, 56, 57, 58, 59, 60, 61, 62, 63]
2021-08-18 13:10:48.809 4266-4377/com.mrousavy.camera.example D/UseCaseAttachState: Active and attached use case: [] for camera: 0
2021-08-18 13:10:48.809 4266-4377/com.mrousavy.camera.example D/UseCaseAttachState: Active and attached use case: [] for camera: 0
2021-08-18 13:10:48.810 4266-4377/com.mrousavy.camera.example D/Camera2CameraImpl: {Camera@411c358[id=0]} Use case androidx.camera.core.VideoCapture@db6f47e ACTIVE
2021-08-18 13:10:48.810 4266-4377/com.mrousavy.camera.example D/UseCaseAttachState: Active and attached use case: [] for camera: 0
2021-08-18 13:10:48.811 4266-4377/com.mrousavy.camera.example D/Camera2CameraImpl: {Camera@411c358[id=0]} Use case ImageCapture:androidx.camera.core.ImageCapture-dad931b5-98e9-4272-a022-dd7336aa8274 ACTIVE
2021-08-18 13:10:48.811 4266-4377/com.mrousavy.camera.example D/UseCaseAttachState: Active and attached use case: [] for camera: 0
2021-08-18 13:10:48.811 4266-4377/com.mrousavy.camera.example D/Camera2CameraImpl: {Camera@411c358[id=0]} Use case Preview:androidx.camera.core.Preview-37cb4361-9622-4717-bddb-54e2afb184e1 INACTIVE
2021-08-18 13:10:48.811 4266-4377/com.mrousavy.camera.example D/UseCaseAttachState: Active and attached use case: [] for camera: 0
2021-08-18 13:10:48.811 4266-4377/com.mrousavy.camera.example D/Camera2CameraImpl: {Camera@411c358[id=0]} Use case androidx.camera.core.VideoCapture@db6f47e ACTIVE
2021-08-18 13:10:48.811 4266-4377/com.mrousavy.camera.example D/UseCaseAttachState: Active and attached use case: [] for camera: 0
2021-08-18 13:10:48.811 4266-4377/com.mrousavy.camera.example D/Camera2CameraImpl: {Camera@411c358[id=0]} Use case ImageCapture:androidx.camera.core.ImageCapture-dad931b5-98e9-4272-a022-dd7336aa8274 ACTIVE
2021-08-18 13:10:48.812 4266-4377/com.mrousavy.camera.example D/UseCaseAttachState: Active and attached use case: [] for camera: 0
2021-08-18 13:10:48.812 4266-4377/com.mrousavy.camera.example D/UseCaseAttachState: Active and attached use case: [] for camera: 0
2021-08-18 13:10:48.812 4266-4377/com.mrousavy.camera.example D/Camera2CameraImpl: {Camera@411c358[id=0]} Use cases [Preview:androidx.camera.core.Preview-37cb4361-9622-4717-bddb-54e2afb184e1, androidx.camera.core.VideoCapture@db6f47e, ImageCapture:androidx.camera.core.ImageCapture-dad931b5-98e9-4272-a022-dd7336aa8274] now ATTACHED
2021-08-18 13:10:48.812 4266-4377/com.mrousavy.camera.example D/UseCaseAttachState: All use case: [androidx.camera.core.VideoCapture-0a092f75-7367-432a-a76b-bbacdbc56284230093950, androidx.camera.core.Preview-37cb4361-9622-4717-bddb-54e2afb184e1186893881, androidx.camera.core.ImageCapture-dad931b5-98e9-4272-a022-dd7336aa8274215526879] for camera: 0
2021-08-18 13:10:48.812 4266-4377/com.mrousavy.camera.example D/Camera2CameraImpl: mMeteringRepeating is ATTACHED, SessionConfig Surfaces: 3, CaptureConfig Surfaces: 1
2021-08-18 13:10:48.813 4266-4377/com.mrousavy.camera.example D/UseCaseAttachState: Active and attached use case: [androidx.camera.core.VideoCapture-0a092f75-7367-432a-a76b-bbacdbc56284230093950, androidx.camera.core.ImageCapture-dad931b5-98e9-4272-a022-dd7336aa8274215526879] for camera: 0
2021-08-18 13:10:48.813 4266-4377/com.mrousavy.camera.example D/Camera2CameraImpl: {Camera@411c358[id=0]} Resetting Capture Session
```

### Device

Pixel 4a

### VisionCamera Version

current react-native-vision-camera example

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

Kypsis, 2021-08-18:
Should be fixed with [https://github.com/mrousavy/react-native-vision-camera/pull/367](https://github.com/mrousavy/react-native-vision-camera/pull/367)

mrousavy, 2021-08-19:
Can you test if https://github.com/mrousavy/react-native-vision-camera/pull/368 fixes the issue for you?

Kypsis, 2021-08-19:
#368 works to solve the initial issue of starting in landscape but does not fix the issue if previous screen was portrait and camera screen is in landscape. Probably due to the rotation value being portrait when you "enter" the screen and then it still does the rotation effect/animation to landscape.

If you add `rotation = context.displayRotation` to `private fun updateLifecycleState()` then it works for the locked screen case as well. Not sure though if that is the right spot for it though to update the real post rotation "animation" state.

mrousavy, 2021-08-20:
@Kypsis pushed another commit to that PR, can you let me know if that fixes the issue for you? If not, I'm just going to put it in the `updateLifecycleState` func I guess

Kypsis, 2021-08-20:
I'm unavailable until September so I'm unable to check.

But I made a fork of the repo and applied the fix in updateLifecycleState and using that in our project temporarily. Unless someone else can verify then this issue can probably be kept open or closed as you see fit. 🙂

Menardi, 2021-08-21:
@mrousavy I'm seeing a similar issue to this on 2.6.1. My app is limited to portrait mode, but if the phone is held in landscape orientation when the app is opened (it opens straight to the camera), then the image is distorted. The PR at https://github.com/mrousavy/react-native-vision-camera/pull/368 fixes this for me :+1:

mrousavy, 2021-08-22:
Okay so the first bug is fixed, can you test this too?:

> the issue if previous screen was portrait and camera screen is in landscape. Probably due to the rotation value being portrait when you "enter" the screen and then it still does the rotation effect/animation to landscape.

Kypsis, 2021-09-01:
I can also confirm that both bugs have been fixed.

mrousavy, 2021-09-01:
awesome! thanks for testing